zynthian/zynthian-sys #13: The 3rd of update.sh call causes compiling problems for Linuxsampler

dcoredump, 2017-09-02:
...
```
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I/usr/local/include/libgig -Wreturn-type -ffast-math -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mneon-for-64bits -mfloat-abi=hard -mvectorize-with-neon-quad -pthread -MT lscpparser.lo -MD -MP -MF .deps/lscpparser.Tpo -c lscpparser.cpp  -fPIC -DPIC -o .libs/lscpparser.o
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I/usr/local/include/libgig -Wreturn-type -ffast-math -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mneon-for-64bits -mfloat-abi=hard -mvectorize-with-neon-quad -pthread -MT lscpserver.lo -MD -MP -MF .deps/lscpserver.Tpo -c lscpserver.cpp  -fPIC -DPIC -o .libs/lscpserver.o
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I/usr/local/include/libgig -Wreturn-type -ffast-math -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mneon-for-64bits -mfloat-abi=hard -mvectorize-with-neon-quad -pthread -MT lscp_shell_reference.lo -MD -MP -MF .deps/lscp_shell_reference.Tpo -c lscp_shell_reference.cpp  -fPIC -DPIC -o .libs/lscp_shell_reference.o
lscp_shell_reference.cpp:2361:7: error: stray '\' in program
       "DEPRECATED: THIS COMMAND WILL DISAPPEAR!\n\nThe front-end   "RIVER,ACTIVE\"\n\n\".\"\n\n"
       ^
lscp_shell_reference.cpp:2361:69: error: expected '}' before 'RIVER'
       "DEPRECATED: THIS COMMAND WILL DISAPPEAR!\n\nThe front-end   "RIVER,ACTIVE\"\n\n\".\"\n\n"
                                                                     ^
lscp_shell_reference.cpp:2361:69: error: expected '}' before 'RIVER'
lscp_shell_reference.cpp:2361:69: error: expected ',' or ';' before 'RIVER'
lscp_shell_reference.cpp:2362:5: error: expected declaration before '}' token
     },
     ^
Makefile:437: recipe for target 'lscp_shell_reference.lo' failed
make[3]: *** [lscp_shell_reference.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
lscpserver.cpp: In member function 'String LinuxSampler::LSCPServer::GetChannelInfo(uint)':
lscpserver.cpp:1253:101: warning: 'LinuxSampler::MidiInputDevice* LinuxSampler::SamplerChannel::GetMidiInputDevice()' is deprecated (declared at ../Sampler.h:258) [-Wdeprecated-declarations]
         result.Add("MIDI_INPUT_DEVICE", GetMidiInputDeviceIndex(pSamplerChannel->GetMidiInputDevice()));
                                                                                                     ^
lscpserver.cpp:1254:73: warning: 'int LinuxSampler::SamplerChannel::GetMidiInputPort()' is deprecated (declared at ../Sampler.h:237) [-Wdeprecated-declarations]
         result.Add("MIDI_INPUT_PORT", pSamplerChannel->GetMidiInputPort());
                                                                         ^
lscpserver.cpp: In member function 'String LinuxSampler::LSCPServer::SetMIDIInputPort(uint, uint)':
lscpserver.cpp:2004:51: warning: 'void LinuxSampler::SamplerChannel::SetMidiInputPort(int)' is deprecated (declared at ../Sampler.h:169) [-Wdeprecated-declarations]
         pSamplerChannel->SetMidiInputPort(MIDIPort);
                                                   ^
lscpserver.cpp: In member function 'String LinuxSampler::LSCPServer::SetMIDIInputDevice(uint, uint)':
lscpserver.cpp:2035:52: warning: 'void LinuxSampler::SamplerChannel::SetMidiInputDevice(LinuxSampler::MidiInputDevice*)' is deprecated (declared at ../Sampler.h:143) [-Wdeprecated-declarations]
         pSamplerChannel->SetMidiInputDevice(pDevice);
                                                    ^
lscpserver.cpp: In member function 'String LinuxSampler::LSCPServer::SetMIDIInputType(String, uint)':
lscpserver.cpp:2073:52: warning: 'void LinuxSampler::SamplerChannel::SetMidiInputDevice(LinuxSampler::MidiInputDevice*)' is deprecated (declared at ../Sampler.h:143) [-Wdeprecated-declarations]
         pSamplerChannel->SetMidiInputDevice(pDevice);
                                                    ^
lscpserver.cpp: In member function 'String LinuxSampler::LSCPServer::SetMIDIInput(uint, uint, uint, uint)':
lscpserver.cpp:2094:83: warning: 'void LinuxSampler::SamplerChannel::SetMidiInput(LinuxSampler::MidiInputDevice*, int, LinuxSampler::midi_chan_t)' is deprecated (declared at ../Sampler.h:207) [-Wdeprecated-declarations]
         pSamplerChannel->SetMidiInput(pDevice, MIDIPort, (midi_chan_t) MIDIChannel);
                                                                                   ^
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I/usr/local/include/libgig -Wreturn-type -ffast-math -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mneon-for-64bits -mfloat-abi=hard -mvectorize-with-neon-quad -pthread -MT lscpparser.lo -MD -MP -MF .deps/lscpparser.Tpo -c lscpparser.cpp -o lscpparser.o >/dev/null 2>&1
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I/usr/local/include/libgig -Wreturn-type -ffast-math -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mneon-for-64bits -mfloat-abi=hard -mvectorize-with-neon-quad -pthread -MT lscpserver.lo -MD -MP -MF .deps/lscpserver.Tpo -c lscpserver.cpp -o lscpserver.o >/dev/null 2>&1
make[3]: Leaving directory '/zynthian/zynthian-sw/linuxsampler/src/network'
Makefile:677: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/zynthian/zynthian-sw/linuxsampler/src'
Makefile:498: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/zynthian/zynthian-sw/linuxsampler'
Makefile:402: recipe for target 'all' failed
make: *** [all] Error 2
```
...

jofemodo, 2017-09-05:
I can't reproduce this problem in a fresh Gorgona after running "update_zynthian.sh" until been up-to-date.

jofemodo, 2017-09-05:
I suposse it's related to #12 and it's should be fixed now.

closed 2017-09-05 by jofemodo

dcoredump, 2017-09-05:
Ok, fixed it by myself by completely removing /zynthian/zynthian-sw/linuxsampler and /zynthian/zynthian-plugins/lv2/linuxsampler.lv2. After calling install_linuxsampler.sh again everything works. Thx for testing and sorry for complaining.

jofemodo, 2018-05-08:
OK! Let's try fixing the LinuxSampler version to a "stable" one.

guysoft, 2018-05-08:
Its working again, its really strange, its coming and going and I am not sure what is changed

guysoft, 2018-05-08:
yes look at the nightly build warning

jofemodo, 2018-05-08:
Ups! I also see this error:
```

lv2_ttl_generator: pthread_mutex_lock.c:81: __pthread_mutex_lock: Assertion `mutex->__data.__owner == 0' failed.
qemu: uncaught target signal 6 (Aborted) - core dumped
./scripts/generate-ttl.sh: line 27: 22279 Aborted                 $GEN ./$FILE
Makefile:62: recipe for target 'gen_lv2' failed
make[1]: *** [gen_lv2] Error 134
make[1]: Leaving directory '/home/pi/zynthian-sw/plugins/DISTRHO-Ports'
Makefile:104: recipe for target 'lv2' failed
make: *** [lv2] Error 2
```
while building DISTHRO. I will investigate ...

Regards,

commit be8fb1ee (2017-09-05): Fix LinuxSampler install recipe: Fix #12, fix #13
commit d338d64a (2017-09-06): Fix #12, fix #13: Build from scratch => Delete linuxsampler directory before ;-)