zynthian/zynthian-sys #21: Diagnosis Script for helping in error solving and improving support proccess

jofemodo, 2017-11-30:
**@mheidt proposal:**

Could you write a diagnosis script, that displays everything we need to know for all those support cases, when the ERROR-Screen on the zynthian shows up?

In webconf, should we call the menu item "Maintenance"?

mheidt, 2017-12-12:
When we are at it, we should have the backup/restore in mind as well.
Following usecase should be possible:
Backup System, Update, Restore System.
And the system should work afterwards :)
The issue now is that zynthian_setenv need to be backed up.
But can contain new values from the update.
We need two setenvs. The real values, that have been changed, need to be stored in a separate file.
This file can be backed up. Not the zynthian_setenv.

jofemodo, 2017-12-18:
In fact, it's more complex yet, because some updates are not backward-compatible (file-name or directory structure changes). Most of times, running "update_zynthian_sys.sh" after restoring should fix the problems, but we can't be completely sure that this will work in every case. 
The "update_zynthian_sys.sh" script will create the new **mandatory** env-vars if it doesn't exist. Also, the webconf tool will add new env-vars to the config file when needed. Finally, when an env-var is not defined in the config file "zynthian_envars.sh", it will take the default value defined in the UI config initialization.
Regarding the changes in file names/location and directory structure, the script "zynthian_update_sys" will try to fix it, for working with current code, but again, i'm not sure that this will work in every case with the backup/restoring system.
The only way of implementing a truly reliable backup/restoring system is saving the git versions for every repo in the backup and "rolling back" the git versions when restoring.

jofemodo, 2017-12-18:
OK. Let's try calling "update_zynthian_sys.sh" after restoring, and test the whole system with the workflow you propose:

+ New SD "Gorgona Next"
+ Customize Config, add some presets, soundfonts, snapshots, ...
+ Backup system
+ [time lapse: changes in repositories]
+ New SD "Gorgona Next"
+ Update
+ Restore Backup + update_zynthian_sys.sh
+ Reboot

And will see how it works ;-)

riban-bw, 2024-03-12:
Backup / restore is implemented. Ability to report issues directly from webconf with system config auto-populated is implemented in oram. I will close this issue as most of the issues are implemented or obselete. There could be further improvement to capture failure conditions automatically but let's consider those under a new ticket in the issue tracker.

closed 2024-03-12 by riban-bw