zynthian/zynthian-sys #24: install_jpmidi.sh recipe links to broken url

guysoft, 2018-01-31:
``setup_system_rbpi_raspbian_lite_stretch.sh`` gives this error:
```
+ bash /zynthian/zynthian-sys/scripts/recipes/install_jpmidi.sh
--2018-01-31 14:44:14--  http://juliencoder.de/jpmidi/jpmidi-0.2.tar.gz
Resolving juliencoder.de (juliencoder.de)... 138.201.83.61, 2a01:4f8:172:1ed8::3
Connecting to juliencoder.de (juliencoder.de)|138.201.83.61|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2018-01-31 14:44:15 ERROR 404: Not Found.
```

Line that needs to be fixed: https://github.com/zynthian/zynthian-sys/blob/master/scripts/recipes/install_jpmidi.sh#L3

guysoft, 2018-01-31:
From http://juliencoder.de/ :
```
18.01.2018 - Finally removed jpmidi and songanalysis. jack_midi_clock is temporarily unavailable, but will be back soon.
12.01.2018 - NOTICE: the hosted software package jpmidi and songanalysis will be removed shortly. It appears that songanalysis is no longer used, try aubio, the bpm-tools or sonic-annotator. jpmidi is now mirrored on github, see the Github jpmidi project page. jack_midi_clock will still be hosted, since at least I use it and haven't find an alternative solution.
```

jerash, 2018-01-31:
just created a v0.21 release here https://github.com/jerash/jpmidi/archive/v0.21.tar.gz
this is the source code compress, is this what is needed for the script ?

jofemodo, 2018-02-01:
OK! I've updated the install recipe;-)

closed 2018-02-01 by jofemodo

guysoft, 2018-02-01:
For future reference: https://github.com/zynthian/zynthian-sys/commit/e910f10416e84a195de2ad3f18096b38fbdbde55
(Btw if you commit with the message " fixes #25 " it would automatically close the issue)

jofemodo, 2018-02-01:
OK! Yes ... i know that ... but i always forget it ;-)