zynthian/zynthian-sys #25: waf: error: no such option: --lv2-only when running recipes/install_guitarix.sh

guysoft, 2018-01-31:
When running ``setup_system_rbpi_raspbian_lite_stretch.sh`` I get the error:

```
waf [command] [options]

Main commands (example: ./waf build -j4)
  build     : builds the project
  clean     : removes the build files
  configure : configures the project
  dist      : makes a tarball for redistributing the sources
  distcheck : checks if the sources compile (tarball from 'dist')
  distclean : removes the build directory
  etags     : Create an Emacs ETAGS file for the src directory
  install   : installs the build files
  uninstall : removes the installed files
  update_pot: extract all text for localization into po/untitled.pot

waf: error: no such option: --lv2-only
```

From what I can see, ``setup_plugins_rbpi.sh`` calls ``recipes/install_guitarix.sh``. which has the line:

```
./waf configure --no-lv2-gui --lv2-only --disable-sse --lv2dir=$ZYNTHIAN_PLUGINS_DIR/lv2 --no-avahi --no-bluez --no-ladspa --no-faust 
```

guysoft, 2018-01-31:
Also found that if you remove the lv2-only flag you also get:
```
waf: error: no such option: --no-ladspa
```

jofemodo, 2018-02-01:
Sorry, don't use the "stretch" script. It's not finished nor tested at all. It's a work in progress and i commited it by error to the master branch.

The only build script that is currently updated and working is this:

setup_system_rbpi_minibian_jessie.sh

Anyway, i've fixed the guitarix install recipe, including the needed dependencies ;-)

Regards,

closed 2018-02-01 by jofemodo

guysoft, 2018-02-01:
@jofemodo Hey, so not sure if this is the place to open this, but we could continue this by email. Since I am not 100% how you work and maintain stuff here.
You probbably saw I've been locating and sending pull requests about that script.

I'm the developer of OctoPi and FullPageOS, and I was looking in to writing a script that builds automatically a zip with an img file of zynthian using [CustomPiOS](https://github.com/guysoft/CustomPiOS). it means we can even make nightly/weekly builds. I reticently 3D printed and built my own electric violin and I would really love to have zynthian as my FX system :)

I am also trying to build zynthian on top of [RealtimePi](https://github.com/guysoft/RealtimePi), which is on Stretch, Jessie does not work with it.
I don't mind running it several times and keeping it up to date, seems to me IMHO that you will have to move to stretch eventually as Jessie is falling out of date. I can push the current distro build system if you want to have a look.

jofemodo, 2018-02-05:
Hi Guy!

Sorry for the delay in the response. I'm quite busy these days. My
second daughter/son will be out in a few days, my new house must be
reformed, zynthian orders must be prepared and shipped, and ....
puuuuf!! my head is exploding!!! ;-DDD

Regarding your proposal ...

As you can imagine, preparing the zynthian images is a hard work for me
and i really would be very happy of getting some help with it.

CustomPiOS seems to be a good tool for simplifying thing and if you know
the system, please, go forward with it. I could prepare a clean virtual
server for mounting the environment and i will try to help you (and
learn from you ;-)) as much as i can.

Some months ago i was thinking of configuring a cross-compiling
environment for the Zynthian images, but my time is limited and my task
list too long.

I have seen that QEMU is used in CustomPiOS ... i suposse that this
approach is better, as we could automatically test the images before
distribution. Do you think we could have a virtual audio device and a
virtual display (emulated) so we could make a full-featured test?

Of course, let's move to stretch. Although Holger is working with
Pianoteq engine that only works with Jessie, i hope it will be upgraded
very soon, so there is no reason to stay in Jessie.

Kind Regards,

guysoft, 2018-02-06:
Oh. so
1. Yay!
2. CustomPiOS needs the ability to loopback mount. If that exists on the server I'd be glad to set up a build system on it. If Vagrant works its a good bonus to keep the environment identical. 
3. CustomPiOS does emulation and not cross-compile. It makes things reproducible on the Pi, but it does mean compiling takes longer. As a busy person I found I prefer deterministic over fast. As long as you can press a button on Jenkins or nightly build and you get a disto ready to ship in 12 hours I don't care. But its a limitation.
4. I've just added [qemu howto to FAQ](https://github.com/guysoft/CustomPiOS/wiki/FAQ#can-i-run-my-distro-on-qemu-to-simulate) It uses dhruvvyas90 qemu kernel (https://githudf b.com/dhruvvyas90/qemu-rpi-kernel/) which lets you boot an image, I haven't tested for an audio device, but I think it should be there. I haven't gotten the networking working on it, which was bothering me more. It gives a little strange behavior on current zynthian. I had to tweak the ``/etc/fstab``, but it boots :)
[image]

guysoft, 2018-02-06:
Replied on github where its public. Should I open a new issue and document
progress there?

jofemodo, 2018-02-06:
Great @guysoft !

+ This log is from the official SD image (gorgona-next) or it's a fresh compiled image?
+ Emulation is fine. I also prefer the deterministic way ;-)

I think we should open a new thread for this conversation ...

jofemodo, 2018-02-06:
https://github.com/zynthian/zynthian-sys/issues/26

guysoft, 2018-02-06:
The image taken is from zynthian_gorgona_next_rbpi3_pitft28-2017-11-14, moving to new topic