zynthian/zynthian-sys #26: Setup an automated Image Building system with CustomPiOS

jofemodo, 2018-02-06:
Comes from: https://github.com/zynthian/zynthian-sys/issues/25

guysoft, 2018-02-06:
I'll push a repo and link here once a build will finish here.
(Its at the ``Generating harmonizer.wisdom file, this might take a while...`` part now.)

guysoft, 2018-02-06:
Ok, repo we can start talking about is here:
https://github.com/guysoft/ZynthianOS

guysoft, 2018-02-07:
Ok, so it builds. It seems like the image needs a few things to tweak in order to boot.
Also there are a few compile errors (search for Error 2 to find them in the build.log).
I am uploading the image so we can figure out what is missing. [Should be here in about 9 hours](http://unofficialpi.org/Distros/ZynthianOS/)

[build.log](https://github.com/zynthian/zynthian-sys/files/1701238/build.log)

jofemodo, 2018-02-07:
The build hangs up in a git command. It has been completely frozen for hours, so i cancelled and restarted after having fixed a few errors in the setup scripts.

guysoft, 2018-02-08:
Blocked git happends in some distros, using vagrant build method solves it.

guysoft, 2018-02-09:
Ok, so in progress, 
There are several errors now in the build, mostly makes that fail. But there is an image poping up.

Here is the latest log:
http://gnethomelinux.com:8080/job/ZynthianOS/8/parsed_console/

guysoft, 2018-02-10:
Ok, I have an image you can test.
There is a known bug with the host name I fixed and I am building a new image with it fixed, however, it will take another 12+ hours to build and upload so suggest you use the workaround for now.

Thinks that I found that dont work are:

    Zynthian-webconf password
    Mod-ui (see bug).
    The gui boots black - is that expected?
    There are two build errors in the console that you can see

So image (suggest you use a download accelerator, hosted by http://docstech.net/, who hosts my FullPageOS images):
http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-02-09_2017-11-29-realtimepi-stretch-lite-0.2-0.1.zip
md5:
http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-02-09_2017-11-29-realtimepi-stretch-lite-0.2-0.1.zip.md5

Build log:
http://gnethomelinux.com:8080/job/ZynthianOS/8/consoleFull

There is still a few things to fix, but once they are people could actually use this.
Enjoy,
Guy

jofemodo, 2018-02-20:
Hi @guysoft!

This week i hope to have some time to test the image and help you with this.

Thanks a lot!!

jofemodo, 2018-02-20:
Is this the last image you have compiled?

http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-02-14_2017-11-29-zynthianos-stretch-lite-0.1.zip

If not, post the latest one here, or better, in the forum thread.

Thanks

guysoft, 2018-02-20:
Yes,
Note there are a few fixes didn't made it to that image, but it should at least get you the idea. The critical stuff should be working.

guysoft, 2018-02-21:
Status update - all compile errors have solutions. Now its just a matter of implementing them in the scripts (aka seeing one is merged, and the other is in our own repo).

guysoft, 2018-02-23:
Build is green, uploading

guysoft, 2018-03-06:
@jofemodo Should we close this?

jofemodo, 2018-03-07:
OK Let's close it ...

guysoft, 2018-03-09:
wait, we still didn't fix this. I need to merge the realtimepi-test branch for this to work, which I will do now

jofemodo, 2018-03-09:
OK! ;-)
Let's work on it ... MIDI-learning development is finished. I hope to merge with master today and release an updated Gorgona SD image the next week. After that, i would like to "stop" development of new features and focus in the ZynthianOS ...

guysoft, 2018-03-09:
I must adding now switches similar to:
```
if [[ "$ZYNTHAIN_SETUP_APT_CLEAN" = "TRUE" ]]; then # Clean apt cache (if instructed via zynthian_envars.sh)
	apt-get clean
fi
```

Can I use:
$ZYNTHAIN_SETUP_APT_CLEAN" = "yes"?

``yes`` is the convention in CustomPiOS

mheidt, 2018-03-09:
Let us finish the 4 open webconf tasks as well before we freeze.

guysoft, 2018-03-09:
Yay we are all here - can't we start a chat or something?

guysoft, 2018-03-09:
Ok if we merge that it should get it all working. I am running a test build

guysoft, 2018-03-09:
Build looking good:
http://gnethomelinux.com:8080/job/ZynthianOS/30/consoleFull

If its OK I wanna merge CustomPiOS branch to master.

closed 2018-03-13 by jofemodo

guysoft, 2018-03-13:
Ok, Updated ZynthianOS sources too . Yay!

commit 150646ac (2018-03-13): Fixes #26, lets CustomPiOS set variables and skip things it does not need
commit 73ea8aa8 (2018-03-13): Merge pull request #38 from zynthian/CustomPiOS

Fixes #26, lets CustomPiOS set variables and skip things it does not …