zynthian/zynthian-sys #27: #ZYNTHIAN_SW_DIR# does not seem to get replaced in system/mod-ui.service and system/mod-sdk.service

guysoft, 2018-02-06:
Hey,
Not sure how it should look, but mod-ui and mod-sdk dont seem to start on my custom build, error below.
Seems like ``#ZYNTHIAN_SW_DIR#`` is not replaced in ``/etc/systemd/system/mod-ui.service`` and ``/etc/systemd/system/mod-sdk.service``

```
● mod-ui.service - MOD-UI
   Loaded: error (Reason: Invalid argument)
   Active: inactive (dead)

Feb 06 20:57:55 jackpi systemd[1]: [/etc/systemd/system/mod-ui.service:17] Working directory path '#ZYNTHIAN_SW_DIR#/mod-ui' is not absolute, ignoring.
Feb 06 20:57:55 jackpi systemd[1]: [/etc/systemd/system/mod-ui.service:18] Executable path is not absolute, ignoring: #ZYNTHIAN_SW_DIR#/mod-ui/server.py
Feb 06 20:57:55 jackpi systemd[1]: mod-ui.service: Service lacks both ExecStart= and ExecStop= setting. Refusing.
Feb 06 21:13:05 jackpi systemd[1]: [/etc/systemd/system/mod-ui.service:17] Working directory path '#ZYNTHIAN_SW_DIR#/mod-ui' is not absolute, ignoring.
Feb 06 21:13:05 jackpi systemd[1]: [/etc/systemd/system/mod-ui.service:18] Executable path is not absolute, ignoring: #ZYNTHIAN_SW_DIR#/mod-ui/server.py
Feb 06 21:13:05 jackpi systemd[1]: mod-ui.service: Service lacks both ExecStart= and ExecStop= setting. Refusing.
```

jofemodo, 2018-02-07:
I've fixed both setup scripts (minibian jessie & stretch lite). It should work now. I'm trying to build an image for test the fixes.

guysoft, 2018-02-08:
Ok, saw so in https://github.com/zynthian/zynthian-sys/commit/992506cd3416bac472cc542ca23a66fd926e5c4e

I am running a nightly build to test fix.

guysoft, 2018-02-10:
Tested, works now, fixed

closed 2018-02-10 by guysoft