zynthian/zynthian-sys #28: ImportError: cannot import name '_imagingtk' in zynthian.service

guysoft, 2018-02-06:
Hey,
So I am trying to start ``zynthian.service``, but I see it fails with this error:
```
Feb 06 22:43:55 jackpi startx[20337]: Traceback (most recent call last):
Feb 06 22:43:55 jackpi startx[20337]:   File "/usr/local/lib/python3.5/dist-packages/PIL/ImageTk.py", line 176, in paste
Feb 06 22:43:55 jackpi startx[20337]:     tk.call("PyImagingPhoto", self.__photo, block.id)
Feb 06 22:43:55 jackpi startx[20337]: _tkinter.TclError: invalid command name "PyImagingPhoto"
Feb 06 22:43:55 jackpi startx[20337]: During handling of the above exception, another exception occurred:
Feb 06 22:43:55 jackpi startx[20337]: Traceback (most recent call last):
Feb 06 22:43:55 jackpi startx[20337]:   File "./zynthian_gui.py", line 39, in <module>
Feb 06 22:43:55 jackpi startx[20337]:     import zynautoconnect
Feb 06 22:43:55 jackpi startx[20337]:   File "/home/pi/zynthian-ui/zynautoconnect/__init__.py", line 5, in <module>
Feb 06 22:43:55 jackpi startx[20337]:     from zynautoconnect.zynthian_autoconnect import *
Feb 06 22:43:55 jackpi startx[20337]:   File "/home/pi/zynthian-ui/zynautoconnect/zynthian_autoconnect.py", line 34, in <module>
Feb 06 22:43:55 jackpi startx[20337]:     from zyngui import zynthian_gui_config
Feb 06 22:43:55 jackpi startx[20337]:   File "/home/pi/zynthian-ui/zyngui/__init__.py", line 22, in <module>
Feb 06 22:43:55 jackpi startx[20337]:     import zyngui.zynthian_gui_config as zynthian_gui_config
Feb 06 22:43:55 jackpi startx[20337]:   File "/home/pi/zynthian-ui/zyngui/zynthian_gui_config.py", line 332, in <module>
Feb 06 22:43:55 jackpi startx[20337]:     loading_imgs.append(ImageTk.PhotoImage(pil_frame2))
Feb 06 22:43:55 jackpi startx[20337]:   File "/usr/local/lib/python3.5/dist-packages/PIL/ImageTk.py", line 115, in __init__
Feb 06 22:43:55 jackpi startx[20337]:     self.paste(image)
Feb 06 22:43:55 jackpi startx[20337]:   File "/usr/local/lib/python3.5/dist-packages/PIL/ImageTk.py", line 180, in paste
Feb 06 22:43:55 jackpi startx[20337]:     from PIL import _imagingtk
Feb 06 22:43:55 jackpi startx[20337]: ImportError: cannot import name '_imagingtk'
```
What version of PIL is needed?

guysoft, 2018-02-06:
The following solved the issue:

```
sudo pip3 install -I --no-cache-dir Pillow
sudo apt-get remove python-pip python3-pip
cd /tmp
wget https://bootstrap.pypa.io/get-pip.py
sudo python get-pip.py ; sudo python3 get-pip.py
sudo pip3 install requests
```

jofemodo, 2018-02-07:
This package should be installed during mod-ui building process. If it's not installed, then something was wrong with the mod-ui recipe. I'm trying to build an image for test the fixes and see if the problems is solved.

guysoft, 2018-02-07:
Ok found why,
This is the console output in the installation. pip explodes:
```
+ ./setup_mod.sh
Cloning into 'mod-host'...
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/mod-monitor.o src/monitor/monitor-client.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/socket.o src/socket.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/uridmap.o src/uridmap.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/mod-host.o src/mod-host.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/protocol.o src/protocol.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/effects.o src/effects.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/utils.o src/utils.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/completer.o src/completer.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/lv2_evbuf.o src/lv2_evbuf.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/monitor.o src/monitor.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/symap.o src/symap.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/worker.o src/worker.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/sha1/sha1.o src/sha1/sha1.c
cc -I/usr/local/include/lilv-0 -I/usr/local/include/sratom-0 -I/usr/local/include -I/usr/local/include/sord-0 -I/usr/local/include/serd-0  -DHAVE_FFTW335 -DHAVE_NEW_LILV -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -O3 -Wall -Wextra -c -std=gnu99 -fPIC -D_GNU_SOURCE -fvisibility=hidden -o src/rtmempool/rtmempool.o src/rtmempool/rtmempool.c
cc src/mod-monitor.o -Wl,--no-undefined -s -L/usr/local/lib -ljack -llilv-0 -ldl -lsratom-0 -lsord-0 -lserd-0 -lreadline -lpthread -lrt -lm -lfftw3 -lfftw3f -lfftw3_threads -lfftw3f_threads -shared -o mod-monitor.so
[... cut ...]
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'
```

guysoft, 2018-02-10:
I did a quick and dirty fix on my end for now: https://github.com/guysoft/ZynthianOS/commit/71fa1329dbc05286f0c83cbd1d1e776a4ff728b2

jofemodo, 2018-02-13:
OK! By now, the Q&D fix is OK. When the process runs without problems, i will see what should be moved from the "start_chroot_script" to the zynthian setup script.

closed 2018-02-13 by jofemodo