zynthian/zynthian-sys #36: Waveshare32 screen not found

mheidt, 2018-02-25:
waveshare32 doesn’t work.
My screen stays white

more /var/log/Xorg.0.log
[ 3159.292]
X.Org X Server 1.19.2
Release Date: 2017-03-02
[ 3159.292] X Protocol Version 11, Revision 0
[ 3159.292] Build Operating System: Linux 4.9.41-v7+ armv7l Raspbian
[ 3159.292] Current Operating System: Linux zynthian 4.9.59-v7+ #1047 SMP Sun O
ct 29 12:19:23 GMT 2017 armv7l
[ 3159.292] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=656 bcm2708
_fb.fbheight=416 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=
0x40000000 dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext
4 elevator=deadline fsck.repair=yes rootwait
[ 3159.292] Build Date: 18 October 2017 04:55:30PM
[ 3159.292] xorg-server 2:1.19.2-1+rpt1+deb9u2 (https://www.debian.org/support)

[ 3159.292] Current version of pixman: 0.34.0
[ 3159.292] Before reporting problems, check http://wiki.x.org
to make sure that you have the latest version.
[ 3159.292] Markers: (–) probed, () from config file, (==) default setting,
(++) from command line, (!!) notice, (II) informational,
(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[ 3159.292] (==) Log file: “/var/log/Xorg.0.log”, Time: Sun Feb 25 08:57:12 201
8
[ 3159.293] (==) Using config directory: “/etc/X11/xorg.conf.d”
[ 3159.293] (==) Using system config directory “/usr/share/X11/xorg.conf.d”
[ 3159.294] (==) No Layout section. Using the first Screen section.
[ 3159.294] (==) No screen section available. Using defaults.
[ 3159.294] () |–>Screen “Default Screen Section” (0)
[ 3159.294] () | |–>Monitor “”
[ 3159.294] (==) No device specified for screen “Default Screen Section”.
Using the first device section listed.
[ 3159.294] () | |–>Device “Display”
[ 3159.294] (==) No monitor specified for screen “Default Screen Section”.
Using a default monitor configuration.
[ 3159.294] (==) Automatically adding devices
[ 3159.294] (==) Automatically enabling devices
[ 3159.294] (==) Automatically adding GPU devices
[ 3159.294] (==) Max clients allowed: 256, resource mask: 0x1fffff
[ 3159.294] (WW) The directory “/usr/share/fonts/X11/cyrillic” does not exist.
[ 3159.295] Entry deleted from font path.
[ 3159.295] (WW) The directory “/usr/share/fonts/X11/100dpi/” does not exist.
[ 3159.295] Entry deleted from font path.
[ 3159.295] (WW) The directory “/usr/share/fonts/X11/75dpi/” does not exist.
[ 3159.295] Entry deleted from font path.
[ 3159.295] (WW) The directory “/usr/share/fonts/X11/Type1” does not exist.
[ 3159.295] Entry deleted from font path.
[ 3159.295] (WW) The directory “/usr/share/fonts/X11/100dpi” does not exist.
[ 3159.295] Entry deleted from font path.
[ 3159.295] (WW) The directory “/usr/share/fonts/X11/75dpi” does not exist.
[ 3159.295] Entry deleted from font path.
[ 3159.295] (==) FontPath set to:
/usr/share/fonts/X11/misc,
built-ins
[ 3159.295] (==) ModulePath set to “/usr/lib/xorg/modules”
[ 3159.295] (II) The server relies on udev to provide the list of input devices
.
If no devices become available, reconfigure udev or disable AutoAddDevic
es.
[ 3159.295] (II) Loader magic: 0x1fbf40
[ 3159.295] (II) Module ABI versions:
[ 3159.295] X.Org ANSI C Emulation: 0.4
[ 3159.295] X.Org Video Driver: 23.0
[ 3159.295] X.Org XInput driver : 24.1
[ 3159.295] X.Org Server Extension : 10.0
[ 3159.297] (++) using VT number 3

[ 3159.297] (II) systemd-logind: logind integration requires -keeptty and -keep
tty was not provided, disabling logind integration
[ 3159.297] (II) no primary bus or device found
[ 3159.297] (II) LoadModule: “glx”
[ 3159.298] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[ 3159.302] (II) Module glx: vendor=“X.Org Foundation”
[ 3159.302] compiled for 1.19.2, module version = 1.0.0
[ 3159.302] ABI class: X.Org Server Extension, version 10.0
[ 3159.302] (II) LoadModule: “fbdev”
[ 3159.303] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[ 3159.303] (II) Module fbdev: vendor=“X.Org Foundation”
[ 3159.303] compiled for 1.19.0, module version = 0.4.4
[ 3159.303] Module class: X.Org Video Driver
[ 3159.303] ABI class: X.Org Video Driver, version 23.0
[ 3159.303] (II) FBDEV: driver for framebuffer: fbdev
[ 3159.306] (WW) Falling back to old probe method for fbdev
[ 3159.306] (II) Loading sub module “fbdevhw”
[ 3159.306] (II) LoadModule: “fbdevhw”
[ 3159.306] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[ 3159.307] (II) Module fbdevhw: vendor=“X.Org Foundation”
[ 3159.307] compiled for 1.19.2, module version = 0.0.2
[ 3159.307] ABI class: X.Org Video Driver, version 23.0
[ 3159.307] (EE) open /dev/fb1: No such file or directory
[ 3159.307] (EE) No devices detected.
[ 3159.307] (EE)
Fatal server error:
[ 3159.307] (EE) no screens found(EE)
[ 3159.307] (EE)
Please consult the The X.Org Foundation support
at http://wiki.x.org
for help.
[ 3159.307] (EE) Please also check the log file at “/var/log/Xorg.0.log” for ad
ditional information.
[ 3159.307] (EE)
[ 3159.313] (EE) Server terminated with error (1). Closing log file.
pi@zynthian:/zynthian/zynthian-sys/sbin $

*** /etc/X11/xorg.conf.d: directory ***

pi@zynthian:/zynthian/zynthian-sys/sbin $ more /usr/share/X11/xorg.conf.d

*** /usr/share/X11/xorg.conf.d: directory ***

pi@zynthian:/zynthian/zynthian-sys/sbin $

guysoft, 2018-02-26:
@jofemodo Any idea where the Waveshare32 sources are? I can't find them

mheidt, 2018-02-26:
I reckon, you need to get in contact with the ppl. Its not open source

https://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)#Support

guysoft, 2018-02-27:
Ok,
Says:
```
Question:
Why does the touchscreen not work well?

Answer:
--
Now, we only provide touch screen drivers for Raspbian and Ubuntu Mate. How to install the driver? See:#Method_1._Driver_installation
```
And instructions are here:
https://www.waveshare.com/wiki/3.2inch_RPi_LCD_(B)#Method_1._Driver_installation

I guess this means its good we moved to rapsbian :-/ 

There is no LICENSE provided, I am not even sure its legal for us to ship this in a distro.
@jofemodo Do you have anyone in the company we can talk to?

mheidt, 2018-02-27:
I was the one who invited waveshare32 to the party. It's the cheapest screen you can buy in one of the most biggest electronic stores in germany. 
It works with our image, but somehow yours doesn't. Maybe it is configuration only.

guysoft, 2018-02-27:
Looks like in ``zynthian_gorgona_next_rbpi3_pitft28-2017-11-14.img`` there are waveshare dtb and dtbo files:

```
find . -name '*waveshare*'
./boot/overlays/waveshare32b-overlay.dtb
./boot/overlays/waveshare35a-overlay.dtb
./boot/overlays/waveshare35b-overlay.dtb
./boot/overlays/waveshare35cpld-overlay.dtb
./boot/overlays/waveshare32b-overlay.dtbo
./boot/overlays/waveshare35a-overlay.dtbo
./boot/overlays/waveshare35b-overlay.dtbo
./boot/overlays/waveshare35cpld-overlay.dtbo
./zynthian/zynthian-sys/boot/overlays/waveshare35a-overlay.dtbo
./zynthian/zynthian-sys/boot/overlays/waveshare35cpld-overlay.dtbo
./zynthian/zynthian-sys/boot/overlays/waveshare32b-overlay.dtbo
./zynthian/zynthian-sys/boot/overlays/waveshare35b-overlay.dtbo
```

in ``2018-02-23_2017-11-29-zynthianos-stretch-lite-0.1.zip`` image These are the files with that name:
```
$ sudo find . -name '*waveshare*'
./home/pi/zynthian-sys/custom/display/WaveShare 3.2B/boot/overlays/waveshare32b-overlay.dtbo
./home/pi/zynthian-sys/boot/overlays/waveshare32b-overlay.dtbo
./home/pi/zynthian-sys/boot/overlays/waveshare35cpld-overlay.dtbo
./home/pi/zynthian-sys/boot/overlays/waveshare35a-overlay.dtbo
./home/pi/zynthian-sys/boot/overlays/waveshare35b-overlay.dtbo
./home/pi/build/zynthian-sys/custom/display/WaveShare 3.2B/boot/overlays/waveshare32b-overlay.dtbo
./home/pi/build/zynthian-sys/boot/overlays/waveshare32b-overlay.dtbo
./home/pi/build/zynthian-sys/boot/overlays/waveshare35cpld-overlay.dtbo
./home/pi/build/zynthian-sys/boot/overlays/waveshare35a-overlay.dtbo
./home/pi/build/zynthian-sys/boot/overlays/waveshare35b-overlay.dtbo
./boot/overlays/waveshare32b-overlay.dtbo
./boot/overlays/waveshare35a-overlay.dtbo
./boot/overlays/waveshare35b-overlay.dtbo
./boot/overlays/waveshare35cpld-overlay.dtbo
```

Looks like there are files, but they are different.
I can't also determinate if they are meant to be kernel specific.

mheidt, 2018-02-27:
the ones in custom will be copied to boot/overlays, when the user changes the display in the webconf.
fyi
I can do some tests in the next days.

mheidt, 2018-02-28:
First the good news.
When you copy the dtb files into /boot/overlays, the screen works.
@jofemodo 
I wonder, if the dtb should be part of the custom-mechanism as well.

The bad news:
I don't get pianoteq up and running.
Installation and Licence input works.
When I try to open it with a running x11 forwarding and clicking in the zynthian-ui, nothing happens.
When I open it up via ssh, I have this ALSA / JACK issue.
But in contrast to the normal system, I cannot switch to JACK.
It says that the JACK server is not running.
I cannot open qjackctl either. It made my x11 forwarded session stall, so that I had to reboot my Linux Mint machine !?

Audioinjector works by the way. Zynaddsubfx as well.

Nice work, guysoft

guysoft, 2018-03-01:
Hey, 
I suggest opening issues for all of those, if we close them all, one by one, we can release a release candidate ( @jofemodo is that something you want to release?)

mheidt, 2018-03-06:
I opened a new task
https://github.com/zynthian/zynthian-sys/issues/37

This one can be closed, when the dtb files for waveshare are part of the custom-mechanism

jofemodo, 2018-03-07:
Of course, i want to release a "release candidate" ... but currently i'm too busy for getting involved as i would like ... sorry! 
Anyway, i see you do it in the forum:

https://discourse.zynthian.org/t/realtime-kernel-for-raspi/1063/5

that it's fine for me ;-)

Go forward and i will try to join this task as soon as i finish my current tasks ...

Regards,

guysoft, 2018-03-07:
@jofemodo [the discourse talk](https://discourse.zynthian.org/t/realtime-kernel-for-raspi/1063/5) is more of a nightly. A release candidate is when you stop all development, release and make sure this is what you will officially ship asking users to test it and report anything you might have missed.
Those should come out every week/day. So we don't get problems like in https://github.com/zynthian/zynthian-sys/issues/37#issuecomment-371036961 where we don't have a ready image with latest ready to be tested. 

Anyway, keeping this open for you or anyone else to fix, because I can't test it.

mheidt, 2018-03-11:
Added the dtb.  Going to test soon

mheidt, 2018-03-11:
@guysoft 
Should I use the one from 2/23
http://unofficialpi.org/Distros/ZynthianOS/nightly/

or the one in realtime?

guysoft, 2018-03-11:
@mheidt A new build was added this morning, dont use the realtime kernel yet, its in the works.
This is latest:
http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-03-10_2017-11-29-zynthianos-stretch-lite-0.1.zip

http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-03-10_2017-11-29-zynthianos-stretch-lite-0.1.zip.md5

Once we confirm it works on the main kernel, we can test and see what heppends in the realtime one

mheidt, 2018-03-11:
Do you use the latest master branch? Webconf returns with 500, because default.sh midiprofile is missing

guysoft, 2018-03-11:
* Idealy yes, each issue should go in its own issue. But for now we can use this one since its only the two of us.

* There is a build log, you can see exactly what was built:
http://gnethomelinux.com:8080/job/ZynthianOS/33/consoleFull

Says:
```
14:11:27 + git clone https://github.com/zynthian/zynthian-webconf.git
14:11:28 Cloning into 'zynthian-webconf'...
```
So should be master

* if you are getting connectivity issues it might be the access point, and I can revert that. Not that if you connect to the access point there would be no internet.

guysoft, 2018-03-11:
[image]
Strange - why is wpa-spplucant is being created with windows file endings? (visible in vi)
With them my wifi fails to connect, if I remove them it works fine.

The file was created by the "add network" option in zynthian-webconf.

mheidt, 2018-03-11:
I don't have connectivity issues. I am on that machine via ssh, I can ping the github via iPhone but not via dns. It's the network settings. First boot, no changes...Ethernet!

guysoft, 2018-03-11:
@mheidt I dont undertand what you did. Please list step by step.

I can tell you I managed to get zynthian running by:
* Flashing the image
* booting and connecting the the wifi access point
*  as root running ``ln -s /zynthian/zynthian-my-data/midi-profiles  /zynthian/zynthian-data/``
* Rebooting and getting in to the web conf via accesspoint
* Create a wifi network via the UI
* SSH and remove all the windows endings from ``/etc/wpa_supplicant/wpa_supplicant.conf``
* Rebooting
```
pi@zynthian:~ $ ping github.com
PING github.com (192.30.253.113) 56(84) bytes of data.
64 bytes from lb-192-30-253-113-iad.github.com (192.30.253.113): icmp_seq=1 ttl=47 time=189 ms
```

mheidt, 2018-03-11:
I did almost everything you did besides the access point parts.
I am connected via ethernet and don't have the need for that.
In this case the outbound connection doesn't work.

Could not resolve host: github.com.
Maybe we should use metric in /etc/network/interfaces and give eth0 a higher priority?

Regarding ^M .
Maybe a git setting.

guysoft, 2018-03-11:
Ok, for a temp solution do
```
echo "nameserver 8.8.8.8" >>/etc/resolv.conf
```
I am looking in to it now.

mheidt, 2018-03-11:
what is the zynthianos wifi password?

guysoft, 2018-03-11:
"raspberry"
Default settings are here:
https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/auto-hotspot/config

mheidt, 2018-03-12:
Could you build a new "nightly"?
Step no4 shouldn't be possible. If it is, there is still a bug.
But I like to have an image with the correct Pianoteq binary.

guysoft, 2018-03-12:
Starting a nightly now:
http://gnethomelinux.com:8080/job/ZynthianOS/34/console

Should be ready by tomorrow morning (if no error occurs).

mheidt, 2018-03-13:
wow 11hrs already...better call it weekly :)

guysoft, 2018-03-13:
It usually takes 9 hours, but at the moment there is also a kernel compilation going, which means all cores are working at 100%. I really suggest we have a dedicated build server. I maintain 6 distros on one Intel i3 CPU-530. Thats a cpu from nearly 10 years ago. I have so much CPU I can spare for zynthian build.

guysoft, 2018-03-14:
@mheidt new raspberrypi released. New base image:

```
2018-03-13:
  * Raspberry Pi 3 B+ support
  * WiFi is disabled until wireless regulatory domain is set (Pi 3 B+ only)
    - The domain can be done through 'Raspberry Pi Configuration' (rc_gui),
      'raspi-config' or by setting 'country=' to an appropriate ISO 3166
      alpha2 country code in /etc/wpa_supplicant/wpa_supplicant.conf.
  * Default wireless regulatory domain is now unset
  * Added support to desktop for different screen sizes and resolutions,
    including multiple preset options in Appearance Settings and pixel doubling
    option in Raspberry Pi Configuration
  * Version 2.1.16 of Thonny included
  * Version 29.0.0.113 of Adobe PepperFlash player included
  * Version 1.2.post1 of Pygame Zero included
  * Bluetooth plugin now supports connection to Bluetooth LE HID devices
  * Network plugin now indicates 5G-compatible APs
  * Latest changes to Bluez ALSA service merged
    - service now started on CLI boot as well as GUI boot
  * Latest changes to dhcpcd networking plugin merged
  * Improved support for running on pi-top devices
  * Small design changes to PiX theme and icons
  * Bug fix - hide spurious window resize handles
  * Bug fix - Scratch 2 remote GPIO state block now works correctly
  * Updated WiFi Firmware
    - brcmfmac43455-sdio 7.45.154
    - brcmfmac43430-sdio 7.45.98.38
```

guysoft, 2018-03-17:
New image:
http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-03-16_2018-03-13-zynthianos-stretch-lite-0.1.zip
Md5: ``bc0b2dc38368ab1e9b0a016367c86d7e``

Effected by: #39 

Will not have Realtime until this is fixed: https://github.com/guysoft/RealtimePi/issues/9

mheidt, 2018-03-17:
Does it contain a fix for the eth/wlan issue?

I will work on the webconf/hotspot soon

guysoft, 2018-03-17:
Yes now dnsmaq uses resovlconf to get the DNS. Or in human words - internet should work when connected to cable.

closed 2018-03-18 by mheidt