zynthian/zynthian-sys #37: CustomPiOs and Pianoteq (6.0)

mheidt, 2018-03-06:
I don't get pianoteq up and running.
Installation and Licence input works.
When I try to open it with a running x11 forwarding and clicking in the zynthian-ui, nothing happens.
When I open it up via ssh, I have this ALSA / JACK issue.
But in contrast to the normal system, I cannot switch to JACK.
It says that the JACK server is not running.
I cannot open qjackctl either. It made my x11 forwarded session stall, so that I had to reboot my Linux Mint machine !?

guysoft, 2018-03-06:
Can you give me steps to reproduce? Hopefully as a list.
What do I click on the Zynthian GUI?

mheidt, 2018-03-06:
When you go to the webconf, there is a Software/Pianoteq menu entry.
On that page you find the instructions how to install Pianoteq.
After that it should be visible in the zynthian-ui.

I am not sure, how @dcoredump is providing the demo image right now.
Maybe you need to download the 6.0 (not 6.1) demo from the webpage of Pianoteq.

guysoft, 2018-03-06:
Where? I see no "Software" entry. I tried all the help entries one by one and saw none had the word "Pianoteq" in them. Not did all the other ones.
[image]

I found this related thread, it sounds like it was not part of the last release:
https://discourse.zynthian.org/t/pianoteq-is-now-pi3-friendly/1497/21

mheidt, 2018-03-07:
You are not running the latest master branch

guysoft, 2018-03-07:
Which image are you using? And did you run any update commands before?
I want to get the exact same environment you have so I can test it.

mheidt, 2018-03-07:
I used stretch from 2018-02-23_2017-11-29

And ran 
update_zynthian.sh
and
update_zynthian_data.sh

guysoft, 2018-03-11:
[image]
Can you email me or something a license I can test with? (I am not going to be a user do this)

guysoft, 2018-03-12:
Ok, downloaded stage. now it says"
* Start Pianoteq on your zynthian

How do I start it? what is the binary? Where is the Pianoteq stuff?
I can see a new layer available.

mheidt, 2018-03-12:
Layer? You mean engine after mod-ui? That's it. If it shows, the demo is already installed.when nothing happens, that's the bug. The binary is in Zynthian-sw/Pianoteq. In case you want to start it manually. Have a x11forward running

guysoft, 2018-03-12:
But I didn't follow the screen instructions. isn't there some license activation? Perhaps something interactive in the install?

mheidt, 2018-03-12:
I think, you don't even need to upload the demo version since Omega:

https://discourse.zynthian.org/t/software-upgrade-omega-global-midi-learning-pianoteq-demo/1709/4

The licence key is only necessary, if you have a licenced binary.
The demo version should run out of the box in omega. Am i right, @jofemodo ?

Do you have a folder /zynthian/zynthian-sw/pianoteq6 ?

Do you see the engine in zynthian-ui as last item?

mheidt, 2018-03-12:
This might explain your problems 
https://discourse.zynthian.org/t/software-upgrade-omega-global-midi-learning-pianoteq-demo/1709/7

guysoft, 2018-03-12:
> Do you have a folder /zynthian/zynthian-sw/pianoteq6 ?

Yes
```
pi@zynthian:~ $ ls -l /zynthian/zynthian-sw/pianoteq6
total 38792
-rwxr-xr-x 1 root root 39716752 Mar  7 18:06 Pianoteq 6 STAGE
drwxr-xr-x 3 root root     4096 Mar 11 23:33 Pianoteq 6 STAGE.lv2

```
> Do you see the engine in zynthian-ui as last item?

Yes 
[image]

> This might explain your problems
> https://discourse.zynthian.org/t/software-upgrade-omega-global-midi-learning-pianoteq-demo/1709/7

Did you push changes to a git repo? Can you link the commit? (pro tip, if you paste the link to the issue int he commit it references it automatically).

mheidt, 2018-03-12:
I didn't push anything yet. Fernando changed the binary. But wait a little. There is a different issue.
Another pro tip.
I just use # and the number in the commit message and it links automatically.

guysoft, 2018-03-12:
I mean across github repos. This repo I follow :)

I am working on the ZynthianOS realtime variant in the meantime.

So when we release we can release both a realtime and non-realtime kernel

mheidt, 2018-03-18:
In the latest build pianoteq couldn't start jack still. Just fyi.Afaik we didn't do in that area anything anyways.

mheidt, 2018-03-20:
pianoteq works again. But did you @guysoft  try to run the Demo version with your latest build by any chance?
I still believe, that it won't work in CustomPiOS.

guysoft, 2018-03-20:
I don't know how to run it.
Also, my internet is down in my apartment. Technician only would arrive Thursday,  so no builds until then.

mheidt, 2018-03-21:
because it doesn't.
In a fresh installation you should be able to create a Pianoteq layer and should be able to hear something but some notes (lower and higher black keys) are blank.

But as I said, engine doesn't start.

Have a ssh -X pi@zynthian.local session open,
Create a layer with Pianoteq, select Steinway D / Default
Alternatively if you don't want to start it via zynthian ui, you can start the binary in zynthian-sw/pianoteq6

Now you can check, whether the engine needs user input...like entering the licence, when you bought one and installed the binary you got.
But this is not important right  now.
But important is, that in CustomPiOS it probably popping up a dialog telling you to select some audio settings.

And when you select Jack, the logs (that are hopefully running, not the systemctrl jobs) show you that it isn't possible.
This bug is CustomPiOS image specific.

guysoft, 2018-04-09:
@mheidt  any news on this?
@jofemodo  are you doing anything to the image which CustomPiOS is not?

mheidt, 2018-04-09:
The image i tried last week still has the webconf bug. That's why I couldn't test this.

guysoft, 2018-04-09:
Ok. The instarucitons now in the web conf are:

Reboot machine
* Open X11 forwarding session (ssh -X your.zynthian)
* Start Pianoteq on your zynthian
* Pianoteq UI opens in the X11 session
* Follow screen instructions

**Question**: how does Pianoteq know to open in YOUR ssh -X session? 
It might be that this step is the cause, because at the moment ZynthianOS only lets you ssh as user. And not as root. So X sessions which run as root users would not work.

This can be solved by running as root:
```
echo "PermitRootLogin yes" >> /etc/ssh/sshd_config
service ssh restart
```
I can also ship ZynthianOS like this. But would appreciate joint tests here. I opened https://github.com/zynthian/zynthian-ui/issues/32 because I have no way to test without a sound card. Which is on my main zynthian.

jofemodo, 2018-04-09:
No other user than root should exist in ZynthianOS. Current "Gorgona" image work in that way and it's perfectly OK. Of course, ssh as root must be allowed in sshd config.

guysoft, 2018-04-09:
1. @jofemodo Already commited and building. It makes sense to preserve the current behavior.
2. What I gave here is again - a guess, which needs testing. Don't rush to conclusions before you test.
3. I really really suggest keeping at least a single non-root user, even if its not used. There are certain programs her forbid running as root (chromium for example). Also, you already have other users if you run ``cut -d: -f1 /etc/passwd`` you will see a list of existing users.

jofemodo, 2018-04-09:
OK! Of course, no problem with other users. I only try to explain that all the zynthian stuff use root user and all the setup process should do it too. I suggest removing/disabling "pi" user for avoiding missunderstandings.

guysoft, 2018-04-09:
We can put a huge message when you log in, like in OctoPi, that does the job.

We can put that on e separate issue. I really wanna solve this one so we can continue making usable nightlies.

guysoft, 2018-04-10:
Image with the fix (this is the build speed I want): http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-04-09_2018-03-13-zynthianos-stretch-lite-1.0.0.zip
md5: ``6778405ad5ea75c6cc39be8ff36d8af3``

jofemodo, 2018-04-10:
Nice! I'm "burning" a SD with the new image ;-)

guysoft, 2018-04-10:
Note fluidsynth's ``AVL_Drumkits_Perc.lv2`` started failing since yesterday.

jofemodo, 2018-04-10:
Test report of last image:

**Success:**

1.) I can access to webconf and configure correctly my wiring scheme (MCP2317_ENCODERS), display (piscreen 3.5 v2) and fontsize (14).

2.) I've created layers of several engines and it works.

3.) I've installed the Pianoteq binary (6.11 trial) and it works!!! Although i can hear some clicks while playing normally (not hard), so a deeper look is needed, perhaps trying the 6.0.3 version.

**Problems:**

1.) It check the SD disk on every boot, although it's correctly powered off.
2.) I can't login by ssh:

```
[txino@txinoleta .ssh]$ ssh root@zynthian.local
Connection reset by 192.168.1.40 port 22
[txino@txinoleta .ssh]$ ssh pi@zynthian.local
Connection reset by 192.168.1.40 port 22

```
Everything seems to work, although i need to test it more. Anyway, we are really close now!

Thanks a lot, @guysoft !!

jofemodo, 2018-04-10:
Regarding the next build, i've added 2 new recipes and fixed several ones:

+ DISTRHO ports (new)
+ DPF plugins (new)
+ guitarix (fixed to use a stable version)

What about linuxsampler? Should i fix the recipe to use a stable version too? Other candidates to be "fixed" to a stable version?

Regards,

dcoredump, 2018-04-10:
> 3.) I've installed the Pianoteq binary (6.11 trial) and it works!!! Although i can hear some clicks while > playing normally (not hard), so a deeper look is needed, perhaps trying the 6.0.3 version.

In PT6.1.1 for some presets the audio output is too high. Perhaps turning down the volume may help to avoid clicks?

Regards, Holger

jofemodo, 2018-04-10:
Let's close the issue. If there are performance problems it's a different issue ;-)

closed 2018-04-10 by guysoft

guysoft, 2018-04-11:
FYI build faild on ``fatal: unable to access 'https://github.com/zynthian/zynthian-sys.git/': Failed to connect to github.com port 443: No route to host``
By ISP sucks