zynthian/zynthian-sys #39: Linux sampler compilation breaks

guysoft, 2018-03-10:
Error:
```
lscp_shell_reference.cpp:120:33: error: missing terminating " character
lscp_shell_reference.cpp:959:149: warning: missing terminating " character
       "CE INFO <effect-instance>\n\nWhere <effect-instance> is the numeri    "this setting defines the audio channel routing to that effect instance"
                                                                                                                                                     ^
lscp_shell_reference.cpp:959:149: error: missing terminating " character
lscp_shell_reference.cpp:120:2: error: expected '}' before 'uctures'
  uctures, so both applications c"
  ^~~~~~~
lscp_shell_reference.cpp:120:2: error: expected '}' before 'uctures'
lscp_shell_reference.cpp:120:2: error: expected ',' or ';' before 'uctures'
lscp_shell_reference.cpp:131:5: error: expected declaration before '}' token
     },
     ^
Makefile:448: recipe for target 'lscp_shell_reference.lo' failed
make[3]: *** [lscp_shell_reference.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I/usr/local/include/libgig -Wreturn-type -ffast-math -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -pthread -MT lscpevent.lo -MD -MP -MF .deps/lscpevent.Tpo -c lscpevent.cpp -o lscpevent.o >/dev/null 2>&1
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I/usr/local/include/libgig -Wreturn-type -ffast-math -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -pthread -MT lscpserver.lo -MD -MP -MF .deps/lscpserver.Tpo -c lscpserver.cpp -o lscpserver.o >/dev/null 2>&1
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../.. -I/usr/local/include/libgig -Wreturn-type -ffast-math -mcpu=cortex-a7 -mthumb -mfpu=neon-vfpv4 -mfloat-abi=hard -mvectorize-with-neon-quad -pthread -MT lscpparser.lo -MD -MP -MF .deps/lscpparser.Tpo -c lscpparser.cpp -o lscpparser.o >/dev/null 2>&1
make[3]: Leaving directory '/home/pi/zynthian-sw/linuxsampler/src/network'
Makefile:689: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/home/pi/zynthian-sw/linuxsampler/src'
Makefile:509: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/home/pi/zynthian-sw/linuxsampler'
Makefile:414: recipe for target 'all' failed
make: *** [all] Error 2
```

Full log:
http://gnethomelinux.com:8080/job/ZynthianOS/31/parsed_console/

guysoft, 2018-03-10:
Stopped breaking. not sure how it was fixed, i see no commits.

guysoft, 2018-03-15:
Broke again. this is coming and going.

guysoft, 2018-03-18:
This is breaking constantly
@mheidt @jofemodo  any clues? You can test it in the nightly

jofemodo, 2018-03-19:
Perhaps we should think about using "stable" versions and no taking the last from master branch. We could start with the packages giving problems, like LinuxSampler.

guysoft, 2018-03-19:
I would not jump to conclusions.
The first step is to find what is causing the break, it might be a commit or something else. 

Having a main config file with all the versions of the repos is something different and should be considered independently.

guysoft, 2018-03-25:
Bug report in linuxsampler forum:
http://bb.linuxsampler.org/viewtopic.php?f=6&t=3403

jofemodo, 2018-03-26:
Good! Thanks for managing this.
As i told before, perhaps we should consider the option of using "stable" versions for some "problematic" packages ... it shouldn't be difficult to patch the recipes and we could create a "stable versions" file for the recipes.

Regards,

guysoft, 2018-03-26:
I can put things in order, but I have no solution and as I said - we might be missing something, I see no commit in linuxsampler that might be causing this.

guysoft, 2018-04-10:
Its been two weeks and no answer from linuxsampler :-/

guysoft, 2018-05-09:
Do we have a way to call them? Know someone in the project?  Because they ignore bug reports.
We could post in the sourceforge maillist

jofemodo, 2018-06-05:
After some good builds, LS is failing again. I'm changing the building script for using stable versions.

jofemodo, 2018-06-05:
It's done! Awaiting for the next build ... ;-)

dreaming-augustin, 2018-09-22:
For reference, I just hit what appears to be the same bug or a very similar one today.
Full report here:
Bug 666738 - media-sound/linuxsampler-2.0.0-r2: lscp_shell_reference.cpp: error: missing terminating " character
https://bugs.gentoo.org/666738

a17r, 2019-05-25:
The build system is not parallel build safe.

guysoft, 2019-05-26:
For reference the fix in Gentoo was:
```
 src_compile() {
-	default
+	emake -j1 # fails with parallel jobs, bug #666738
 	use doc && emake docs
 }
```
aka not make it parallel.

guysoft, 2019-05-26:
@dreaming-augustin @a17r Thanks for bringing that to us, fix implemented here too and notified back to linuxsampler forum.

dreaming-augustin, 2019-06-09:
Thank you @a17r and @guysoft  for the fix. :)

commit bb1fa9d0 (2018-06-06): Use new LinuxSampler recipe that uses stable versions. Fixes #39
commit f74764d9 (2019-05-26): Fix #39 as mentined in https://bugs.gentoo.org/666738