zynthian/zynthian-sys #40: Bounch of new compile errors (and B+)

guysoft, 2018-03-15:
Hey,
Lots of new compile errors in the new image.
I hadn't tested it it was the new additions to the zynthian repos, or pacahging missing for the new 3B+

Output is here with errors highlight.:
http://gnethomelinux.com:8080/job/ZynthianOS/38/parsed_console/

Already detected:

```
E: Unable to locate package deb-multimedia-keyring
...
gpg: connecting dirmngr at '/tmp/apt-key-gpghome.Cni6VvBXRX/S.dirmngr' failed: No such file or directory
gpg: keyserver receive failed: No dirmngr
```

```
src/ttymidi.c:28:23: fatal error: jack/jack.h: No such file or directory
 #include <jack/jack.h>
```

```
vringbuffer.h:7:29: fatal error: jack/ringbuffer.h: No such file or directory
 #include <jack/ringbuffer.h>
```

```
testsndfile.c:20:21: fatal error: sndfile.h: No such file or directory
 #include <sndfile.h>
```

```
  src/liblo.c:435:19: fatal error: lo/lo.h: No such file or directory
   #include "lo/lo.h"
```
```
  src/RtMidi.cpp:1099:28: fatal error: alsa/asoundlib.h: No such file or directory
   #include <alsa/asoundlib.h>
```

```
/zynthian/zynthian-sw/plugins/eq10q-code/eq6qs.c:29:19: fatal error: fftw3.h: No such file or directory
 #include <fftw3.h>
```

dcoredump, 2018-03-15:
It seems that there are the wrong compiler options chosen. It looks like the options for Raspi2 and not for Raspi3...

guysoft, 2018-03-15:
@dcoredump refresh, I updated the issue

guysoft, 2018-03-15:
The package that causes the line to break is: ``liblrdf-dev``

guysoft, 2018-03-18:
Fixed

closed 2018-03-18 by guysoft