zynthian/zynthian-sys #43: midi-profiles does not exist in /zynthian/zynthian-data/

guysoft, 2018-03-20:
Hey,
The issue mentioned in #36 is still not resolved.
In order for the installation to be functional one must run:

```
ln -s /zynthian/zynthian-my-data/midi-profiles /zynthian/zynthian-data/
```
Otherwise zynthian-webconf would not start.

As @mheidt said in https://github.com/zynthian/zynthian-sys/issues/36#issuecomment-372472733 :
>  Step no4 shouldn't be necessary. If it is, there is still a bug.

Please fix this.

mheidt, 2018-03-20:
Have you checkout out the latest code of webconf, sys and ui?
Especially in sys/scripts/zynthian_envars.sh

ZYNTHIAN_SCRIPT_MIDI_PROFILE

should contain the my-data path.

guysoft, 2018-03-20:
@mheidt I have the code that is provided in the build from [2018-03-16_2018-03-13-zynthianos-stretch-lite-0.1.zip](http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-03-16_2018-03-13-zynthianos-stretch-lite-0.1.zip). If this has been fixed, please LINK THE COMMIT to where it was, so I can compare what I have to latest. This would be the first time I wait 9 hours building and uploading to find that the issue was not fixed.

mheidt, 2018-03-20:
https://github.com/zynthian/zynthian-sys/commit/f430ea2b12f3c15379ba47ab3e00a31db2b4ff7b 

Best practice is to always run a zynthian-update.sh and test again before posting a bug

guysoft, 2018-03-20:
Ok, this was posted 3 days ago, I will make another build and test against it.

mheidt, 2018-03-20:
better wait until omega is stable. Pianoteq has an issue right now.

guysoft, 2018-03-20:
As long as I have CPU time I am gonna build it. Its new a click of a button.

I also wanna release the realtime build so we can test that.

jofemodo, 2018-03-20:
It should be solved from 2-3 days ago ...
Anyway, i wait for confirmation from @guysoft  before closing

guysoft, 2018-03-26:
Still haven't tested it. I was working on the ftp code so the build would be uploaded.
(I have disconnections here and a fixed build has not been uploaded yet. Now I have an FTP upload script that retries and resumes upload).

jofemodo, 2018-04-03:
Hi @guysoft,
Can i close this issue? ;-)

guysoft, 2018-04-03:
No, we haven't tested, and we haven't tested because I can't upload any Zynthian images. See: https://discourse.zynthian.org/t/issues-uploading-images-to-my-hosting-service/1771/1

I know its fun closing issues, but is much more fun closing them after testing and knowing you are not going to repoen.

jofemodo, 2018-04-03:
No problem! A good amount of disk space is being prepared in zynthian server. SFTP is good for you?

guysoft, 2018-04-03:
Yes, or rsync, rsync probably best because you can resume partial uploads.

guysoft, 2018-04-04:
Finally went up:
http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-03-27_2018-03-13-zynthianos-stretch-lite-0.1.zip
http://unofficialpi.org/Distros/ZynthianOS/nightly/2018-03-27_2018-03-13-zynthianos-stretch-lite-0.1.zip.md5

jofemodo, 2018-04-04:
Downloading ;-)
Do you still need the space in the zynthian server?

jofemodo, 2018-04-04:
Don't worry about it. I will prepare a fresh server instance with stretch and give you root access so you can install what you need (no RKs, please ;-)). I really will do it, but you know i'm really busy, so please, don't hesitate to push a little bit ;-)