zynthian/zynthian-sys #44: Version release numbering

guysoft, 2018-03-22:
Hey,
So CustomPiOS tends to assume you have a version number, as specified in: https://github.com/guysoft/ZynthianOS/blob/devel/src/config#L2

What should I put in it for ZynthianOS? It gets appended to the image name and is stored in ``/etc/zynthianos_version``, helps debuging.

jofemodo, 2018-03-26:
I think we should start with version 1.0 and consider previous "Gorgona" releases as 0.x ;-)
So, we are working for ZynthianOS v1.0 ...

guysoft, 2018-03-26:
I am making it V1.0.0 , so we have a minor for bug fixes.

closed 2018-03-26 by guysoft

jofemodo, 2018-03-26:
Perfect. I suposse every "night build" doesn't increase this number. It should be something like ZynthianOS v1.0.0 build-20180326 ... am i right?

guysoft, 2018-03-26:
No, 
The version is part of the build, so what Jenkins does is renames the zip to include the build date, that way you can release a nightly as a release candidate and eventually a release. You don't want to rebuild especially for a release,  defies the point.

jofemodo, 2018-03-26:
Oko. Understand. It seems good to me ;-)