zynthian/zynthian-sys #46: Zynthian instalation overwrites example of how to use wpa zynthianos-wpa-supplicant.txt

guysoft, 2018-04-09:
CustomPiOS provides a detailed example and explanation, warning against using wrong editors (which we got lots of user complaining about).
https://github.com/guysoft/CustomPiOS/blob/CustomPiOS/src/modules/network/filesystem/boot/custompios-wpa-supplicant.txt

However it gets overwritten by:
https://github.com/zynthian/zynthian-sys/blob/master/etc/wpa_supplicant/wpa_supplicant.conf

Which is much more minimal.

jofemodo, 2018-06-06:
The wpa_supplicant file is automatically generated by the webconf tool. The CustomPiOS version could be saved in the same directory, as example and useful information for users wanting to tweak it.

guysoft, 2018-06-06:
@jofemodo no no no.
Please look at wpa_supplicant.conf in the zynthian-sys compared  to the one in CustomPiOS.

The one in zynthian is 3 lines, with no documentation, no way to understand how to set up anything else.

I would consider changing it if you actually give a legitimate explanation.

jofemodo, 2018-06-06:
I've no problem with including all the documentation you want in the auto-generated wpa_supplicant.conf. It's easy to add ;-)

Anyway, in ZynthianOS nobody would need to modify this file by hand because this configuration is done by webconf tool and the changes made by users will be overwritten.

jofemodo, 2018-06-06:
OK! The last is not completely true because the config tool try to conserve the configuration as much as possible, replacing only the needed fields.

jofemodo, 2018-06-06:
Could you point me to the CustomPiOS's version of wpa_supplicant.conf? I don't find it ...

riban-bw, 2024-03-12:
Wifi management from UI has been improved. We don't wan't manual editing of the configs. I will close this issue. Please open a new ticket in the issue tracker if there is still a desire to implement this enhancement.

closed 2024-03-12 by riban-bw