zynthian/zynthian-sys #47: fluidsynth build is failing

guysoft, 2018-04-10:
Getting:

```
Cloning into 'FluidPlug'...
(cd AirFont320.lv2 && \
	wget http://download.linuxaudio.org/musical-instrument-libraries/sf2/airfont_a340u.tar.7z && \
	7z x airfont_a340u.tar.7z && \
	7z x airfont_a340u.tar && \
	mv airfont_a340u/* . && \
	mv airfont_a320u.sf2 FluidPlug.sf2 && \
	mv readme README && \
	rmdir airfont_a340u)
cc source/FluidPlug.c -DFLUIDPLUG_LABEL=\"AirFont320\" -Wall -Wextra -Wshadow -pipe -fPIC -DPIC -DNDEBUG -O2  -fvisibility=hidden -std=gnu99 -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mneon-for-64bits -mfloat-abi=hard -mvectorize-with-neon-quad -I/usr/local/include -fdata-sections -ffunction-sections -Wl,--gc-sections -Wl,-O1 -Wl,--as-needed -Wl,--strip-all  -Wl,--no-undefined -L/usr/local/lib -lfluidsynth -shared -o AirFont320.lv2/FluidPlug.so
--2018-04-09 16:00:35--  http://download.linuxaudio.org/musical-instrument-libraries/sf2/airfont_a340u.tar.7z
Resolving download.linuxaudio.org (download.linuxaudio.org)... (cd AVL_Drumkits_Perc.lv2 && \
	wget http://download.linuxaudio.org/musical-instrument-libraries/sf2/AVL-Drumkits-1.1-SF2-splitted/AVL_Drumkits_Perc_1.1.tar.7z && \
	7z x AVL_Drumkits_Perc_1.1.tar.7z && \
	7z x AVL_Drumkits_Perc_1.1.tar && \
	mv AVL_Drumkits_Perc_1.1/* . && \
	mv AVL_Drumkits_Perc_1.1.sf2 FluidPlug.sf2 && \
	mv 'AVL-Drumkits CC-BY-SA License.pdf' License.pdf && \
	rmdir AVL_Drumkits_Perc_1.1)
185.54.115.200cc source/FluidPlug.c -DFLUIDPLUG_LABEL=\"AVL_Drumkits_Perc\" -Wall -Wextra -Wshadow -pipe -fPIC -DPIC -DNDEBUG -O2  -fvisibility=hidden -std=gnu99 -mcpu=cortex-a53 -mfpu=neon-fp-armv8 -mneon-for-64bits -mfloat-abi=hard -mvectorize-with-neon-quad -I/usr/local/include -fdata-sections -ffunction-sections -Wl,--gc-sections -Wl,-O1 -Wl,--as-needed -Wl,--strip-all  -Wl,--no-undefined -L/usr/local/lib -lfluidsynth -shared -o AVL_Drumkits_Perc.lv2/FluidPlug.so

Connecting to download.linuxaudio.org (download.linuxaudio.org)|185.54.115.200|:80... --2018-04-09 16:00:35--  http://download.linuxaudio.org/musical-instrument-libraries/sf2/AVL-Drumkits-1.1-SF2-splitted/AVL_Drumkits_Perc_1.1.tar.7z
Resolving download.linuxaudio.org (download.linuxaudio.org)... connected.
HTTP request sent, awaiting response... 200 OK
Length: 7799067 (7.4M) [application/x-7z-compressed]
Saving to: 'airfont_a340u.tar.7z'

     0K .......... ...185.54.115.200
Connecting to download.linuxaudio.org (download.linuxaudio.org)|185.54.115.200|:80... ....... .......... .......... ..connected.
HTTP request sent, awaiting response... ........  0%  279K 27s
    50K .......... .......... .......... .......... ........200 OK
..  1%  570K 20s
   100K .Length: 1806647 (1.7M) [application/x-7z-compressed]
Saving to: 'AVL_Drumkits_Perc_1.1.tar.7z'

     0K .......... ............ .......... .......... .......... ..........  1% 2.96M 14s
   150K .......... .......... .......... .......... ..........  2% 19.5M 11s
   200K .......... ........ .......... .......... ........... .......... .......... ..........  3%  500K 11s
   250K .......... ..source/FluidPlug.c: In function 'lv2_instantiate':
source/FluidPlug.c:133:20: error: storage size of 'preset' isn't known
     fluid_preset_t preset;
                    ^~~~~~
source/FluidPlug.c:135:10: error: dereferencing pointer to incomplete type 'fluid_sfont_t {aka struct _fluid_sfont_t}'
     sfont->iteration_start(sfont);
          ^~
........ .......... .......... ..........  3% 2.65M 10s
   300K .......... .......... .......... ....source/FluidPlug.c:133:20: warning: unused variable 'preset' [-Wunused-variable]
     fluid_preset_t preset;
                    ^~~~~~
...... ..........  4% 8.77M 9s
   350K .......... .......... .......... .......... ..........  5% 13.8M 7s
   400K .......... .......... .......... ................  2%  342K 5s
    50K .......... .......... .......... .......... .......... ..........  5%  464K 8s
   450K .......... .......... ....Makefile:666: recipe for target 'AirFont320.lv2/FluidPlug.so' failed
make: *** [AirFont320.lv2/FluidPlug.so] Error 1
make: *** Waiting for unfinished jobs....
...... .......... ..........  6% 3.79M 8s
   500K .......... .......... .......... .......... ..........  7% 3.09M 7s
   550K .......... .......... .......... .......... ..........  7% 2.00M 7s
   600K .......... .......... .......... .......... .........  5%  313K 5s
   100K .......... .......... .......... .......... ..........  8% 11.9M 3s
   150K .......... .......... .......... .......... .......... 11% 19.3M 2s
   200K ........source/FluidPlug.c: In function 'lv2_instantiate':
.. .source/FluidPlug.c:133:20: error: storage size of 'preset' isn't known
     fluid_preset_t preset;
                    ^~~~~~
source/FluidPlug.c:135:10: error: dereferencing pointer to incomplete type 'fluid_sfont_t {aka struct _fluid_sfont_t}'
     sfont->iteration_start(sfont);
          ^~
source/FluidPlug.c:133:20: warning: unused variable 'preset' [-Wunused-variable]
     fluid_preset_t preset;
                    ^~~~~~
......... .......... .......... .......... 14%  402K 3s
   250K .......... .......... .......... .......... .......... 17% 6.09M 2s
   300K .....  8%  337K 8s
   650K .......... .......... .......... .......... ..........  9% 24.1M 7s
   700K .......... .......... .......... .......... ..................  9% ...... 40.1M.. 7s
   750K. . ............ .......... .......... .......... .......... 10% 55.5M 6s
   800K .......... .................. .......... .......... .... .............. 11% 93.4M 6s.. 
   850K.. ................. 19% 6.31M 2s. ......
   350K. ....... ........... ....... ............... .......... 11% 66.0M 5s... .....Makefile:666: recipe for target 'AVL_Drumkits_Perc.lv2/FluidPlug.so' failed
make: *** [AVL_Drumkits_Perc.lv2/FluidPlug.so] Error 1
..... .......... .......... 22%  406K 2s
   400K .......... .......... .......... .......... .......... 25% 5.99M 2s
   450K .......... .......... .......... .......... .......... 28% 14.2M 1s
   500K .......... .......... .......... .......... .......... 31% 14.2M 1s
   550K .......... .......... .......... .......... .......... 34% 9.52M 1s
   600K .......... .......... .......... .......... .......... 36%  782K 1s
   650K .......... .......... .......... .......... .......... 39% 1.01M 1s
   700K .......... .......... .......... .......... .......... 42% 7.95M 1s
   750K .......... .......... .......... .......... .......... 45% 17.9M 1s
   800K .......... .......... .......... .......... .......... 48% 29.6M 1s
   850K .......... .......... .......... .......... .......... 51% 3.84M 1s
   900K .......... .......... .......... .......... .......... 53%  558K 1s
   950K .......... .......... .......... .......... .......... 56% 11.3M 1s
  1000K .......... .......... .......... .......... .......... 59% 7.60M 1s
  1050K .......... .......... .......... .......... .......... 62% 29.4M 1s
  1100K .......... .......... .......... .......... .......... 65% 35.1M 0s
  1150K .......... .......... .......... .......... .......... 68%  521K 0s
  1200K .......... .......... .......... .......... .......... 70% 7.92M 0s
  1250K .......... .......... .......... .......... .......... 73% 7.07M 0s
  1300K .......... .......... .......... .......... .......... 76% 21.1M 0s
  1350K .......... .......... .......... .......... .......... 79% 25.7M 0s
  1400K .......... .......... .......... .......... .......... 82%  528K 0s
  1450K .......... .......... .......... .......... .......... 85% 6.27M 0s
  1500K .......... .......... .........
   900K .......... .......... .......... ........... .... .......... .......... 87% 7.89M 0s
  1550K .......... ............. .......... .......... .......... 90% 30.6M 0s
  1600K .......... .......... .......... .......... .......... 93% 28.8M 0s
  1650K .......... .......... .......... .......... .......... 96%  664K 0s
  1700K .......... .......... .......... .......... .......... 99% 24.2M 0s
  1750K. .......... ....                                       100% 3.65M=1.1s
```

Full log:
http://gnethomelinux.com:808/job/ZynthianOS/51/parsed_console/job/ZynthianOS/51/parsed_console/log_content.html#WARNING5

Rebuilding to confirm.

guysoft, 2018-04-30:
cc @jofemodo  This is the only thing that is breaking now.

jofemodo, 2018-06-01:
I think this is solved now: edf70ab
The next automatic build will confirm ;-)

guysoft, 2018-06-01:
Still failing, I cant see the apt-get install commands happening. Perhaps we need ``set -x`` to see its actually executing them  #50 ).

http://os.zynthian.org:8080/job/ZynthianOS/59/parsed_console/

jofemodo, 2018-06-05:
Build #59 didn't use the latest changes, but fluidplug is compiling without errors from build #60 ;-)
Build #61 finished with 2 warnings only, but LinuxSampler is failing again from build #62 ...
I'm tired of that ... I'm changing the script for using LS stable versions only ;-)

closed 2018-06-05 by jofemodo