zynthian/zynthian-sys #48: Zynthian build gest stuck, looks like something in guitarix

guysoft, 2018-04-12:
```
08:45:38 Writing manifest.ttl... done!
08:45:38 Writing TAL-Vocoder-2.ttl... done!
08:45:38 Writing presets.ttl...
08:45:38 Saving preset 1/11...
08:45:38 Saving preset 2/11...
08:45:38 Saving preset 3/11...
08:45:38 Saving preset 4/11...
08:45:38 Saving preset 5/11...
08:45:38 Saving preset 6/11...
08:45:38 Saving preset 7/11...
08:45:38 Saving preset 8/11...
08:45:38 Saving preset 9/11...
08:45:38 Saving preset 10/11... done!
08:45:38 Generate ttl data for './drowaudio-reverb.so', basename: 'drowaudio-reverb'
08:45:38 Writing manifest.ttl... done!
08:45:38 Writing drowaudio-reverb.ttl... done!
```
Hangs here for ~4 hours.

Note:
```
/home/pi/zynthian-sw/plugins/guitarix-0.36.1/wscript:685: error: the configuration failed (see '/home/pi/zynthian-sw/plugins/guitarix-0.36.1/build/config.log')
```
attached: [config.log](https://github.com/zynthian/zynthian-sys/files/1902352/config.log)

Says at the end:
```
----------------------------------------
Checking for lrdf
pkg-config --errors-to-stdout --print-errors --cflags --libs lrdf
Package lrdf was not found in the pkg-config search path.
Perhaps you should add the directory containing `lrdf.pc'
to the PKG_CONFIG_PATH environment variable
No package 'lrdf' found
not found
```

Full log:

http://gnethomelinux.com:8080/job/ZynthianOS/59/consoleFull

guysoft, 2018-04-12:
Ran again, got stuck exactly at the same place.
Log of second build stuck: http://gnethomelinux.com:8080/job/ZynthianOS/60/console
No builds until this is fixed cc @jofemodo

guysoft, 2018-04-12:
This commit broke the build:
https://github.com/zynthian/zynthian-sys/commit/e57a43cc9b0dc250e3158e897e1457052946efa1

jofemodo, 2018-04-13:
OK! I've added the liblrdf package to setup_system_rbpi_raspbian_lite_stretch.sh. This should solve the problem with DISTRHO.

I will try to give a deeper try to the FluidSynth problem ...

Regards,

guysoft, 2018-04-13:
Rebuilding

guysoft, 2018-04-13:
Its still getting stuck:
http://gnethomelinux.com:8080/job/ZynthianOS/62/console

```
configure: error: *** FFTW3 library not found.
```

jofemodo, 2018-04-14:
Sorry for the pitfall. It takes time to check this changes. Burn a new image (15 min), put in the raspberry, launch the pre-setup procedure, first reboot, launch the install, wait until finish or error, analyze logs ... Your system simplify hugely all this process, and we need it!! ;-)

I'm preparing the VM for the zynthianOS builder infrastructure: Debian Stretch, 2 cores, 2 GB RAM and 100GB FS.

Thanks,

guysoft, 2018-04-14:
Still timing out:
```
05:37:36 Saving preset 10/11... done!
05:37:36 Generate ttl data for './drowaudio-reverb.so', basename: 'drowaudio-reverb'
05:37:36 Writing manifest.ttl... done!
05:37:37 Writing drowaudio-reverb.ttl... done!
11:37:37 Build timed out (after 360 minutes). Marking the build as failed.
```
I am throwing it out on a branch and building without it until you fix it.

guysoft, 2018-04-16:
Its not utilizing any CPU. Something is not right there. 
It might be that it dies on some kind of runtime error. Once we have the build server up we can look at dmesg and figure out. I want to get an env we are both looking at first.

jofemodo, 2018-04-17:
But it'sfrozen in a different point. Now it seems to have finished the DISTHRO building and  it's waiting in the last stage, generating the TTL files.  Again, any error message, only a frozen process ...

Could be a question of memory? In RBPi i have to reduce the number of cores in the "make" for avoiding getting out of memory.

Regards.

guysoft, 2018-04-17:
I have no idea what it is, so I have no way of telling. 
All i know is:

[works]
lots of big commits
[not works]

You know what you changed, you are welcome to try and change it.
I am building against ``CustomPiOS-test``branch at the moement to see if I can get it working, then diff between it and master, then slowly figure out what made it break in master.

jofemodo, 2018-04-17:
We could try to disable DISTRHO building by commenting the line 69 in file setup_plugins_rbpi.sh ...

guysoft, 2018-04-17:
All of this?
```
# dcoredump Stuff
bash $ZYNTHIAN_RECIPE_DIR/install_lvtk.sh
bash $ZYNTHIAN_RECIPE_DIR/install_dxsyx.sh
#bash $ZYNTHIAN_RECIPE_DIR/install_deeaxe7.sh
bash $ZYNTHIAN_RECIPE_DIR/install_dexed_dcoredump.sh
```

I can try if it helps, would be better to know exact line.
Thats why I stared adding ``set -x`` to bash scripts. Now that the output is logged to jenkins its easier to trace back stuff. But you need it to be printed to screen first.

guysoft, 2018-04-24:
Fixed by removing libcurl4-openssl-dev. Does not jam anymore, building from master

jofemodo, 2018-06-06:
It fails in the automated building process but it works perfectly when executed from command line in a RBPi with the latest ZynthianOS image. You can see the error in the Jenkins parsed console output.
It remains the last 2 warnings in the list ;-)

guysoft, 2018-06-06:
Thia bug is when it was hannging. If its building then irs not hanging

jofemodo, 2018-06-07:
Ups! It's true ... i open a new issue ...

closed 2018-06-07 by jofemodo