zynthian/zynthian-sys #49: Raspi3 B+ and WIFI

mheidt, 2018-04-12:
We need this to get WIFI working with the new Raspi3 B+

https://discourse.zynthian.org/t/raspbi3-b-and-wifi/1806/3

@guysoft 
I can check later when my new sd card arrived, if this is a problem for CustomPiOs as well.

mheidt, 2018-04-14:
I am using Zynthian. Holger showed a fix on a non Zynthian platform.

jofemodo, 2018-06-06:
Commit 3e8bd09 install the non-free WIFI firmware from repository. It should solve partially the problem. 
Now we have to see what to do with "/etc/network/interfaces". The problem here is that some people (metal case owners) needs to use a USB wifi dongle, so 2 interfaces are needed (wlan0 and wlan1), but when both interfaces are defined in the "/etc/network/interfaces" file, the wap_supplicant seems to fail for the first one. I'm doing some research for solving it, but any help will be very appreciated here.

jofemodo, 2018-06-07:
From discourse:

```
wpa_supplicant -iwlan0 -c /etc/wpa_supplicant/wpa_supplicant.conf

But the ‘Start WiFi’ option on the gui itself does nothing. Well, it starts up wpa_supplicant in the background, but that doesn’t appear to actually try to do anything useful.
```
https://discourse.zynthian.org/t/new-zynthian-build-from-kit-no-sound-wifi-doesnt-work/1965/6

mheidt, 2018-06-07:
No it doesn't.
I only implemented the UI only as I didn't know how to start it.

jofemodo, 2018-06-08:
I think it's solved with 87df53a ;-)

commit 87df53a5 (2018-06-08): + Install CRDA to allow enabling WIFI channels 12-14.
+ Fix /etc/network/interfaces. Solve problema with wlan0 and wpa_supplicant. Fix #49.
+ Install LV2 versions for synthv1, samplv1 and drumkv1 (stretch).
closed 2018-06-08 by jofemodo