zynthian/zynthian-sys #55: Why are we building against out of date mod-ui?

guysoft, 2018-04-29:
Repo seems to be behind by a few good months:
https://github.com/zynthian/mod-ui/network

jofemodo, 2018-05-08:
We use a customized MOD-UI version. I sync with upstream from time to time and verify that everything is working OK. We should automatize this ...

guysoft, 2018-05-08:
It might be easier to maintain if I know what changes were committed (as in list of commits)

jofemodo, 2018-05-08:
Easy:

https://github.com/zynthian/mod-ui/commits?author=jofemodo

;-)

guysoft, 2018-05-10:
Apart for the midi_map functionality, there is barely anything changed there. 
It might be easier to just maintain a patch and a SHA1 that points to the last build tested against

guysoft, 2018-07-25:
@jofemodo 
Ok, haven't tested but I squished all changes to a single commit against the latest mod-ui/mod-host:

https://github.com/guysoft/mod-ui/commit/ee7b36c6f3482d9be8f0ea6bf2f073ee30d29277

https://github.com/guysoft/mod-host/commit/3c39954f75f22c9bd73dfdf1ea3864362968cbc5

Haven't tested, so if you have anyp ointers what to test would be great.
I guess we could use this branch and push to nightly?

guysoft, 2018-08-13:
Hey, tested, it all works, so I am merging it

guysoft, 2018-08-14:
The branch has been updated from the commits above. See:
https://github.com/guysoft/mod-ui/tree/zynthian-single-commit

and:
https://github.com/guysoft/mod-host/tree/zynthian-single-commit

They both have one commits which is the last branch update. You can also use that to create a patch against current moddevices repos

jofemodo, 2018-08-14:
Regarding mod-host changes ...

These "mod-ui"  changes would be really useful for a pure "mod-host" integration (without mod-ui), but currently we don't need it and don't use it, as zynthian interfaces with mod-ui, no mod-host. Current zynthian images use the official mod-host, not the zynthian's one.

I think these changes are interesting, but i prefer to stick to the official mod-host because MOD's developers didn't seem favorable to include it.

guysoft, 2018-08-14:
@jofemodo They are planning to talk about it next week, see https://github.com/moddevices/mod-host/pull/26

We can also maintain a patch if the decide otherwise. I think your input there would be helpful since its your code.

guysoft, 2018-08-18:
Ok - Can you please give me access or push the branch from ``guysoft/zynthian-single-commit`` to ``zynthian/zynthian-single-commit`` so we can use that in the nightly builds?

jofemodo, 2018-08-25:
Hi @guysoft!

You are invited to both repositories now (mod-ui & mod-host).
Excuse the delay. The last weeks i'm living like a farmer ... a kind of "yearly brain reset" ;-)

I see that mod's team reject to merge the changes. It will be difficult to push changes in its repository because although mod-host & mod-ui are open source software, both are designed for working with the MOD's hardware and they don't want to waste its time merging changes that they don't need and increase the complexity (and i understand their point of view ;-))

You have made a great job by compiling all Zynthian's changes in a single commit. This will ease the task of merging the last version of mod-ui with our repo. Perhaps we could automate the process ...

Of course, new changes to MOD-UI are coming ...  ;-)

Thanks a lot!

guysoft, 2018-08-26:
Fixed via https://github.com/zynthian/zynthian-sys/commit/609162fdcf30d368d5da2f137cb849de2f387a6c

steveb, 2020-07-07:
I've re-applied and cleaned up the patches against the current upstream master of mod-ui. The result works for me. https://github.com/steveb/mod-ui/tree/zyn-mod-merge

I've got a corresponding change to zynthian-ui which detects changes in the mod-ui API (set_midi_devices and pedalpresets)

steveb, 2020-07-07:
If we're going to use this issue to track getting back to latest upstream mod-ui, can it be repoened?

jofemodo, 2020-07-08:
Hi @steveb !
What do you think is the best way of getting your updated branch in the zynthian's mod-ui fork?

jofemodo, 2020-07-08:
Regarding the last commit on zynthian's fork, that you didn't apply:

https://github.com/zynthian/mod-ui/commit/4d8e7e351da12847b86f94bf43ffaf5ea13c77e5

It's a little bit "ugly", but it simply tries to avoid failing to load a pedalboard because some of the included plugins has a different version and some control port changed. Instead of "pass", a warning message would be better ;-)

Regards,

steveb, 2020-07-08:
I didn't apply that last patch because it looks like the current mod-ui code might avoid the errors which were being raised when there is no `symbol` key in the `ports` dict:
https://github.com/moddevices/mod-ui/blob/master/mod/host.py#L3062-L3070

steveb, 2020-07-08:
I just tried loading Crazy Kalimba and it loaded without any errors being logged

jofemodo, 2020-07-08:
OK!! The fewer custom code, the better.

steveb, 2020-07-09:
We can always add something back if there are issues

jofemodo, 2020-07-10:
OK! What do you think is the best way of getting your updated branch into the zynthian's mod-ui fork?

steveb, 2020-07-11:
Maybe leave master as it is and create a new branch which the install script can checkout. I can keep an eye on upstream master and keep the zynthian branch up-to-date

jofemodo, 2020-09-02:
But this branch can't be forked from the current master. It should be forked at some point on the past, so i can merge your updated branch on top of it, true?

jofemodo, 2020-09-02:
Until the repository question is solved, i have changed the zynthian nightly builds for including the branch from your repo. Please, could pull the latest changes from MOD-UI? ;-)

Also, regarding the requirements.txt ... did you installed all the required packages? Specially these 2:

+ Pillow==6.2.2
+ tornado==4.3

Zynthian webconf use older versions (5.4 & 4.1), and we could have issues with this. Perhaps it's time to use a python virtualenv, as recommended.

Regards,

steveb, 2020-09-02:
Yes, I was thinking the master branch would be left to rot, and the install script would switch to whichever branch it needs to.

Master its only the default branch by convention (and we should maybe consider switching away from master due to its historic connotations, but thats unrelated to this ;)

I'll pull in the latest upstream changes and test zyn-mod-merge branch before pushing

steveb, 2020-09-02:
Oh I see, could you create a zyn-mod-merge branch in https://github.com/zynthian/mod-ui based off my zyn-mod-merge branch?

Then the install script can be pointed back to https://github.com/zynthian/mod-ui zyn-mod-merge and I can manage changes by approving PRs from https://github.com/steveb/mod-ui

jofemodo, 2020-09-02:
>Oh I see, could you create a zyn-mod-merge branch in https://github.com/zynthian/mod-ui based off my zyn-mod-merge branch?
I will try to do this ...

jofemodo, 2021-01-26:
I think this can be closed, right? ;-)

closed 2021-01-26 by jofemodo