zynthian/zynthian-sys #56: Stretch Building Script: ZynAddSubFX building is broken

jofemodo, 2018-05-31:
ZynADdSubFX building script gives this error:

```
CMake Error at rtosc/cmake/ColorMessage.cmake:53 (_message):
  The imported target "fluid" references the file

     "/usr/bin/fluid"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/fltk/FLTK-Targets.cmake"

  but not all the files it references.

  
Call Stack (most recent call first):
  /usr/lib/fltk/FLTK-Targets.cmake:99 (message)
  /usr/lib/fltk/FLTKConfig.cmake:21 (include)
  /usr/share/cmake-3.7/Modules/FindFLTK.cmake:136 (include)
  rtosc/CMakeLists.txt:92 (find_package)

-- Configuring incomplete, errors occurred!
See also "/zynthian/zynthian-sw/zynaddsubfx/build/CMakeFiles/CMakeOutput.log".

```
But Jenkins console parser didn't seem to detect it.

jofemodo, 2018-05-31:
Install "fluid" package solved the problem ;-)

closed 2018-05-31 by jofemodo

guysoft, 2018-05-31:
Yay! Lets see how nightly goes (also I think the server needs to be restarted)

jofemodo, 2018-05-31:
Restarted no ... cleaned up!! The disk was completely full. I had to delete by hand a few images from "/home/guysoft/builds".  Is there some easy way of limiting the number of builds saved from Jenkins or i should write e little cron script?

jofemodo, 2018-05-31:
Some idea of why the Jenkins console parser didn't account/detect this error?

guysoft, 2018-05-31:
The builds are copied there. We can delete it and leave thr last x buida by cron or Jenkins.

jofemodo, 2018-05-31:
BTW ... did you read my email? ;-)

jofemodo, 2018-05-31:
Also ... the Jenkin's workspace is empty. No zip files to download ...
I restarted the server after cleaning the files ... but nothing ;-(

guysoft, 2018-05-31:
I Don understand what youbare expecting. Tyebjob copies thr result to the build folder, not the workspace.

jofemodo, 2018-06-01:
Normally the Jenkins workspace have a link to the last image successfully built.
This morning have it again ;-)

jofemodo, 2018-06-01:
BTW, Did you read my encrypted email? I don't remember the password for the Jenkins admin user ...

guysoft, 2018-06-01:
I couldn't decrypt it with my key.
I can send you the password

commit 2e8a089b (2018-05-31): Fixes #56: Install "fluid" package for fixing ZynAddSubFX building.