zynthian/zynthian-sys #57: ZynthianOS images: SSH access is failing

jofemodo, 2018-06-06:
```
ssh root@zynthian.local
Connection reset by 192.168.1.38 port 22

``` 
I've tried to reboot the system several times but it still failing. Same with "pi" user.

jofemodo, 2018-06-06:
Regenerating keys from webconf tool solved the problem, but it should work out the box.
I will try again from a fresh-burned image.

jofemodo, 2018-06-06:
I see wlan0 is configured as a WIFI access point. Super Niceeeeeeeeeeee!! It's a great improvement!

Anyway, it should be allowed to disable from webconf, for "normal" WIFI setup. Could you point me to the right direction? What services are involved in that? etc.

Thanks!

jofemodo, 2018-06-25:
I'm not alone:

https://discourse.zynthian.org/t/zynthianos-the-upcoming-zynthian-sd-image/1957/17

guysoft, 2018-07-14:
Hey,
Same here, however running "regenerate ssh keys" in the gui solved the issue.
So its probably related to that.

guysoft, 2018-10-25:
Yws, look at the base module. Hss a setting

jofemodo, 2018-10-25:
I see this:

```
# if set will resize root partition on image after build to minimum size + 
# provided size in MB
export BASE_IMAGE_RESIZEROOT=200
`````

OK! This is done "after build", but is the system "rebooted" after finishing the build, so the "/etc/rc.local" is executed? It's from there that zynthian's first boot script is called and the line is deleted, so the next boot wont call the script...

jofemodo, 2018-10-29:
Hi @guysoft !
This is still failing. The last build image is very close to OK. It work out-the-box, and most of engines works perfectly. Only this error (SSH key generation on first boot) and the unstable MOD-UI engine remains as "important" problems.

guysoft, 2018-10-29:
The base module has a script that enables ssh:
https://github.com/guysoft/CustomPiOS/blob/devel/src/modules/base/config#L44

```
# Enable ssh daemon
[ -n "$BASE_SSH_ENABLE" ] || BASE_SSH_ENABLE=yes
```

This script was tested many times with OctoPi, so it should work.
However I recall there was a talk about enabling ssh also for root. So I suggest you test both user and root.

This code handles that:
https://github.com/guysoft/ZynthianOS/blob/devel/src/modules/zynthianos/start_chroot_script#L54

```
# Make root password raspberry
echo root:raspberry | chpasswd
echo "PermitRootLogin yes" >> /etc/ssh/sshd_config
sed -i 's/PermitRootLogin without-password//g' /etc/ssh/sshd_config
```

It might be that new versions you added stuff that clashes with those lines.

jofemodo, 2018-10-29:
But this is not the problem @guysoft !
SSH is enabled and root login allowed. The problem is with key generation and it's solved when the keys are regenerated.

guysoft, 2018-10-29:
Ok, another direction then.
This could be related to https://github.com/guysoft/OctoPi/issues/424
This issue was solved locally in OctoPi with:
https://github.com/guysoft/OctoPi/blob/devel/src/modules/octopi/start_chroot_script#L190

```
### Try and fix https://github.com/guysoft/OctoPi/issues/424

sed -i "s@ExecStart=/usr/bin/ssh-keygen -A -v@ExecStart=/bin/bash -c ' /usr/bin/ssh-keygen -A -v >> /var/log/regenerate_ssh_host_keys.log 2>\&1'@g" /lib/systemd/system/regenerate_ssh_host_keys.service
sed -i "s@ExecStartPost=/bin/systemctl disable regenerate_ssh_host_keys@ExecStartPost=/bin/bash -c 'for i in /etc/ssh/ssh_host_*_key*; do actualsize=\$(wc -c <\"\$i\") ;if [ \$actualsize -eq 0 ]; then echo size is 0 bytes ; exit 1 ; fi ; done ; /bin/systemctl disable regenerate_ssh_host_keys'@g" /lib/systemd/system/regenerate_ssh_host_keys.service
```

If this solves the problem for ZynthianOS too, I will merge it in to CustomPiOS

jofemodo, 2018-10-30:
Hi @guysoft !

ZynthianOS already have something similar:

```
https://github.com/zynthian/zynthian-sys/blob/75eb232e9e8d639292d4131a0df7f61e9115080c/scripts/setup_system_rbpi_raspbian_lite_stretch.sh#L241

https://github.com/zynthian/zynthian-sys/blob/master/scripts/set_first_boot.sh

https://github.com/zynthian/zynthian-sys/blob/master/scripts/first_boot.sh

```
But it's not working and i don't understand why. In the jessie SD image it works like a charm.
Something is failing on the first boot and the "first_boot.sh" script is not finishing its tasks.

guysoft, 2018-10-30:
The script I linked uses logs. I suggest you add logs or use what I provided,  so far no report came, and you can check its logs if it does not work.

jofemodo, 2019-05-07:
A "first_boot" service has been configured.

jofemodo, 2019-05-07:
The regeneration seems to block awaiting user interaction. I'm trying to fix it on 63aacbe

jofemodo, 2019-05-10:
Solved! ;-)

commit 62f72e51 (2018-10-25): Fix #57:Generate SSH keys before deleting the "first_boot.sh" line from /etc/rc.local.
commit d2475809 (2019-05-21): Fix #57: Call ssh-keygen with the -N "" option to avoid answering for a passphrase.
closed 2019-05-10 by jofemodo