zynthian/zynthian-sys #59: DISTHRO ports building fails

jofemodo, 2018-06-07:
It fails in the automated building process but it works perfectly when executed from command line in a RBPi with the latest ZynthianOS image. You can see the error in the Jenkins parsed console output.
It remains the last 2 warnings in the list ;-)

```
lv2_ttl_generator: pthread_mutex_lock.c:81: __pthread_mutex_lock: Assertion `mutex->__data.__owner == 0' failed.
qemu: uncaught target signal 6 (Aborted) - core dumped
./scripts/generate-ttl.sh: line 27: 16524 Aborted                 $GEN ./$FILE
Makefile:62: recipe for target 'gen_lv2' failed
make[1]: *** [gen_lv2] Error 134
make[1]: Leaving directory '/home/pi/zynthian-sw/plugins/DISTRHO-Ports'
Makefile:104: recipe for target 'lv2' failed
make: *** [lv2] Error 2
```

guysoft, 2018-06-07:
Ok, it seems to be a bug in qemu and implementation of vfork:
https://bugs.launchpad.net/ubuntu/+source/qemu/+bug/1737312

Will try find more

guysoft, 2018-06-07:
Can you paste the build instructions?  I want to reproduce in the chroot

guysoft, 2018-06-10:
Ok I found [a fix was released in qemu 2.10](https://bugs.launchpad.net/qemu/+bug/1701971) , and we are using qemu 2.8.

But I can't apt-get update/apt-get install because you changed the root password on the machine.

guysoft, 2018-06-10:
I added sid repo so i can install qemu-user-static 2.12. So we will see how it goes tonight.

jofemodo, 2018-06-11:
Same result ... seems like still using a buggy qemu version ;-(

guysoft, 2018-06-11:
I can try a build from master

guysoft, 2018-06-11:
Ok, there is now qemu built from master on the server now:
```
qemu-arm-static --version
qemu-arm version 2.12.50 (v2.12.0-1182-ga7a7309ca5-dirty)
Copyright (c) 2003-2017 Fabrice Bellard and the QEMU Project developers
```

jofemodo, 2018-06-12:
Not solved yet. Same error on DISTHRO. I will try to write a simple recreation script for reporting the issue ...

guysoft, 2018-06-12:
Also reported to qemu: https://bugs.launchpad.net/qemu/+bug/1776478

guysoft, 2018-06-30:
Did anyone push anything? It started hanging again.

jofemodo, 2018-06-30:
AFAIK, no changes in Zynthian's side.
If it's hanging, it means a qemu's bug?

Regards,

guysoft, 2018-07-03:
Hey,
Hey, managed to reproduce.
The hanging is happening in the new image of Raspbian released last week.

If you run the reproduction above with ``2018-06-27-raspbian-stretch-lite.zip`` the build hangs at:
```
Writing drowaudio-flanger.ttl... done!
Generate ttl data for './drowaudio-tremolo.so', basename: 'drowaudio-tremolo'
```

guysoft, 2018-07-03:
Suggestion:
We could perhaps make it that first boot generates the so ttls first time.
It's a workaround until we know what to do, because it seems ``lv2_ttl_generator`` is causing havoc on qemu.

guysoft, 2018-07-03:
Workaround to disable the generation of lv2_ttl_generator:
```
sed -i 's@\t\@./scripts/generate-ttl.sh@\t@g' Makefile
```

I am pushing this to master just so we can build. We can figure out with qemu what is going on/per-build this outside the chroot.

jofemodo, 2018-07-03:
Yes, generating the TTLs on the first boot could do the trick. Let me take a look to the makefile to see what is the best way of doing it ... probably installation of plugins has no sense until TTL are generated.

Thanks a lot!

guysoft, 2018-07-03:
The ttl files might also be platform-independent, which means we can generate them outside the build and copy them. Its a little ugly, but it might do the trick and put us back on the automation track.
Meanwhile you can follow the qemu bug. I got as far as running gdb on qemu, but not sure how to go from there.

guysoft, 2018-07-31:
@hammerandtongs 
There is a bug in either qemu or DISTRHO, which makes the build crash qemu or in some case and makes it hang. Please read the thread.
The only workaround at the moment is to make it build after its flashed to the Pi, or build it externally to ZynthianOS build and copy it.
I am talking to qemu people about it, but its really really hard to understand what is going on.

hammerandtongs, 2018-07-31:
Yes the bugs you've opened are very interesting.

I'm working my way through figuring out how zynthian installs etc.

I'm not finding a qemu setup script for a build environment?

I was confused I think looking at the output of jenkins and expecting DISTHRO somewhere but I guess that wouldn't make sense as you are generating offline and then copying.

Is it most straightforward for me to just build obxd on the zynthian?

Btw just for reference these ttl files are https://en.wikipedia.org/wiki/Turtle_(syntax) they are plaintext readable and pretty clear.

Yes they look to be platform independent.

They don't seem to have any build specific info that I see. Just higher level descriptive metadata that can be copied like you are doing with the hack.

jofemodo, 2019-04-29:
Using a repo with pre-built plugins until TTL generation is solved with qemu.
Building in a true RBPi works OK.

jofemodo, 2019-09-19:
Generation of TTLs on first boot is not working. I've disabled the build again and back to install from pre-built binaries. I've included Helm in the pre-built repo as it's failing to generate TTLs too.

jofemodo, 2021-01-26:
Since we are now using KXstudio repos, we can close this.

closed 2021-01-26 by jofemodo

guysoft, 2021-05-02:
FYI QEMU asked if this is still the case, I said it was:
https://bugs.launchpad.net/qemu/+bug/1776478

pguyot, 2021-05-03:
As mentioned on launchpad, using taskset -c 0 could be a workaround.

https://github.com/pguyot/pynab/commit/91011e68e446c69e317fd1198c58f85ff0cd5fb1

commit 1c60bfaa (2019-09-09): Fix #59: Generate TTLs for some presets that fail to generate on emulated environment (ZynthianOS).
commit 485eba4a (2019-09-09): Fix #59: Move TTL generation code to a recipe script.
commit d345fdb2 (2019-09-09): Fix #59: Re-enable DISTHRO plugins build.
commit 21d457e9 (2019-09-09): Fix #59: Add "./" to DLL path.