zynthian/zynthian-sys #60: Pianoteq not showing up on GUI

edwardjarchibald, 2018-06-20:
I tried, a number of times, to use the web UI on my Zynthian 2 to upload a licensed copy of Pianoteq. A number of times, after the progress bar showed that, apparently, something was being copied 100%, I got a dialog box that said: 

192.168.2.148 says

Error. Please, contact the webmaster!

Finally, the upload appeared to work without the error. Then I rebooted the Zynthian and where there was previously the menu item for the 'Pianoteq Stage - Demo' there was nada.  I logged into the Zynthian via ssh and saw that the previously populated 'pianoteq6' directory was now there but files were missing.

Next, I tried copying a correctly populated pianoteq6 directory into place - I created it from the files that I got for my licensed copy of Zynthian and used the binaries for 'arm. I did the copy via scp, into /zynthian/zynthian-sw.  I named the directory the same as the previous pianoteq6 directory, did a proper power-off on the Zynthian and restarted.  nada - no menu item shows up.  

I can start the pianoteq binary manually by ssh into the host, and did so to activate it via a proxied X-windows session but what I can't figure out is how to get the menu entry to show up on the Zynthian built-in GUI.  Pianoteq does show up in the web UI.  Seems to me that there must be some metadata stored somewhere that tells the built-in GUI what to load from what appears in a given directory etc. Any tips would be most appreciated.

I'm running on a Zynthian v2 that was delivered 'pre-built'.  

This is what I'm running as far as software:

zyncoder: master (3e47ce0) 
zynthian-ui: master (bae73f4) 
zynthian-sys: master (1e87b96) 
zynthian-data: master (80a6459) 
zynthian-webconf: master (5b70cee)

mheidt, 2018-06-20:
I just created a branch that shows more messages.
Besides that. Start the webconf manually, so that you see the logging messages.
Do you know how to do that?
Ssh, stopping service, run script manually.

mheidt, 2018-06-20:
Check the latest webconf error.
@jofemodo knows that it currently doesn't work.

edwardjarchibald, 2018-06-21:
Hi,
Thanks for the tip! That's just what I needed. 
The issue that I found is that the code that installs Pianoteq assumes that the version is 'Pianoteq Stage' and a bunch of stuff is hard-coded in the python code to that effect starting with the extraction from the .7z file. I have 'Pianoteq 6 Pro' so the qualified extraction from the .7z failed. To get it working for myself, quickly, I just edited my own .py files to remove the references to STAGE as needed. 
I'll document what I did more fully later - a little pressed for time right now - but unless someone else prefers to alter the code to make it work for the more general case i.e. that it could install any version of Pianoteq, I could take a pass at it myself.  I think I know where all of the bodies are buried at this point so I don't think it will be too hard. Please let me know if this makes sense.
Regards,
Edward

jofemodo, 2018-06-21:
OK! Take into account that your changes will be overwriten when you update your zynthian software.

+ You could try to "repackage" your pro-package  so it seems like a stage-package, changing the names of binaries and everything ...

+ You also could try to use the stage binary with your "pro" license key. I can send you the package by email if you can't get it ... i don't know if it will work, but it's easy to test ;-)

Regards,

mheidt, 2018-06-21:
But would't it be best, if we had a generic solution for both?
In the end it might be  a regexp somewhere.

edwardjarchibald, 2018-06-22:
This is my proposal. I suspect that a solution something like the following would work:
- generate a listing of the .7z file via '7z l <archive>' prior to extraction to determine what version of Pianoteq needs to be installed.- set up appropriate variables in the python code based on the above.- voila!
I'll work this out sometime this weekend and post the solution in a branch.  I'm looking forward to this as a first step to learning more about how all things Zynthian work.
Cheers,
Edward

jofemodo, 2018-06-25:
Hi @edwardjarchibald!

Could you share your modified code for installing the "pro" version?

Thanks!

jofemodo, 2018-06-25:
This subject should be discussed in the zynthian-webconf repo, so i've created a new topic there:

https://github.com/zynthian/zynthian-webconf/issues/39

for forwarding the discussion.

closed 2018-06-25 by jofemodo