zynthian/zynthian-sys #61: helm build started breaking

guysoft, 2018-07-08:
Cloning into 'helm'...
/usr/bin/make -C builds/linux/LV2 CONFIG=Release DEBCXXFLAGS="-O2 -fstack-protector-strong -Wformat -Werror=format-security -mcpu=cortex-a7 -mthumb -Ofast -mfpu=neon-vfpv4 -mfloat-abi=hard -mlittle-endian -munaligned-access" DEBLDFLAGS="-Wl,-z,relro" SIMDFLAGS="-mcpu=cortex-a7 -mthumb -Ofast -mfpu=neon-vfpv4 -mfloat-abi=hard -mlittle-endian -munaligned-access -mfpu=neon-fp-armv8 -mfloat-abi=hard"
make[1]: Entering directory '/home/pi/zynthian-sw/plugins/helm/builds/linux/LV2'
/usr/bin/make -f Makefile.binary CONFIG=Release DEBCXXFLAGS="-O2 -fstack-protector-strong -Wformat -Werror=format-security -mcpu=cortex-a7 -mthumb -Ofast -mfpu=neon-vfpv4 -mfloat-abi=hard -mlittle-endian -munaligned-access" DEBLDFLAGS="-Wl,-z,relro"
/usr/bin/make -f Makefile.ttl_generator CONFIG=Release
make[2]: Entering directory '/home/pi/zynthian-sw/plugins/helm/builds/linux/LV2'
g++ lv2_ttl_generator.c -o lv2_ttl_generator -ldl
make[2]: Entering directory '/home/pi/zynthian-sw/plugins/helm/builds/linux/LV2'
make[2]: *** No rule to make target '../../../JUCE/modules/juce_audio_plugin_client/LV2/juce_LV2_Wrapper.cpp', needed by 'build/intermediate/Release/juce_LV2_Wrapper_fb90cd9c.o'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/pi/zynthian-sw/plugins/helm/builds/linux/LV2'
Makefile:8: recipe for target 'binary' failed
make[1]: *** [binary] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/pi/zynthian-sw/plugins/helm/builds/linux/LV2'
make[1]: Leaving directory '/home/pi/zynthian-sw/plugins/helm/builds/linux/LV2'
Makefile:82: recipe for target 'lv2' failed
make: *** [lv2] Error 2

guysoft, 2018-07-08:
Totally their fault, they pushed 4 days ago something that broke the build, even says so on their CI

[image]

jofemodo, 2018-07-08:
Curiously, the problem seems to be related with ttl_generator ...

guysoft, 2018-07-08:
No, ttl_generator built fine, its after it. Also see talks in commit, it looks like all the lv2 plugin part was removed

mtytel, 2018-07-08:
Yeah I accidentally pushed a newer version of JUCE without the LV2 files.
Working on a fix

mtytel, 2018-07-08:
Ok it's working again.

guysoft, 2018-07-08:
Great, nightly should build and we will see it tomorrow morning :)

guysoft, 2018-07-10:
Fixed by https://github.com/mtytel/helm/commit/abdedd527e6e1cf86636f0f1e8a3e75b06ed166a . Yay!

closed 2018-07-10 by guysoft