zynthian/zynthian-sys #62: ImportError: No module named 'psutil'

guysoft, 2018-07-14:
Hey,
Ran latest nightly, (``/tmp/2018-07-14_2018-06-27-zynthianos-stretch-lite-1.0.0.zip``) And zynthian did not start because psutil is missing.
```
● zynthian.service - Zynthian User Interface
   Loaded: loaded (/etc/systemd/system/zynthian.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-07-14 15:05:59 UTC; 38s ago
 Main PID: 430 (startx)
   CGroup: /system.slice/zynthian.service
           ├─430 /bin/sh /usr/bin/startx ./zynthian.sh -- :0 vt3
           ├─496 xinit ./zynthian.sh -- /usr/bin/X :0 vt3 -auth /tmp/serverauth.IANQWfSLHr
           ├─497 /usr/lib/xorg/Xorg :0 vt3 -auth /tmp/serverauth.IANQWfSLHr
           └─529 /bin/bash ./zynthian.sh

Jul 14 15:06:37 zynthian startx[430]:     import zynautoconnect
Jul 14 15:06:37 zynthian startx[430]:   File "/home/pi/zynthian-ui/zynautoconnect/__init__.py", line 5, in <module>
Jul 14 15:06:37 zynthian startx[430]:     from zynautoconnect.zynthian_autoconnect import *
Jul 14 15:06:37 zynthian startx[430]:   File "/home/pi/zynthian-ui/zynautoconnect/zynthian_autoconnect.py", line 34, in <
Jul 14 15:06:37 zynthian startx[430]:     from zyngui import zynthian_gui_config
Jul 14 15:06:37 zynthian startx[430]:   File "/home/pi/zynthian-ui/zyngui/__init__.py", line 26, in <module>
Jul 14 15:06:37 zynthian startx[430]:     from zyngui.zynthian_gui_admin import zynthian_gui_admin
Jul 14 15:06:37 zynthian startx[430]:   File "/home/pi/zynthian-ui/zyngui/zynthian_gui_admin.py", line 31, in <module>
Jul 14 15:06:37 zynthian startx[430]:     import psutil
Jul 14 15:06:37 zynthian startx[430]: ImportError: No module named 'psutil'
```

Workaround is to run:
```
sudo pip3 install psutil
```

mheidt, 2018-07-19:
I am trying to follow your path, but I can't get the image on disk successfully.
The rootfs has a broken zynthian symbolic link.
How do you get the img to a card?
I use sudo dd if=... of=....

mheidt, 2018-07-20:
i might struggle here with hardware issues. Need more investigations.
But you could tell me, what the destination of the symbolic link is.
I like to change the wiring before I boot the device.

guysoft, 2018-07-20:
What nightly build are you using?
You should flash the imagw with diskimiger or with dd command

mheidt, 2018-07-20:
I tried 14th and 19th.
But I try to copy my working image first to that SD card I am using for CusomPiOs.
Maybe the card or sth else is the problem.

But nevertheless...where is the Zynthian directory? You seem to use a symbolic link

guysoft, 2018-07-20:
I have no idea what would work in the way you arr mixing builds.

Its in ``/home/pi/zynthian``

``ls -l /zynthian`` will show you where a symbolic link is pointing

mheidt, 2018-07-20:
When I open the image in Mint, the link shows as broken. I haven't checked in shell yet. But the link should work there as well.
But as I said, I have to check the sanity of the SD card first.
A build label file in root directory which shows the label of the daily build would be nice.

mheidt, 2018-07-20:
What do you mean with mixing builds?

guysoft, 2018-07-20:
1. The link will show up as broken, because a symbolic link holds the absolute path to where it should go. Your Mint installation does not have a path ``/zynthian`` so it shows up as broken.
2. This image is intended to be dumped on an SD and booted on a Pi. That's it. Beyond that the image should maintain itself, perhaps have a mechanism to import/export presets and settings. You are copying it on to another system I have no idea how it works, how to test and even what architecture. So I can't predict, reproduce or support what you are doing. Perhaps @jofemodo might, or decide its not supported.
3. Regardless its off-topic of this issue and should be discussed in a separate one.

commit cf32c459 (2018-07-21): Fixes #62
closed 2018-07-21 by guysoft