zynthian/zynthian-sys #63: mod-ui will not start, crashes on KeyError: 'DEL1_BPMHOST'

guysoft, 2018-07-14:
Hey, on latest nightly (``2018-07-14_2018-06-27-zynthianos-stretch-lite-1.0.0.zip``) Mod ui crashes when a layer is added with this error:

```
● mod-ui.service - MOD-UI
   Loaded: loaded (/etc/systemd/system/mod-ui.service; disabled; vendor preset: enabled)
   Active: active (running) since Sat 2018-07-14 15:08:18 UTC; 5min ago
 Main PID: 1288 (mod-ui)
   CGroup: /system.slice/mod-ui.service
           └─1288 python3 /zynthian/zynthian-sw/mod-ui/server.py

Jul 14 15:13:24 zynthian server.py[1288]:     result = func(*args, **kwargs)
Jul 14 15:13:24 zynthian server.py[1288]:   File "/home/pi/zynthian-sw/mod-ui/mod/webserver.py", line 1098, in post
Jul 14 15:13:24 zynthian server.py[1288]:     name = SESSION.load_pedalboard(bundlepath, isDefault)
Jul 14 15:13:24 zynthian server.py[1288]:   File "/home/pi/zynthian-sw/mod-ui/mod/session.py", line 297, in load_pedalboa
Jul 14 15:13:24 zynthian server.py[1288]:     title = self.host.load(bundlepath, isDefault)
Jul 14 15:13:24 zynthian server.py[1288]:   File "/home/pi/zynthian-sw/mod-ui/mod/host.py", line 2014, in load
Jul 14 15:13:24 zynthian server.py[1288]:     self.load_pb_plugins(pb['plugins'], instances, rinstances)
Jul 14 15:13:24 zynthian server.py[1288]:   File "/home/pi/zynthian-sw/mod-ui/mod/host.py", line 2169, in load_pb_plugins
Jul 14 15:13:24 zynthian server.py[1288]:     if pluginData['ports'][symbol] != value:
Jul 14 15:13:24 zynthian server.py[1288]: KeyError: 'DEL1_BPMHOST'
```

guysoft, 2018-07-21:
Seems to work after a restart, not sure why.

jofemodo, 2019-04-29:
It's working now ...

closed 2019-04-29 by jofemodo