zynthian/zynthian-sys #65: ZynthianOS: ERROR screen on official v2 kit

mheidt, 2018-07-20:
I downloaded the latest image.
****
After burning if from a different machine (Windows 10), I got it working. My sd card reader or Linux Mint must have been my problem (side note for @guysoft )
****
I connected via Wifi Hotspot.
Used webconf, changed wiring and display and regenerated the keys.

I get the red Error white screen.

I can't send you logs, because the following fails
pi@zynthian:/zynthian/zynthian-sys/sbin $ ./zynthian.sh
./zynthian.sh: line 6: /scripts/zynthian_envars.sh: No such file or directory
xauth: file /home/pi/.Xauthority does not exist

How do you run zynthian-ui outside of the service?

guysoft, 2018-07-22:
Why is this assigned to me? I have no idea how to fix this and how its related to what I am doing here.

jofemodo, 2018-10-29:
I think this is solved. The last build images are perfectly usable out-the-box ;-)

closed 2018-10-29 by jofemodo