zynthian/zynthian-sys #68: Issues with certain plugins not loading.

guysoft, 2018-08-13:
When starting, mod-ui gives these errors in ```systemctl status mod-ui``:

```
Aug 13 10:15:24 zynthian server.py[1760]: lilv_world_load_bundle(): error: Error reading file:///zynthian/zynthian-plugins/lv2/TheFunction.lv2/manifest.ttl
Aug 13 10:15:24 zynthian server.py[1760]: error: failed to open file /zynthian/zynthian-plugins/lv2/drowaudio-distortionshaper.lv2/manifest.ttl (No such file or directory)
Aug 13 10:15:24 zynthian server.py[1760]: lilv_world_load_file(): error: Error loading file `file:///zynthian/zynthian-plugins/lv2/drowaudio-distortionshaper.lv2/manifest.ttl'
Aug 13 10:15:24 zynthian server.py[1760]: lilv_world_load_bundle(): error: Error reading file:///zynthian/zynthian-plugins/lv2/drowaudio-distortionshaper.lv2/manifest.ttl
Aug 13 10:15:24 zynthian server.py[1760]: error: failed to open file /zynthian/zynthian-plugins/lv2/Vex.lv2/manifest.ttl (No such file or directory)
Aug 13 10:15:24 zynthian server.py[1760]: lilv_world_load_file(): error: Error loading file `file:///zynthian/zynthian-plugins/lv2/Vex.lv2/manifest.ttl'
Aug 13 10:15:24 zynthian server.py[1760]: lilv_world_load_bundle(): error: Error reading file:///zynthian/zynthian-plugins/lv2/Vex.lv2/manifest.ttl
Aug 13 10:16:03 zynthian server.py[1760]: error: /zynthian/zynthian-plugins/lv2/gx_mbcompressor.lv2/modgui.ttl:139:12: missing ';' or '.'
Aug 13 10:16:03 zynthian server.py[1760]: error: /zynthian/zynthian-plugins/lv2/gx_mbcompressor.lv2/modgui.ttl:139:12: expected `]', not `l'
Aug 13 10:16:03 zynthian server.py[1760]: lilv_world_load_file(): error: Error loading file `file:///zynthian
/zynthian-plugins/lv2/gx_mbcompressor.lv2/modgui.ttl'
```

jofemodo, 2018-10-01:
Probably this error is due to the disabling of ttl-generator in DISTHRO compiling. Remember this line?

`sed -i 's@\t\@./scripts/generate-ttl.sh@\t@g' Makefile`

guysoft, 2018-10-01:
We might want to write that line differently. At least I have a reproduction steps at: https://github.com/zynthian/zynthian-sys/issues/59#issuecomment-396739645

You are welcome to play with that.

jofemodo, 2018-10-01:
No. We need to generate the TTLs. If we can't do it on build time because of the qemu problem, we should do it on the first boot. Don't worry about it. I will solve it ;-)

jofemodo, 2019-04-29:
DISTHRO TTL generation is broken when building with qemu, so i have created a github repo with the pre-built DISTHRO plugins. This repo is used for installing the DISTHRO plugins on ZynthianOS builds. When the TTL problem is solved, i will re-enable the DISTHRO building ...

closed 2019-04-29 by jofemodo