zynthian/zynthian-sys #70: infamousPlugins breaks buiuld

guysoft, 2018-09-27:
```
-- Build files have been written to: /zynthian/zynthian-sw/plugins/infamousPlugins/build
Scanning dependencies of target infamous-rule
[  1%] Generating casynth_ui.h
[  2%] Building C object CMakeFiles/infamous-rule.dir/src/rule.c.o
Scanning dependencies of target envfollower
Scanning dependencies of target casynth
[  4%] Building C object src/casynth/CMakeFiles/casynth.dir/casynth.c.o
[  5%] Building C object src/envfollower/CMakeFiles/envfollower.dir/envfollower.c.o
[  6%] Generating casynth_ui.cxx
In file included from /zynthian/zynthian-sw/plugins/infamousPlugins/src/casynth/casynth.c:3:0:
/zynthian/zynthian-sw/plugins/infamousPlugins/src/casynth/./casynth.h:4:16: fatal error: lv2.h: No such file or directory
 #include<lv2.h>
                ^
compilation terminated.
src/casynth/CMakeFiles/casynth.dir/build.make:62: recipe for target 'src/casynth/CMakeFiles/casynth.dir/casynth.c.o' failed
make[2]: *** [src/casynth/CMakeFiles/casynth.dir/casynth.c.o] Error 1
CMakeFiles/Makefile2:156: recipe for target 'src/casynth/CMakeFiles/casynth.dir/all' failed
make[1]: *** [src/casynth/CMakeFiles/casynth.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/zynthian/zynthian-sw/plugins/infamousPlugins/src/envfollower/envfollower.c:9:16: fatal error: lv2.h: No such file or directory
 #include<lv2.h>
                ^
compilation terminated.
src/envfollower/CMakeFiles/envfollower.dir/build.make:62: recipe for target 'src/envfollower/CMakeFiles/envfollower.dir/envfollower.c.o' failed
make[2]: *** [src/envfollower/CMakeFiles/envfollower.dir/envfollower.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[  8%] Building C object src/envfollower/CMakeFiles/envfollower.dir/rms_calc.c.o
Scanning dependencies of target casynth_ui
[  9%] Linking C executable infamous-rule
CMakeFiles/Makefile2:252: recipe for target 'src/envfollower/CMakeFiles/envfollower.dir/all' failed
make[1]: *** [src/envfollower/CMakeFiles/envfollower.dir/all] Error 2
[ 12%] Building CXX object src/casynth/CMakeFiles/casynth_ui.dir/casynth_ui_main.cxx.o
[ 12%] Building CXX object src/casynth/CMakeFiles/casynth_ui.dir/casynth_ui.cxx.o
[ 12%] Built target infamous-rule
In file included from /zynthian/zynthian-sw/plugins/infamousPlugins/build/src/casynth/casynth_ui.h:11:0,
                 from /zynthian/zynthian-sw/plugins/infamousPlugins/src/casynth/casynth_ui_main.cxx:8:
/zynthian/zynthian-sw/plugins/infamousPlugins/src/casynth/./casynth.h:4:16: fatal error: lv2.h: No such file or directory
 #include<lv2.h>
                ^
In file included from /zynthian/zynthian-sw/plugins/infamousPlugins/build/src/casynth/casynth_ui.h:11:0,
                 from /zynthian/zynthian-sw/plugins/infamousPlugins/build/src/casynth/casynth_ui.cxx:3:
/zynthian/zynthian-sw/plugins/infamousPlugins/src/casynth/./casynth.h:4:16: fatal error: lv2.h: No such file or directory
 #include<lv2.h>
                ^
compilation terminated.
compilation terminated.
src/casynth/CMakeFiles/casynth_ui.dir/build.make:72: recipe for target 'src/casynth/CMakeFiles/casynth_ui.dir/casynth_ui_main.cxx.o' failed
make[2]: *** [src/casynth/CMakeFiles/casynth_ui.dir/casynth_ui_main.cxx.o] Error 1
make[2]: *** Waiting for unfinished jobs....
src/casynth/CMakeFiles/casynth_ui.dir/build.make:96: recipe for target 'src/casynth/CMakeFiles/casynth_ui.dir/casynth_ui.cxx.o' failed
make[2]: *** [src/casynth/CMakeFiles/casynth_ui.dir/casynth_ui.cxx.o] Error 1
CMakeFiles/Makefile2:193: recipe for target 'src/casynth/CMakeFiles/casynth_ui.dir/all' failed
make[1]: *** [src/casynth/CMakeFiles/casynth_ui.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
[  2%] Built target infamous-rule
[  4%] Building C object src/casynth/CMakeFiles/casynth.dir/casynth.c.o
In file included from /zynthian/zynthian-sw/plugins/infamousPlugins/src/casynth/casynth.c:3:0:
/zynthian/zynthian-sw/plugins/infamousPlugins/src/casynth/./casynth.h:4:16: fatal error: lv2.h: No such file or directory
 #include<lv2.h>
                ^
compilation terminated.
src/casynth/CMakeFiles/casynth.dir/build.make:62: recipe for target 'src/casynth/CMakeFiles/casynth.dir/casynth.c.o' failed
make[2]: *** [src/casynth/CMakeFiles/casynth.dir/casynth.c.o] Error 1
CMakeFiles/Makefile2:156: recipe for target 'src/casynth/CMakeFiles/casynth.dir/all' failed
make[1]: *** [src/casynth/CMakeFiles/casynth.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
/bin/mv: cannot stat '/zynthian/zynthian-plugins/bin/infamous-rule': No such file or directory
/bin/cat: /sys/firmware/devicetree/base/model: No such file or directory
```

jofemodo, 2018-09-28:
The problem is in LV2 library. It seems to have moved the header to a new location. I've created a symlink. Let's see if it solves the problem.

Anyway, is this OK at the end of the process?

```
/home/guysoft/ZynthianOS
++ head -n 1
++ ls './src/workspace/*.zip'
ls: cannot access './src/workspace/*.zip': No such file or directory
+ ZIPNAME=
+++ ls './src/workspace/*.zip'
+++ head -n 1
ls: cannot access './src/workspace/*.zip': No such file or directory
++ basename
basename: missing operand
Try 'basename --help' for more information.
+ BASENAME=
+ DIRNAME=./src/workspace
++ date --rfc-3339=date
+ NEW_ZIP_NAME=./src/workspace/2018-09-28_
+ mv ./src/workspace/2018-09-28_
mv: missing destination file operand after './src/workspace/2018-09-28_'
Try 'mv --help' for more information.
++ basename ./src/workspace/2018-09-28_
+ mv ./src/workspace/2018-09-28_ /var/lib/jenkins/workspace/ZynthianOS/2018-09-28_
mv: cannot stat './src/workspace/2018-09-28_': No such file or directory
+ pushd /var/lib/jenkins/workspace/ZynthianOS
~/workspace/ZynthianOS /home/guysoft/ZynthianOS ~/workspace/ZynthianOS
++ basename ./src/workspace/2018-09-28_
+ md5sum 2018-09-28_
++ basename ./src/workspace/2018-09-28_
md5sum: 2018-09-28_: No such file or directory
+ popd
/home/guysoft/ZynthianOS ~/workspace/ZynthianOS
++ basename ./src/workspace/2018-09-28_
+ cp -v /var/lib/jenkins/workspace/ZynthianOS/2018-09-28_ /home/guysoft/builds
cp: cannot stat '/var/lib/jenkins/workspace/ZynthianOS/2018-09-28_': No such file or directory
++ basename ./src/workspace/2018-09-28_
+ cp -v /var/lib/jenkins/workspace/ZynthianOS/2018-09-28_.md5 /home/guysoft/builds
'/var/lib/jenkins/workspace/ZynthianOS/2018-09-28_.md5' -> '/home/guysoft/builds/2018-09-28_.md5'
+ popd
~/workspace/ZynthianOS

```

guysoft, 2018-09-29:
Its still failing, can't see why hecauon mobile

Do you want it to email you in failure oon something? So you know when a
commit breaks rhe build.

jofemodo, 2018-10-01:
The remaining error seems to be unrelated with any package or compilation. It seems to be a problem when the image file creation process. It says something like this:

```
/home/guysoft/ZynthianOS
++ head -n 1
++ ls './src/workspace/*.zip'
ls: cannot access './src/workspace/*.zip': No such file or directory
+ ZIPNAME=
+++ ls './src/workspace/*.zip'
+++ head -n 1
ls: cannot access './src/workspace/*.zip': No such file or directory
++ basename
basename: missing operand
Try 'basename --help' for more information.
+ BASENAME=
+ DIRNAME=./src/workspace
++ date --rfc-3339=date
+ NEW_ZIP_NAME=./src/workspace/2018-10-01_
+ mv ./src/workspace/2018-10-01_
mv: missing destination file operand after './src/workspace/2018-10-01_'
Try 'mv --help' for more information.
++ basename ./src/workspace/2018-10-01_
+ mv ./src/workspace/2018-10-01_ /var/lib/jenkins/workspace/ZynthianOS/2018-10-01_
mv: cannot stat './src/workspace/2018-10-01_': No such file or directory
+ pushd /var/lib/jenkins/workspace/ZynthianOS
~/workspace/ZynthianOS /home/guysoft/ZynthianOS ~/workspace/ZynthianOS
++ basename ./src/workspace/2018-10-01_
+ md5sum 2018-10-01_
++ basename ./src/workspace/2018-10-01_
md5sum: 2018-10-01_: No such file or directory
+ popd
/home/guysoft/ZynthianOS ~/workspace/ZynthianOS
++ basename ./src/workspace/2018-10-01_
+ cp -v /var/lib/jenkins/workspace/ZynthianOS/2018-10-01_ /home/guysoft/builds
cp: cannot stat '/var/lib/jenkins/workspace/ZynthianOS/2018-10-01_': No such file or directory
++ basename ./src/workspace/2018-10-01_
+ cp -v /var/lib/jenkins/workspace/ZynthianOS/2018-10-01_.md5 /home/guysoft/builds
'/var/lib/jenkins/workspace/ZynthianOS/2018-10-01_.md5' -> '/home/guysoft/builds/2018-10-01_.md5'
+ popd
~/workspace/ZynthianOS

```

guysoft, 2018-10-01:
The error is a few lines back:
```
+++ echo 'Unmounting /home/guysoft/ZynthianOS/src/workspace/mount/dev...'
Unmounting /home/guysoft/ZynthianOS/src/workspace/mount/dev...
+++ sudo umount /home/guysoft/ZynthianOS/src/workspace/mount/dev
umount: /home/guysoft/ZynthianOS/src/workspace/mount/dev: target is busy
        (In some cases useful info about processes that
         use the device is found by lsof(8) or fuser(1).)
+ exit 1
```

Something is using ``/dev`` and is still running.

guysoft, 2018-10-01:
Suggestion: You really need to set in jenkins that each build that fails you get an email, then you would have catches the breaking commit right away. Now we have to go over two weeks at least to figure out what went wrong. Nightly build are not worth much if you don't look at them.

jofemodo, 2018-10-01:
My email is added now ;-)

guysoft, 2018-10-01:
Ok, the gpg issue is unrelated to this issue, so I opened #72 . This issue likely can be closed.

I guess another nightly build advantage is I can run kdif3 and grep and compare two builds from 13 days ago and track the issue.

jofemodo, 2018-10-01:
Problems related with the changes in LV2 library package are solved, so i close this issue.

closed 2018-10-01 by jofemodo