zynthian/zynthian-sys #71: Failed to install pexpect

rodamaralgeo, 2018-09-29:
Hi Guys!

After updating my Zynthian, the startup displays an error message.
When I tried to restart zynthian in debug mode, I noticed that pexpect was not installed.
When I try to install, I get the following error:
[image]

mheidt, 2018-09-29:
Are you sure, that the update didn't break your WiFi? Connected via ethernet?
Looks like a internet connectivity issue to me

closed 2018-09-29 by rodamaralgeo

jofemodo, 2018-10-01:
Zynthian uses "htpdate" for time sync. It's loaded on every boot and remains as a daemon, so it shouldn't be needed to fix the time or do anything related.