zynthian/zynthian-sys #72: gpg service stays running at end of build

guysoft, 2018-10-01:
New break reason from #70 

comparing buid [173](http://os.zynthian.org:8080/job/ZynthianOS/173/parsed_console/) and [172](http://os.zynthian.org:8080/job/ZynthianOS/173/parsed_console/) which started failing I see:

```$ grep gpg green
Executing: /tmp/apt-key-gpghome.5Z9Tqwg8Rg/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 5C808C2B65558117
gpg: failed to start the dirmngr '/usr/bin/dirmngr': No such file or directory
gpg: connecting dirmngr at '/tmp/apt-key-gpghome.5Z9Tqwg8Rg/S.dirmngr' failed: No such file or directory
gpg: keyserver receive failed: No dirmngr
```

And the failed one:

```
$ grep gpg red 
Executing: /tmp/apt-key-gpghome.D058wYsbQw/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 5C808C2B65558117
gpg: failed to start the dirmngr '/usr/bin/dirmngr': No such file or directory
gpg: connecting dirmngr at '/tmp/apt-key-gpghome.D058wYsbQw/S.dirmngr' failed: No such file or directory
gpg: keyserver receive failed: No dirmngr
gpg: directory '/root/.gnupg' created
gpg: keybox '/root/.gnupg/pubring.kbx' created
gpg: /root/.gnupg/trustdb.gpg: trustdb created
gpg: key 49B4C67C05277AAA: public key "Thomas Nagy <tnagy@waf.io>" imported
gpg: Total number processed: 1
gpg:               imported: 1
gpg: assuming signed data in 'waf-2.0.11.tar.bz2'
gpg: Signature made Sun Sep  2 20:36:05 2018 UTC
gpg:                using RSA key 8AF22DE5A06822E3474F3C7049B4C67C05277AAA
gpg: Good signature from "Thomas Nagy <tnagy@waf.io>" [unknown]
gpg: WARNING: This key is not certified with a trusted signature!
gpg:          There is no indication that the signature belongs to the owner.
```

Then the gpg systemd service stays on and breaks the unmount of ``/dev``

guysoft, 2018-10-01:
This is the commit that broke the build:
https://github.com/aubio/aubio/commit/e561ca7706d0c94b34da218f440ac06bbc139d79
indeed, 14 days ago aubio added a script that verifies gpg .

jofemodo, 2018-10-01:
We could disable gpg by hand at the end of aubio install script ...

guysoft, 2018-10-01:
Go for it. Strange that the gpg systemd service goes up after running gpg in the first place.

jofemodo, 2018-10-01:
I just ran the aubio install script in a Raspberry Pi and no gpg daemon was running after finished.

jofemodo, 2018-10-02:
Failed again. Can we add this line before trying to umount the "dev":

`fuser -m -k /home/guysoft/ZynthianOS/src/workspace/mount/dev`

it should solve the problem ;-)

jofemodo, 2018-10-02:
I see in the CustmPiOS "nightly_build_scripts/custompios_nightly_build" script:

`for i in `lsof "${DIST_PATH}/workspace${WORKSPACE_POSTFIX}/mount" | awk '{print $2}'`; do kill -9 $i; done || true
`
what should do the trick, but it's not working ...

Curiosly, when a tried the "fuser" command from the terminal, it worked.

Regards,

guysoft, 2018-10-02:
what ``fuser`` command did you run?

jofemodo, 2018-10-02:
I told you 2 comments above:

`fuser -m -k /home/guysoft/ZynthianOS/src/workspace/mount/dev`

jofemodo, 2018-10-02:
I tried to launch a build by hand, but it didn't work. See:

http://os.zynthian.org:8080/job/ZynthianOS/188/parsed_console/

It didn't start the build ...

guysoft, 2018-10-09:
@jofemodo Should have worked, try again.
The unmount commands are to make sure nothing is mounted before CustomPiOS starts.

guysoft, 2018-10-09:
I am restarting the server, looks like something got stuck. 
Ideally we would have used vagrant or to be docker container and restart that before each build.

Sometimes ``/dev`` gets stuck. Its doing this now because of the gpg service running.

guysoft, 2018-10-09:
Started another build

jofemodo, 2018-10-09:
No problem with rebooting the server before each build. It's a virtual server and the reboot is quite fast.

guysoft, 2018-10-09:
1. You don't want to reboot the server each build, its a server people download files from and that would break the connection.
1a., rebooting to solve something like that is a bad way to do things, it means you can't run long processes on the server.

2. You don't need to reboot it, its just that the build did something strange with ``/dev`` and that got the mount point stuck. You just fixed it and now single reboot should be enough.

guysoft, 2018-10-10:
Still getting
```
umount: /home/guysoft/ZynthianOS/src/workspace/mount/dev: target is busy
```

I'll see if I can debug this

guysoft, 2018-10-10:
Ok,
Think I know why its happening.

I had to run ``kill -9`` to kill it, not just a regular term singal.

Workaround is to run this:

```
for i in $(ps aux | grep gpg | grep -v grep | awk '{ print $2 }'); do kill -9 $i; done
```
Might need to update CustomPiOS to run a KILL with ``-9``

guysoft, 2018-10-10:
Running a build to test:
http://os.zynthian.org:8080/job/ZynthianOS/201/

guysoft, 2018-10-10:
Dang, can't run ps to locate the PID of gpg. Gives

```
13:17:47 Error: /proc must be mounted
13:17:47   To mount /proc at boot you need an /etc/fstab line like:
13:17:47       proc   /proc   proc    defaults
13:17:47   In the meantime, run "mount proc /proc -t proc"
```

jofemodo, 2018-10-10:
I just tried "ps" and everything works normally.
Are you running the command in the normal environment or in a chrooted one?

```
txino@zynthianOS:~$ ps
  PID TTY          TIME CMD
11723 pts/0    00:00:00 bash
11736 pts/0    00:00:00 ps
txino@zynthianOS:~$ mount
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
udev on /dev type devtmpfs (rw,nosuid,relatime,size=990788k,nr_inodes=247697,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,noexec,relatime,size=204216k,mode=755)

```

guysoft, 2018-10-10:
yes, its in the chroot, I really suggest you create an enviroment locally
so you can test thing within it. Otherwise you can't test things like this.

guysoft, 2018-10-10:
It happens because /proc is not mounted. Mounting it is a security issue
because it could effect the host.

guysoft, 2018-10-12:
Build succeeded, last commit fixed it.

commit dd1af33f (2018-10-10): Fix #72
closed 2018-10-12 by guysoft