zynthian/zynthian-sys #74: DPF-Plugins build breaks

guysoft, 2018-10-15:
```
# ProM (needs OpenGL + ProjectM)
```

And later:
```
cp -r bin/ProM             /usr/local/bin/
/bin/cp: cannot stat 'bin/ProM': No such file or directory
Makefile:114: recipe for target 'install' failed
```

Source:
https://github.com/DISTRHO/DPF-Plugins/blob/master/Makefile#L48

Full log:
http://os.zynthian.org:8080/job/ZynthianOS/208/parsed_console/job/ZynthianOS/208/parsed_console/log_content.html#WARNING1

jofemodo, 2018-10-18:
It should work fine now ;-)

closed 2018-10-18 by jofemodo

guysoft, 2019-02-27:
Yay, issue also fixed at: https://github.com/DISTRHO/DPF-Plugins/issues/10

jofemodo, 2019-02-27:
Nice! ;-)
We are really close ... latest images are working quite well and i hope to release a new official SD image based in ZynthianOS in the next weeks.

commit c926d717 (2018-10-18): Fix #74: Avoid errors while installing binaries.