zynthian/zynthian-sys #75: Bunch of builds fail on lo/lo.h and fftw3.h: No such file or directory a

guysoft, 2018-10-19:
lo/lo.h: No such file or directory

error: fftw3.h: No such file or directory
 #include <fftw3.h>

http://os.zynthian.org:8080/job/ZynthianOS/213/parsed_console/

guysoft, 2018-10-22:
Looks better, saw also

```
../src/cocoa_in_qt5.mm:17:24: fatal error: Cocoa/Cocoa.h: No such file or directory
```

guysoft, 2018-10-22:
Unrelated but:
```
./setup_plugins_rbpi.sh: line 47: /zynthian/zynthian-sys/scripts/recipes/install-calf.sh: No such file or directory
```

guysoft, 2018-10-22:
Also
```
21:02:08 /usr/bin/touch: missing file operand
21:02:08 Try '/usr/bin/touch --help' for more information.
21:02:09 fatal: Not a git repository (or any of the parent directories): .git
21:02:09 Copying default Aeolus Waves ...
21:02:09 /zynthian/zynthian-sys/scripts/update_zynthian_sys.sh: line 232: cd: /usr/share/aeolus/stops: No such file or directory
```

jofemodo, 2018-10-23:
Hi @guysoft !

I'm working on it and i hope to upload the fixes before this night. Anyway, i've seen this error:

`Failed to connect to bus: No such file or directory`

when calling systemctl. Do you know if we can call systemctl from inside the container or we should be using a different way?

Thanks!

guysoft, 2018-10-23:
In general its a bad idea to run systemd inside the chroot because you are not running a full system, only a chroot jail. The system never booted.
It means you need stuff like ``/proc`` mounted which is shared with the host, and is dependent on it.

You can google and see similar arguments why not o use systemd in docker, because what we are doing is a kind of container.

I would suggest, since I can see systemd is part of your scripts add an environment variable that skips the systemd commands. Or alternatively use a bash alias to override what the commend systemd does.
 
Anyway, I see the builds are green now :+1:

commit 441571da (2018-10-24): Fix #75: Several fixes and improvements related.
commit fe729212 (2018-10-25): Fix #75: Solve problems with some recipes that cause errors during teh ZynthianOS build process.
commit 7f03ead8 (2018-10-25): Fix #75: Disable calf recipe because build fails. Instead, install from repository an older version.
commit 20636a28 (2018-10-26): Fix #75: Fix lilv's python-bindings installation.
closed 2018-10-25 by jofemodo