zynthian/zynthian-sys #79: Add -10dB pad into Audio config

wyleu, 2019-02-14:
Add -10dB pad into Audio config to allow for high volume playing without clipping.

guysoft, 2019-02-14:
Er - I am not sure I understand what this means, can you explain what you are trying to do, with what hardware?
Or perhaps @jofemodo might understand what this is.

wyleu, 2019-02-14:
Add  function that can effectively turn down the overall volume by a specific amount to allow hi volume signals to be played at a 'decent' level. A +10dB pad could also be added to boost low level signals.

From the Nord Modular G2 external input module from the manual...

[image]

guysoft, 2019-02-14:
Ok, why not use an lv2 plugin?
(Just trying to.understand why this is a general function)

wyleu, 2019-02-14:
It's an overall characteristic at the periphery of the jack infrastructure
rather than a component within a particular engine.

wyleu, 2019-02-15:
It's an upshot of the forum thread.
https://discourse.zynthian.org/t/issue-volume-output-too-high-producing-clipping-like-noise/2718/20

Should it be applied to each engine at part of it's initialization.

jofemodo, 2021-07-25:
This should be implemented in the "mixer". BTW we should have an open task for discussing mixer features and implementation.

riban-bw, 2024-03-12:
I am closing this ticket as we now have individual control of each chain and the main mixbus in the mixer as well as more flexible processor / plugin management. If you (@wyleu) still feel stongly there is benefit for a pad in the chain then please raise new ticket in the issue tracker.

closed 2024-03-12 by riban-bw