zynthian/zynthian-sys #81: Report Multiple USB connected error on startup.

wyleu, 2019-03-18:
Connecting multiple usb midi devices on startup produces continual re-boot errors and can place zynthian in state where only a power supply cycle will reset.
This is a pernicious bug; 
could it be handled  with the wonderful new report ip address on failure screen?

guysoft, 2019-03-18:
It might be that you are plugging in too many devices for the power supply to handle.

Is there any output from ``dmesg``?

@jofemodo You might want to be able to catch under voltage and stuff like that with the ``vcgencmd get_throttled`` command. OctoPrint did this (Source: https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/pi_support/__init__.py#L18 )

jofemodo, 2019-03-18:
Hi @guysoft !

Under-voltage & over-temperature is captured and shown in the new Zynthian-UI's "Status Area", in the top-right corner. Take a look to this:

http://wiki.zynthian.org/index.php/Zynthian_UI_Users_Guide#The_Status_Area

Thanks!

wyleu, 2019-03-18:
Pretty sure I've had it with two USB MIDI connectors which are pretty low power, I'll have a look at Power Supply considerations on start up. Largely cos once the zynth is booted you can plug in the second USB device and it will work as you would expect, it's being part of boot up that seems to be a problem

wyleu, 2019-03-21:
[    3.401372] EXT4-fs (mmcblk0p2): re-mounted. Opts: errors=remount-ro,commit=120
[    3.613923] rpi-ft5406 rpi_ft5406: Probing device
[    3.618242] input: FT5406 memory based driver as /devices/virtual/input/input1
[    3.761288] usbcore: registered new interface driver snd-usb-audio
[    3.858199] snd-rpi-hifiberry-dacplus soc:sound: ASoC: CODEC DAI pcm512x-hifi not registered - will retry
[    3.955551] snd-rpi-hifiberry-dacplus soc:sound: pcm512x-hifi <-> 3f203000.i2s mapping ok
[    4.035279] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    4.077282] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    4.132586] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2
[    4.134536] cfg80211: failed to load regulatory.db
[    4.160510] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    4.164257] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    4.166345] usbcore: registered new interface driver brcmfmac
[    4.402918] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    4.404981] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available

jofemodo, 2019-03-27:
I'm pretty sure that the "reboot-loop" problem is is a power-supply problem.

A very different problem is the "soundcard-ordering" problem. Having several USB devices connected to Zynthian while booting securely will prevent that the right soundcard is selected as default, so jackd will fail to init, etc.

I've tried everything:

- Specifying indexes in "/etc/modprobe.d/alsa-base.conf":

```
options snd_soc_hifiberry_dacplus index=0
options snd-usb-audio index=-2
options vc4 index=-2
```

- Specifying default devices in "/etc/asound.conf":

```
pcm.!default {
  type hw
  card sndrpihifiberry
}

ctl.!default {
  type hw
  card sndrpihifiberry
}
```

But nothing. It seems that ALSA card ordering can't be altered when using "overlays" for loading drivers.

If someone have some information about this, please, share it!! ;-)

Regards,

wyleu, 2019-03-27:
Thanks for the work...
I've tried running it with EVERYTHING on powered hubs and it still behaves the same way.
I know it's always been an issue with Alsa ordering and this seems to be what is at the route of the problem. 
When Alsa was first introduced the idea of multi USB probably looked like an un-achievable dream !

wyleu, 2019-04-10:
I will gather more data.

smiths73v3, 2019-06-05:
I stumbled across this issues over the weekend. It looks like an issue with jackctl failing to start with multiple USB MIDI devices. It works fine with one or the other, but not both. This manifests as the "boot log" screen cycling over and over as it restarts Xorg in an attempt to restart the UI.

Removing the USB devices once the failure has occurred does not correct the issue. As @wyleu pointed out, only a reboot seems to resolve the issue. I tried systemctl to restart jack2 and zynthian, but that did nothing.

> [    2.130982] usb 1-1.2: new full-speed USB device number 4 using dwc_otg
> [    2.262043] usb 1-1.2: New USB device found, idVendor=0763, idProduct=019a, bcdDevice= 1.05
> [    2.262054] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [    2.262058] usb 1-1.2: Product: USB Axiom 61
> [    2.262062] usb 1-1.2: Manufacturer: M-Audio
> [    2.560931] usb 1-1.3: new full-speed USB device number 5 using dwc_otg
> [    2.731036] usbcore: registered new interface driver snd-usb-audio
> [    2.838732] usb 1-1.3: New USB device found, idVendor=0763, idProduct=2026, bcdDevice= 1.01
> [    2.838742] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
> [    2.838746] usb 1-1.3: Product: Oxygen 25
> [    2.838750] usb 1-1.3: Manufacturer: M-Audio

>     INFO:root:STARTING ZYNTHIAN-UI ...
>     Cannot connect to server socket err = No such file or directory
>     Cannot connect to server request channel
>     jack server is not running or cannot be started
>     JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
>     JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
>     libjackpeak failed to start jack client: 17
>     /zynthian/config/img/zynthian_logo_error_ip.png is 320x240 PNG image, color type RGB_ALPHA, 8 bit, file gamma 0.4546
>      PNG file: /zynthian/config/img/zynthian_logo_error_ip.png - Application must supply a known background gamma
>       Zooming image by 200%...done
>       Merging...done
>       Building XImage...done
>     INFO:root:ZYNTHIAN-UI CONFIG ...
>     INFO:root:Wiring Layout MCP23017_EXTRA
>     DEBUG:root:ZYNCODER A: [102, 105, 110, 113]
>     DEBUG:root:ZYNCODER B: [101, 104, 109, 112]
>     DEBUG:root:SWITCHES layout: [100, 103, 108, 111, 106, 107, 114, 115]
>     DEBUG:root:DISABLED_MIDI_IN = ['']
>     DEBUG:root:ENABLED_MIDI_OUT = ['ttymidi:MIDI_out']
>     DEBUG:root:ENABLED_MIDI_FB = ['']
>     ^Cxinit: connection to X server lost
> 
>     waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

jofemodo, 2019-06-05:
The problem is related with the way ALSA assign device numbers and it's not easy to solve. I've tried everything and didn't find a solution. I would like other people trying to solve it ... perhaps there is something i didn't understand ...

jofemodo, 2019-06-05:
Anyway, there is another ticket (#114) already addressing this problem. Although the error-case is different, the problem is the same, or very related.

wyleu, 2019-06-05:
Could we have a clear statement of how the issue presents itself? And where
the limitation lies?

smiths73v3, 2019-06-05:
I cannot see ticket (#114).

I'll take a deeper look at this and see what I can find. It sounds like this is related to alsa device mapping, and jack2?

riban-bw, 2019-06-05:
udev rules may be our friend here. Will someone describe a set of steps to reliably reproduce this so that we can have a method of testing any resolution?

smiths73v3, 2019-06-05:
I plug in 2 USB MIDI devices prior to powering on. Then the issue occurs.

smiths73v3, 2019-06-06:
Ok, I managed to get something working on my system. Not sure it's scalable since it depends on the name of the sound device in the /etc/asound.conf I don't have any other hardware to test.

**Changes to asound.conf**
> root@zynthian:~# cat /etc/asound.conf
> pcm.!default {
>         type hw
>         card sndrpihifiberry
> }
> 
> ctl.!default {
>         type hw
>         card sndrpihifiberry
> }
> 
> pcm.zynth {
>         type hw
>         card sndrpihifiberry
> }
> 
> ctl.zynth {
>         type hw
>         card sndrpihifiberry
> }

**Changes to jack2 service**
> root@zynthian:~# cat /etc/systemd/system/jack2.service                                                                      
> [Unit]                                                                                                                      
> Description=JACK2 Audio Server                                                                                              
> #After=sound.target                                                                                                          
>                                                                                                                              
> [Service]                                                                                                                    
> Environment=DBUS_SESSION_BUS_ADDRESS=unix:path=/run/dbus/system_bus_socket                                                  
> Environment=LV2_PATH=/zynthian/zynthian-plugins/lv2:/zynthian/zynthian-my-plugins/lv2:/zynthian/zynthian-my-data/presets/lv2
> LimitRTPRIO=infinity                                                                                                        
> LimitMEMLOCK=infinity                                                                                                                                       
> ExecStart=/usr/bin/jackd -v -P 70 -t 2000 -s -d alsa -d zynth -r 44100 -p 256 -n 2 -X raw                                    
> Restart=always                                                                                                              
> RestartSec=1                                                                                                                
>                                                                                                                              
> [Install]                                                                                                                    
> WantedBy=multi-user.target                                                                                                    
> 

After these changes reboot, and Jack uses the alsa "zynth" sound device, attached to **sndrpihifiberry** Previously, if one of the USB MIDI devices happened to grab the hw:0 device, Jack would fail, and the UI would boot-cycle.

So, it could be possible to add pcm.!zynth with other sound cards to **/etc/asound.conf** to get coverage for other sound cards, OR regenerated when the soundcard is picked from webconf.

riban-bw, 2019-06-06:
The issue is that alsa is enumerating each sound device which includes midi and audio interfaces. In my test case that results in:
```
 cat /proc/asound/cards
 0 [CH345_1        ]: USB-Audio - CH345
                      QinHeng CH345 at usb-3f980000.usb-1.1.3, full speed
 1 [CH345          ]: USB-Audio - CH345
                      QinHeng CH345 at usb-3f980000.usb-1.1.2, full speed
 2 [sndrpihifiberry]: RPi-simple - snd_rpi_hifiberry_dac
                      snd_rpi_hifiberry_dac
```
Because zynthian is configured to start jack2 service using hw:0 it will pick the first sound device which in this case is my first USB MIDI device. JACK2 will attempt to use its audio clock and fail (because there isn't one). We could do something similar to what @smiths73v3 suggests and define the name of supported sound cards in the webconf then use the card name instead of hw:0. We don't need to define this in asound.conf, just pass the card name to jackd, e.g.
```
/usr/bin/jackd -P 70 -t 2000 -s -d alsa -d sndrpihifiberry -r 44100 -p 256 -n 2 -X raw
```
There may be clever ways to automatically select the first PCM device but this may not be desirable. I think we probably want to define the audio device that zynthian uses to avoid it suddenly grabbing some USB device plugged in. (JACK2 expects to use a single PCM device by default for good reason so let's not try to change that just yet!).
So an initial solution would be to add a configuration element to webconf that sets the device name in jack2.service.
*HOWEVER* We should also try to stop this issue happening _before_ a user has accessed webconf, e.g. first boot with usb midi devices connected may trigger this issue. Here we should maybe use some form of automatic detection of first PCM device, i.e. default is to detect first PCM but webconf can override that. (I will look at detecting first PCM - sed, grep, awk may be required....)

riban-bw, 2019-06-06:
Can detect first PCM card with:
```
aplay -l | grep "card 0" | cut -d "[" -f2 | cut -d "]" -f1
```

riban-bw, 2019-06-06:
Example script:
```
# Replaces placeholder ZYNDUMMYALSACARD within jack2.service with the name of the first ALSA PCM card
# Brian Walton 2019-06-06

firstcard=`aplay -l|grep "card 0"|awk '{print $3}'`
if [ "my$firstcard" = "my"]
then
	# No card discovered
else
	# Found a card
	sed "s/ZYNDUMMYALSACARD/$firstcard/" /etc/systemd/system/jack2.service
	systemctl daemon-reload
fi
```
Not yet sure when and how this should be run but the idea would be to deploy zynthian with jack2.service populated with ZYNDUMMYALSACARD and to replace this with the actual soundcard name.

smiths73v3, 2019-06-06:
@riban-bw, the concept of the script works, but the "grep" portion is wrong. Here's the output from my aplay -l:

>root@zynthian:/etc# aplay -l                                                                        
> **** List of PLAYBACK Hardware Devices ****                                                         
> card 2: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 []
>   Subdevices: 0/1                                                                                   
>   Subdevice #0: subdevice #0                                                                        

I think the grep should be `grep "^card "`

>\>aplay -l | grep "^card " | awk '{print $3}'
>sndrpihifiberry

Also, under `#No card discovered` I would perform the sed substitution using **hw:0** as a fallback.

smiths73v3, 2019-06-07:
Further testing, the solution proposed by @riban-bw does not work for me.  I think the card name "sndrpihifiberry" in my case, would need to be used in an asound.conf file that maps the hardware to an alias such as zynthiansnd using the sed replacement listed above. Also note that the pcm.!default entries are not needed. I'm also going to take my comment about #No card discovered back, so that a retry of the script would have a replacement match if it does succeed. 
```
\>cat /etc/asound.conf
 
 pcm.zynthiansnd {            
         type hw              
         card ZYNDUMMYALSACARD 
 }                            
                              
 ctl.zynthiansnd {            
         type hw              
         card ZYNDUMMYALSACARD 
 }                            

```
run the script:
```
firstcard=`aplay -l|grep "^card "|awk '{print $3}'`
if [ "my$firstcard" = "my"]
then
	# No card discovered
else
	# Found a card
	sed -i -e "s/ZYNDUMMYALSACARD/$firstcard/" /etc/asound.conf
fi
```
Check the result:
```
\>cat /etc/asound.conf
 
 pcm.zynthiansnd {            
         type hw              
         card sndrpihifiberry 
 }                            
                              
 ctl.zynthiansnd {            
         type hw              
         card sndrpihifiberry 
 }                            

```

riban-bw, 2019-06-07:
@smiths73v3 good catch with the grep. Of course the first PCM capable audio card might not be card 0 but we should limit to only one result in case user has multiple PCM cards. When testing I had an I2S internal card and a USB external card. This can be done by adding `-m 1` to awk to limit quantity of results to 1, i.e. only return first card found.

Are you sure you restored your asound.conf file to zynthian default before testing this? It _should_ work. ALSA enumerates cards as it finds them and names each. It is this name that the awk should find. Jack expects an ALSA PCM card index or name to be passed hence I don't know why this does not work for you.

It would be good to avoid tweaking too many places in the config chain. We may wish to add multiple card support at some point (although that is thwart with issues). The simpler the implementation the easier it is to unpick issues.

Where would we put such an automated check? It could be a service started before Jack or could be part of the Jack service control itself. It feels wrong to run this on every boot but it is difficult to know how many boots before the zynthian builder has installed a working audio card.

What happens if the audio card changes or is removed? Maybe the zynth uses a USB soundcard and it gets swapped for another card. We need webconf working even without jack - need to validate that the zynth will boot and remain running in such situations.

smiths73v3, 2019-06-07:
@riban-bw Good catch on the ```-m``` 

I think we need to continue with the card **name** not the card index. The card name is more static than the index. The problem is that while the index could be used as hw:2 in jack2.service config, the card name cannot be used in this way. The problem with using card index as you have pointed out is that it will change if the user changes the USB configuration and would require touching the jack2.service config file at each boot.

By using an asound.conf alias **zynthiansnd** this lets alsa pick the sound device by the card name parsed using original ``` {print $3}``` method you proposed method. Jack2.service file would be static, and the alias **zynthiansnd** would always be used. Only the /etc/asound.conf would need to be modified, and also would only need to be modified once at setup, and then if the user makes a change in web-ui config

I think the script to modify /etc/asound.conf would be handled in the same manner as the /boot/config.txt file. It can be set up on the initial boot with the "official" defaults and a web-ui config would regenerate the /etc/asound.conf with the users selected soundcard. The card name could either be scanned by aplay -l, or preferably by using a settings table for each supported soundcard that contains the aplay name similar to the method used for "Display" settings

I'll do some testing this weekend to ensure that JACK2 comes up even when the audio is mis-configured. in this way. Perhaps I can use a "dummy" and override it when a real sound device is detected?

riban-bw, 2019-06-07:
The card name can be used in jack2.service. That is exactly what I am doing and effectively what you are doing by naming the card in asound.conf.

smiths73v3, 2019-06-08:
I can't get jack2.service to start with the sndrpihifiberry card name. Here's some sample output. It looks like ALSA does not see the PCM device when it's specified by the cardname.

```
root@zynthian:~# aplay -l
**** List of PLAYBACK Hardware Devices ****
card 2: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry DAC+ HiFi pcm512x-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
```
root@zynthian:~# aplay -l | grep -m 1 "^card " | awk '{print $3}'
sndrpihifiberry
root@zynthian:~# grep `aplay -l | grep -m 1 "^card " | awk '{print $3}'` /etc/systemd/system/jack2.service | tail
ExecStart=/usr/bin/jackd -P 70 -t 2000 -s -d alsa -d sndrpihifiberry -r 44100 -p 256 -n 2 -X raw
```
```
root@zynthian:~# grep `aplay -l | grep -m 1 "^card " | awk '{print $3}'` zynth.log_bad_jack.txt | tail
Jun 08 01:25:43 zynthian jackd[2744]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM sndrpihifiberry
Jun 08 01:25:44 zynthian jackd[2751]: ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL sndrpihifiberry
Jun 08 01:25:44 zynthian jackd[2751]: creating alsa driver ... sndrpihifiberry|sndrpihifiberry|256|2|44100|0|0|nomon|swmeter|-|32bit
Jun 08 01:25:44 zynthian jackd[2751]: ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL sndrpihifiberry
Jun 08 01:25:44 zynthian jackd[2751]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM sndrpihifiberry
Jun 08 01:25:46 zynthian jackd[2756]: ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL sndrpihifiberry
Jun 08 01:25:46 zynthian jackd[2756]: creating alsa driver ... sndrpihifiberry|sndrpihifiberry|256|2|44100|0|0|nomon|swmeter|-|32bit
Jun 08 01:25:46 zynthian jackd[2756]: ALSA lib control.c:1373:(snd_ctl_open_noupdate) Invalid CTL sndrpihifiberry
Jun 08 01:25:46 zynthian jackd[2756]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM sndrpihifiberry
Jun 08 01:25:46 zynthian jackd[2756]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM sndrpihifiberry

```

riban-bw, 2019-06-08:
@smiths73v3 you need to prefix the card name with `hw:` so that the line in jack2.service becomes:
```
ExecStart=/usr/bin/jackd -P 70 -t 2000 -s -d alsa -d hw:sndrpihifiberry -r 44100 -p 256 -n 2 -X raw
```
I had the idea that we could add an "Auto Detect" configuration option to webconf which could be default. This would perform the detection of first sound card as described here and populate jack2.service. All other options would use predefined card names, e.g. hw:sndrpihifiberry for most I2S cards. I think we should also put in place a check that there is a PCM card and not attempt to start jack2 service unless one exists.

rodamaralgeo, 2019-06-08:
Hi guys,

Using hw:audioinjectorpi in the jackd configuration worked perfectly!

smiths73v3, 2019-06-08:
@riban-bw  hw:sndrpihifiberry in jackd config worked for me. Thanks! It looks like your original script will work. I modified the sed a little bit so that it will search for hw: and can be re-executed, and the script could parse a parameter if a "forced" configuration is picked in the web-config.

```
if [ "my$1" = "my" ]                                                        
then                                                                        
        firstcard=`aplay -l|grep -m 1 "^card "|awk '{print $3}'`            
else                                                                        
        firstcard=$1                                                        
fi                                                                          
                                                                            
if [ "my$firstcard" = "my" ]                                                
then                                                                        
        # No card discovered                                                
        echo "No PCM device found"                                          
else                                                                        
        # Found a card                                                      
        sed "s/hw:[^ ]* /hw:$firstcard /" /etc/systemd/system/jack2.service 
        systemctl daemon-reload                                             
fi  
```

I really like the idea of "auto-config" and also not starting jack if there is no PCM device. The fewer reasons there are for the UI not coming up the better.

riban-bw, 2019-06-10:
A quick win would be to change zynthian-webconf/lib/audio_config_handler.py to replace `hw:0` with the relevant card names. I note that it is already done for some cards, e.g. Behringer UCA222 (USB) has `hw:CODEC`. So the code would look like:
```
...
                ['HifiBerry DAC+ light', {
                        'SOUNDCARD_CONFIG':'dtoverlay=hifiberry-dac',
                        'JACKD_OPTIONS': '-P 70 -t 2000 -s -d alsa -d sndrpihifiberry -r 44100 -p 256 -n 2 -X raw'
                }],
...
```
With this a user can set their primary audio card. This will allow Jack to start up and presents a significant improvement for user experience. (There remains some work to implement the other ideas suggested here.) I have not submitted a patch because I don't know the card names for each of the supported cards.

smiths73v3, 2019-06-11:
That sounds like a great solution.

jofemodo, 2019-09-16:
It's a great solution and i missed to read it ...
Now it's been reported in another issue (#85) and has been implemented for Hifiberry soundcards. We need the device names of all the supported soundcards for extending the solution.

closed 2019-09-16 by jofemodo

wyleu, 2019-09-17:
for the hifiberry amp2 machine…

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 2: sndrpihifiberry [snd_rpi_hifiberry_dacplus], device 0: HiFiBerry
DAC+ HiFi pcm512x-hifi-0 []
Subdevices: 0/1
Subdevice #0: subdevice #0

for the audio injector machine (zynthian-alm)
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: audioinjectorpi [audioinjector-pi-soundcard], device 0:
AudioInjector audio wm8731-hifi-0 []
Subdevices: 0/1
Subdevice #0: subdevice #0

commit 8be8e391 (2019-11-25): Fix #81: Use device name for AudioInjector activation.