zynthian/zynthian-sys #83: Add MIDI and audio device firmware to build

riban-bw, 2019-04-12:
May I request that we include some MIDI and audio device firmware in the image, e.g. midisport-firmware?

guysoft, 2019-04-12:
Can you link to the specific resource?
We need to see
1. who will it server?
2. Is the license compatible?
3. how big is it?

jofemodo, 2019-04-12:
Of course you can request that, but a little bit of extra info would be desirable, as @guysoft  has wrote.
Also, some info about the use-case, the motivations for including that firmware, etc ...

riban-bw, 2019-04-12:
Okay - specifically I would like to see the midisport-firmware debian package installed. This provides firmware for the M-Audio / Midman midisport 2x2 USB interface without which the device does not function. It supports the following devices:
- MidiSport 1x1
 - MidiSport 2x2
 - MidiSport 4x4
 - MidiSport 8x8
 - KeyStation (old models: 49, 61)
 - Oxygen
 - Radium49
 - Radium61
 - Uno

Installed size is 160kB and includes files:
```
/lib/udev/rules.d/42-midisport-firmware.rules
/usr/share/doc/midisport-firmware/README
/usr/share/doc/midisport-firmware/buildinfo.gz
/usr/share/doc/midisport-firmware/changelog.Debian.gz
/usr/share/doc/midisport-firmware/changelog.gz
/usr/share/doc/midisport-firmware/copyright
/usr/share/usb/maudio/MidiSport1x1.ihx
/usr/share/usb/maudio/MidiSport2x2.ihx
/usr/share/usb/maudio/MidiSport4x4.ihx
/usr/share/usb/maudio/MidiSport8x8-2.10.ihx
/usr/share/usb/maudio/MidiSport8x8-2.21.ihx
/usr/share/usb/maudio/MidiSportKS.ihx
/usr/share/usb/maudio/MidiSportLoader.ihx
```
It depends on fxload which is 38kB installed.

midisport-firmware is in the non-free archive area so it does not necessarily comply with the DFSG, probably because it is a binary blob. I cannot see whether there are license restrictions which may conflict with distribution. If this is a concern then maybe we can add an option to install such firmware via the web interface and / or admin menu. Users can ssh and perform the install themselves but I believe we aspire to a more turnkey solution. We are not all tech boffins!

wyleu, 2019-04-12:
I've had real problems with a 4x4 on Linux.... Gave up on it in the end....

riban-bw, 2019-04-12:
> I've had real problems with a 4x4 on Linux.... Gave up on it in the end....

That is odd - it just works for me and very well too! Just `apt install midisport-firmware` and away you go.

jofemodo, 2019-05-07:
It should be available on the next night build ;-)

wyleu, 2019-05-07:
I suspect I may have predated midisport-firmware ... :-D 

I've got one of these boxes so this is interesting!

riban-bw, 2019-05-11:
Tested in Aruk - works fine, thanks.

closed 2019-05-11 by riban-bw