zynthian/zynthian-sys #84: Is Bluetooth enabled?

riban-bw, 2019-04-14:
In the blurb on the [Zynthian homepage](http://zynthian.org/) it lists Bluetooth under connectivity. I am not sure how this is used. config.txt has the following entries:

```
[pi3]
# Disable Blutooth Support for HifiBerry compatibility with RBPi 3
dtoverlay=pi3-disable-bt
# Swap back the ttyAMA0 to GPIO
dtoverlay=pi3-miniuart-bt
```

pi3-disable-bt and pi3-miniuart-bt seem to be mutually exclusive. pi3-disable-bt disables bluetooth whilst pi3-miniuart-bt enables it.

From [RPi docs](https://www.raspberrypi.org/documentation/configuration/uart.md):

> pi3-disable-bt disables the Bluetooth device and restores UART0/ttyAMA0 to GPIOs 14 and 15. It is also necessary to disable the system service that initialises the modem so it doesn't use the UART: sudo systemctl disable hciuart.

> pi3-miniuart-bt switches the Raspberry Pi 3 and Raspberry Pi Zero W Bluetooth function to use the mini UART (ttyS0), and restores UART0/ttyAMA0 to GPIOs 14 and 15.

I have previously read that bluetooth can be disabled to improve RPi performance so it may be desirable to disable Bluetooth by default and maybe offer a configuration to enable it if required. 

I think (but may be mistaken) that your current intention is to disable bluetooth so the pi3-miniuart-bt entry may be removed.

jofemodo, 2019-04-25:
1.) By default the RBPi's Bluetooth uses the standard UART. As we use the standard UART for MIDI-IN/OUT, i used the "pi3-miniuart-bt" flag for forcing to use the second UART, that is a limited UART.

2.) Aditionally, Bluetooth is disabled because most of users doesn't need it. It could be enabled by those users that need it, but if you want to have the MIDI-IN/OUT, the "pi3-miniuart-bt" flag must be set.

So i choosed to explicitly leave both flags in the config file as a remember for me and others ;-)

jofemodo, 2019-05-06:
Can i close this? ;-)

guysoft, 2019-05-06:
It might be worth adding documentation somewhere. Not sure though how critical this is for anyone other than @riban-bw

riban-bw, 2019-05-06:
I would recommend amending the blurb on the project home page and probably in wiki too so that it is clear that Bluetooth is disabled by default. For example, Bluetooth is specifically listed in"Connectivity" section [here](http://zynthian.org).

jofemodo, 2019-05-06:
A new web is on the way and i will fix this and other details then.

Thanks!

closed 2019-05-06 by jofemodo

jofemodo, 2019-05-06:
Until then, i removed the "blutooth" line from the website.
BTW, I just created a wiki account for you. You should have received an email with the password. Please, confirm ...