zynthian/zynthian-sys #85: Failed to start Raise network interfaces

smiths73v3, 2019-04-19:
I am having an issue shortly after boot from a power off state. The UI comes up as expected and USB MIDI works for about 30 seconds. After this audio "glitches out" and no more MIDI "Green M on screen" are seen. The system is not accessible from WIFI or by direct connected ethernet.

Restarting the UI clears up the condition, as does rebooting the system from the onscreen UI.

The only differences I have found between a good boot and a bad boot are related to network interfaces, and the message "Failed to start Raise network interfaces." seen in the bad case (from the /var/log/boot.log file)

I am running a HiFiBerry DAC+ with RPi Foundation 7" touchscreen, and dummies controller. 
Wifi when it is working is connected to a valid SSID, with the correct country code set "US"

zyncoder: master (62b3dc3)
zynthian-ui: master (69883fd)
zynthian-sys: master (4473bea)
zynthian-data: master (6712b1c)
zynthian-webconf: master (81ee0c4)

smiths73v3, 2019-04-24:
Looking at networking.service after restarting wifi shows that networking failed due to a **timeout**

```
root@zynthian:/etc/network# systemctl status networking.service
● networking.service - Raise network interfaces
   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/networking.service.d
           └─reduce-timeout.conf
   Active: failed (Result: timeout) since Fri 2019-04-19 03:20:29 UTC; 4 days ago
     Docs: man:interfaces(5)
  Process: 286 ExecStart=/sbin/ifup -a --read-environment (code=killed, signal=TERM)
  Process: 273 ExecStartPre=/bin/sh -c [ "$CONFIGURE_INTERFACES" != "no" ] && [ -n "$(ifquery --read-environment --list --exclude=lo)" ] && udevadm settle (code= Main PID: 286 (code=killed, signal=TERM)
   CGroup: /system.slice/networking.service

Apr 19 03:20:28 zynthian systemd[1]: Starting Raise network interfaces...
Apr 19 03:20:29 zynthian systemd[1]: networking.service: Start operation timed out. Terminating.
Apr 19 03:20:29 zynthian systemd[1]: networking.service: Main process exited, code=killed, status=15/TERM
Apr 19 03:20:29 zynthian systemd[1]: Failed to start Raise network interfaces.
Apr 19 03:20:29 zynthian systemd[1]: networking.service: Unit entered failed state.
Apr 19 03:20:29 zynthian systemd[1]: networking.service: Failed with result 'timeout'.
Warning: networking.service changed on disk. Run 'systemctl daemon-reload' to reload units.
lines 1-18/18 (END)
```

The  timeout listed above is set in .../reduce-timeout.conf 
```
[Service]           
TimeoutStartSec=1   
```
I am going to experiment with extending this timeout since I think it is taking longer to get an IP address if it has previously expired (vs. a renew almost instantly upon reboot)

smiths73v3, 2019-04-28:
Disabling WIFI in the config file fixes the audio glitch, no MIDI after 30 seconds issue. It appears that this is related to WIFI, and Network connectivity. 

I suspect that an issue with networking is preventing/crashing some other service. I'm not exactly sure where to find that. I'm going to keep digging.

smiths73v3, 2019-05-12:
There is a good possibility that the cause of my issue is related to a low-power condition on boot. I see the lightning bolt on boot while bringing up X.org and getting to a stable UI but once the system is up it stabilizes. Foundation 7" display seems to be taking a ton of power, and I don't have my final power supply finalized yet.

If there is a dependency on wifi/networking to be in an expected state for the zynthian-ui to come up that should still be looked at. I now understand that low power issues can cause many unexpected conditions, and this issue could be caused by low power.

jofemodo, 2019-05-17:
Can you confirm that your problem is related with a low-power condition?
If not, you could try the latest Aruk "green" SD-image:

https://os.zynthian.org/2019-05-17-zynthianos-stretch-lite-1.0.0.zip

WIFI configuration has been improved lately ;-)

smiths73v3, 2019-05-17:
I have upgraded my power as part of my ongoing DIY build. Rock solid power now.  I'll check for the issue tomorrow, and then also test the ARUK "green"

smiths73v3, 2019-05-18:
Ok, I have confirmed that the issue exists after the power issues were resolved. I will try with the suggested ARUK image.

smiths73v3, 2019-05-23:
I was able to reproduce the issue with 2019-05-17 image. The issue still exists.

jofemodo, 2019-05-23:
For me is working perfectly in the last night builds. Try today's build. It's RC-2 ;-)

smiths73v3, 2019-05-25:
I still have the issue with 2019-05-23.

Seems like it is the first boot after DHCP lease expires, about 18 hours after last boot.
after ~30 seconds audio glitches and no midi activity is seen.
WIFI not working.

If I restart UI, everything comes back normally, and no further issues are seen.

smiths73v3, 2019-06-07:
Ok, I think I have log evidence of what is going on, from the looks of things it's indirectly related to NTP and system time that gets updated when the WIFI interface comes up. 

In the logs below, time jumps by roughly 19 hours when wifi connects :
``` 
Jun 07 04:17:57 zynthian startx[6894]: DEBUG:root:export ZYNTHIAN_WIFI_MODE="on"
Jun 07 23:46:58 zynthian systemd[1]: Time has been changed
```
 Then **JACK2** gets a timeout in that same second tick (23:46:58)
```
Jun 07 23:46:58 zynthian startx[6894]: JackPosixSemaphore::TimedWait err = Connection timed out
```
This results in a jackd failure, and it shutsdown, and does not come back up.
More complete logs below:
```
Jun 07 04:17:53 zynthian avahi-daemon[408]: Registering new address record for fdf9:1ff6:a778::4bf on wlan0.*.
Jun 07 04:17:56 zynthian dhcpcd[652]: wlan0: leased 192.168.1.153 for 21600 seconds
Jun 07 04:17:56 zynthian avahi-daemon[408]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.153.
Jun 07 04:17:56 zynthian avahi-daemon[408]: New relevant interface wlan0.IPv4 for mDNS.
Jun 07 04:17:56 zynthian dhcpcd[652]: wlan0: adding route to 192.168.1.0/24
Jun 07 04:17:56 zynthian dhcpcd[652]: wlan0: adding default route via 192.168.1.1
Jun 07 04:17:56 zynthian avahi-daemon[408]: Registering new address record for 192.168.1.153 on wlan0.IPv4.
Jun 07 04:17:56 zynthian dhcpcd[652]: eth0: deleting default route
Jun 07 04:17:57 zynthian startx[6894]: DEBUG:root:export ZYNTHIAN_WIFI_MODE="on"
Jun 07 23:46:58 zynthian systemd[1]: Time has been changed
Jun 07 23:46:58 zynthian systemd[4784]: Time has been changed
Jun 07 23:46:58 zynthian startx[6894]: JackPosixSemaphore::TimedWait err = Connection timed out
Jun 07 23:46:58 zynthian startx[6894]: SuspendRefNum error
Jun 07 23:46:58 zynthian startx[6894]: JackClient::Execute error name = ZynMidiRouter
Jun 07 23:46:58 zynthian startx[6894]: JackPosixSemaphore::TimedWait err = Connection timed out
Jun 07 23:46:58 zynthian startx[6894]: SuspendRefNum error
Jun 07 23:46:58 zynthian startx[6894]: JackClient::Execute error name = jackpeak
Jun 07 23:46:58 zynthian ttymidi[527]: JackPosixSemaphore::TimedWait err = Connection timed out
Jun 07 23:46:58 zynthian ttymidi[527]: SuspendRefNum error
Jun 07 23:46:58 zynthian ttymidi[527]: JackClient::Execute error name = ttymidi
Jun 07 23:46:58 zynthian a2jmidid[524]: JackPosixSemaphore::TimedWait err = Connection timed out
Jun 07 23:46:58 zynthian a2jmidid[524]: SuspendRefNum error
Jun 07 23:46:58 zynthian a2jmidid[524]: JackClient::Execute error name = a2j
Jun 07 23:46:58 zynthian systemd-timesyncd[312]: Synchronized to time server 193.228.143.12:123 (2.debian.pool.ntp.org).
Jun 07 23:46:58 zynthian systemd[1]: apt-daily.timer: Adding 15min 31.615797s random time.
Jun 07 23:46:58 zynthian systemd[1]: apt-daily-upgrade.timer: Adding 9min 49.994237s random time.
Jun 07 23:46:58 zynthian systemd[1]: Reached target Network is Online.
Jun 07 23:46:58 zynthian systemd[1]: Starting Daily apt download activities...
Jun 07 23:46:58 zynthian startx[6894]: ERROR:root:Jack Audio XRUN!
Jun 07 23:46:58 zynthian a2jmidid[524]: WARNING: JACK server shutdown notification received.
Jun 07 23:46:58 zynthian a2jmidid[524]: Bridge stopping...
Jun 07 23:46:59 zynthian startx[6894]: ERROR:root:Jack Audio XRUN!
Jun 07 23:46:59 zynthian a2jmidid[524]: port deleted: Midi Through [14] (capture): Midi Through Port-0
Jun 07 23:46:59 zynthian a2jmidid[524]: Server is not running
Jun 07 23:46:59 zynthian a2jmidid[524]: port deleted: Midi Through [14] (playback): Midi Through Port-0
Jun 07 23:46:59 zynthian a2jmidid[524]: Server is not running
Jun 07 23:46:59 zynthian a2jmidid[524]: Server is not running
Jun 07 23:46:59 zynthian a2jmidid[524]: ERROR: a2j_destroy: Cannot close jack client (-1)
Jun 07 23:46:59 zynthian a2jmidid[524]: Bridge stopped
```

jofemodo, 2019-06-21:
Could you check again with the last night build?

smiths73v3, 2019-06-22:
Looks good! Tested with 2019-06-20-zynthianos-stretch-lite-1.0.0

```
Apr 08 11:45:12 zynthian avahi-daemon[399]: Withdrawing address record for 169.254.218.246 on wlan0.                                         
Apr 08 11:45:12 zynthian avahi-daemon[399]: Leaving mDNS multicast group on interface wlan0.IPv4 with address 169.254.218.246.               
Apr 08 11:45:12 zynthian dhcpcd[699]: wlan0: deleting route to 169.254.0.0/16                                                                
Apr 08 11:45:12 zynthian avahi-daemon[399]: Joining mDNS multicast group on interface wlan0.IPv4 with address 192.168.1.154.                 
Jun 22 01:41:10 zynthian systemd-timesyncd[277]: Synchronized to time server 192.150.149.245:123 (1.debian.pool.ntp.org).                    
Jun 22 01:41:10 zynthian systemd[1]: Time has been changed                                                                                   
Jun 22 01:41:10 zynthian systemd[1]: apt-daily.timer: Adding 9h 46min 18.726938s random time.                                                
Jun 22 01:41:10 zynthian systemd[1]: apt-daily-upgrade.timer: Adding 31min 12.854696s random time.                                           
Jun 22 01:41:10 zynthian systemd[1]: Reached target Network is Online.                                                                       
Jun 22 01:41:10 zynthian systemd[1]: Starting Daily apt download activities...                                                               
Jun 22 01:41:37 zynthian dbus[367]: [system] Activating via systemd: service name='org.freedesktop.PackageKit' unit='packagekit.service'     
Jun 22 01:41:37 zynthian dbus[367]: [system] Activation via systemd failed for unit 'packagekit.service': Unit packagekit.service is masked. 
Jun 22 01:41:37 zynthian dbus[367]: [system] Activating via systemd: service name='org.freedesktop.PackageKit' unit='packagekit.service'     
Jun 22 01:41:37 zynthian dbus[367]: [system] Activation via systemd failed for unit 'packagekit.service': Unit packagekit.service is masked. 
Jun 22 01:41:49 zynthian systemd[1]: Started Daily apt download activities.                                                                  
Jun 22 01:41:49 zynthian systemd[1]: apt-daily.timer: Adding 1h 13min 27.852349s random time.                                                
Jun 22 01:41:49 zynthian systemd[1]: apt-daily.timer: Adding 3h 3min 6.604599s random time.                                                  
Jun 22 01:41:49 zynthian systemd[1]: Starting Daily apt upgrade and clean activities...                                                      
Jun 22 01:42:10 zynthian systemd[1]: Started Daily apt upgrade and clean activities.                                                         
Jun 22 01:42:10 zynthian systemd[1]: apt-daily-upgrade.timer: Adding 51min 37.982167s random time.                                           
Jun 22 01:42:10 zynthian systemd[1]: apt-daily-upgrade.timer: Adding 7min 57.239113s random time.                                            
Jun 22 01:48:18 zynthian startx[517]: ERROR:root:Jack Audio XRUN!                                                                            
Jun 22 01:48:18 zynthian startx[517]: ERROR:root:Jack Audio XRUN!                                                                            
Jun 22 01:48:18 zynthian startx[517]: ERROR:root:Jack Audio XRUN!                                                                            
Jun 22 01:48:18 zynthian startx[517]: ERROR:root:Jack Audio XRUN!                                                                            
```

closed 2019-06-22 by smiths73v3