zynthian/zynthian-sys #87: vi and tabs

mheidt, 2019-05-05:
could we add this in a file ~/.vimrc

set softtabstop=0 noexpandtab

Suddenly I couldn't change files with vi anymore correctly. Tabs were changed to spaces.

mheidt, 2019-06-09:
And I don't want to change this file with every new os build :)

riban-bw, 2019-06-10:
Yes please! Vim in Aruk is driving me up the wall. I can't enter tabs and I can't highlight and copy text. I have to use nano, blurk!!!

jofemodo, 2019-06-12:
I've added a global "vimrc.local" to /etc/vim. Please, confirm that this solves your problems ;-)

mheidt, 2019-10-13:
The file is missing in the latest buster image

jofemodo, 2019-10-13:
Curious ... the file is on the repo:

https://github.com/zynthian/zynthian-sys/tree/master/etc/vim

And it's been copied ...

https://github.com/zynthian/zynthian-sys/blob/fb5e4544c088c31afffbfb43da43af7032d10fcb/scripts/update_zynthian_sys.sh#L267

I will take a look ...

mheidt, 2019-12-03:
But ~/.vimrc still not there

jofemodo, 2019-12-05:
Try updating now ;-)

closed 2019-12-05 by jofemodo