zynthian/zynthian-sys #88: USB Mouse not recognised

wyleu, 2019-05-06:
A USB mouse no longer operates the zynthian GUI.

Without Mouse.
lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 12M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 12M

With Mouse

lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 12M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 12M

WITH MOUSE & AKAI Keyboard....
lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 12M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 12M
        |__ Port 2: Dev 9, If 0, Class=Human Interface Device, Driver=usbhid, 12M
        |__ Port 2: Dev 9, If 1, Class=Audio, Driver=snd-usb-audio, 12M
        |__ Port 2: Dev 9, If 2, Class=Audio, Driver=snd-usb-audio, 12M

guysoft, 2019-05-06:
Ouptut of ``dmesg``?

wyleu, 2019-05-07:
dmesg:-
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.18.7-v7+ (matuschd@storage) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03)) #1 SMP Fri Sep 14 08:48:48 CEST 2018
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 8 MiB at 0x3ac00000
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   Normal zone: 2133 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 242688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x7c/0x490 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu @(ptrval) s39104 r8192 d22336 u69632
[    0.000000] pcpu-alloc: s39104 r8192 d22336 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=1024 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.speed=1 dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 940124K/970752K available (7168K kernel code, 623K rwdata, 2144K rodata, 1024K init, 820K bss, 22436K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xbb800000 - 0xff800000   (1088 MB)
                   lowmem  : 0x80000000 - 0xbb400000   ( 948 MB)
                   modules : 0x7f000000 - 0x80000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (8160 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (1024 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 624 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 821 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 26013 entries in 77 pages
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000003] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000012] Switching to timer-based delay loop, resolution 52ns
[    0.000156] Console: colour dummy device 80x30
[    0.000433] console [tty1] enabled
[    0.000455] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000476] pid_max: default: 32768 minimum: 301
[    0.000666] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000684] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001304] Disabling memory control group subsystem
[    0.001358] CPU: Testing write buffer coherency: ok
[    0.001640] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.001990] Setting up static identity map for 0x100000 - 0x10003c
[    0.002079] Hierarchical SRCU implementation.
[    0.002517] smp: Bringing up secondary CPUs ...
[    0.003031] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.003566] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.004058] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.004117] smp: Brought up 1 node, 4 CPUs
[    0.004155] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.004166] CPU: All CPU(s) started in HYP mode.
[    0.004176] CPU: Virtualization extensions available.
[    0.004722] devtmpfs: initialized
[    0.010506] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.010659] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.010683] futex hash table entries: 1024 (order: 4, 65536 bytes)
[    0.011029] pinctrl core: initialized pinctrl subsystem
[    0.011515] NET: Registered protocol family 16
[    0.012661] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.015450] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.015466] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.015593] Serial: AMBA PL011 UART driver
[    0.016669] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[... 207 lines cut ...]
[    5.052712] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    5.655000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[    5.655950] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xC5E1
[    5.928959] bcm2835-i2s 3f203000.i2s: I2S SYNC error!
[    6.332249] uart-pl011 3f201000.serial: no DMA platform data
[    8.751508] Bluetooth: Core ver 2.22
[    8.751588] NET: Registered protocol family 31
[... cut ...]
[    9.067642] Bluetooth: BNEP filters: protocol multicast
[    9.067660] Bluetooth: BNEP socket layer initialized
[   10.075873] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   10.075895] brcmfmac: power management disabled
[   10.619270] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   10.619310] brcmfmac: power management disabled
[   11.377631] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.541778] random: crng init done
[   33.541789] random: 7 urandom warning(s) missed due to ratelimiting

wyleu, 2019-05-18:
Seems to have cleared itself up with the release of #498

closed 2019-05-18 by wyleu