zynthian/zynthian-sys #89: Qjackctl not displaying

riban-bw, 2019-05-06:
- ssh to Zynthian
- qjackctl

qjackctl seems to be running but the windows do not render in my x-server. Same if it point at a different x-server, e.g.
DISPLAY=192.168.1.8:0 qjackctl.

On my Windows PC running VcXsrv (which works for other applications forwarded from the zynthian) I see two windows (default and connections) in the task bar but neither will render on the screen.

riban-bw, 2019-05-06:
OS: Raspbian GNU/Linux 8.0 (jessie)

jofemodo, 2019-05-06:
qjackctl version?
I tested in old Gorgona Omgea with 3.12 and 5.7.10 (fresh compiled from github) and it works for me connecting from Linux. But have similar problems with stretch's qjackctl package.

riban-bw, 2019-05-15:
I have since seen qjackclt display within Windows vi VcXsv. I have not figured out what causes this to work but it sometimes fails and sometimes works. I suspect it is an issue with VcXsrv or the configuration on the Windows laptop. I will close this issue and investigate the Windows specific issue spearately.

jofemodo, 2019-05-16:
I will keep the issue open because i can't open the qjackctl win from my Fedora computer. Only with ZynthianOS. Gorgona works nicely.

guysoft, 2019-05-19:
> Yes. The OS line in the webconf's dashboard. If it's "Jessie", then it's a Gorgona. If it's "Stretch", then it's Aruk ;-)

Might want to include an "about" or something so people can pull this out easily. Now that its a nightly you can generate all the version info from ``/etc/zynthianos_verion``

mark-orion, 2020-11-02:
Had the same problem. The package seems to be buggy so compiling it myself solved the problem:
apt remove qjackctl
apt install portaudio19-dev
git clone https://git.code.sf.net/p/qjackctl/code qjackctl-code
cd qjackctl-code
./autogen.sh
make
cd src <<---- this is important otherwise make install will fail because of a missing “lupdate” what can only be solved by installing a whole load of in this case not needed QT dev tools.
make install

And then you can enjoy QJackctl - assuming you run tigervnc:
DISPLAY=:1 qjackctl

commit 73396e2d (2020-11-02): Fix #89: Qjackctl UI not showing remotely (X11 or VNC). Install from source code solves the problem.
closed 2020-11-02 by jofemodo