zynthian/zynthian-sys #90: MOD-SDK Start Error

rodamaralgeo, 2019-05-06:
Version: **Zynthian-OS (2019-05-03)**
When trying to start MOD-SDK the following error is displayed:
[image]
[image]

riban-bw, 2019-05-15:
Is this a requirement for ARUK? I am not aware that MOD-SDK is a core function of Zynthian. Will the failure to run MOD-SDK block roll-out of ARUK?

jofemodo, 2019-05-16:
It's not a core function. MOD-SDK is a Web-GUI that is mainly useful for LV2-plugin developpers. It worked in Gorgona and should work in Aruk, but it's not critical at all. Also, i'm pretty sure that the problem is easy to solve ;-)

jofemodo, 2021-09-21:
Since mod-sdk is not present by default on recent SD images, i will close this.

closed 2021-09-21 by jofemodo