zynthian/zynthian-sys #92: ZynthianOS build: rpi-update ??

jofemodo, 2019-05-09:
Hi @guysoft !

What about doing a rpi-update during ZynthianOS building? ;-)

guysoft, 2019-05-12:
Its risky because you are changing the firmware. If there is a realtime kernel of something that it might break the whole system. Welcome to try and see what bugs you encounter ;)

You could add a CustomPiOS module that runs a system update.

Is there an argument why to do that? Assuming you are using the latest base image?

jofemodo, 2019-05-13:
There is a good reason for making this ... the next official Zynthian Kit will include the HifiBerry DAC+ADC, what needs kernel v4.19. As the current raspbian base release comes with 4.14, we need a rpi-update for getting the latest kernel and drivers.

jofemodo, 2019-05-13:
My question is if you think that rpi-update itself would work OK on the CustomPiOS emulated environment ...
I already tested the rpi-update in a recent night-build image and it worked fine, without breaking things ;-)

guysoft, 2020-01-23:
Hey just saw now.
In my builds I don't do it. Since we have no idea what and when Rpi foundation would be pushing stuff there I personally don't do that. But its your distro so you get to choose. Should I close?

jofemodo, 2020-01-23:
Yes. Let's close ....

closed 2020-01-23 by jofemodo