zynthian/zynthian-sys #93: ZynthianOS: Jackd server fails randomly

jofemodo, 2019-05-10:
- The error is random but reproducible if running time enough.
- It occurs also when no layers are defined or no playing at all.
- Zynthian UI logs:
   - Server is not running
   - ERROR:root:Jack XRUN!

jofemodo, 2019-05-10:
Can somebody confirm this problem with recent ZynthianOS images?
I'm testing with this:

wyleu, 2019-05-10:
If you mean that Jack crashes after a period of time (5 minutes or so with
fluidsynth yamaha} then I can confirm it.

wyleu, 2019-05-10:
Doesn't seem to happening on audio injector but I will test that overnight
this evening. I'll try to leave all my instances going to see what I can
collect. It shows up as repeating can't start server in the gui log

wyleu, 2019-05-10:
Arpeggio from akai MPK playing against fluidsynth viola on audio injector zynth...

wyleu, 2019-05-10:
hifiberry Amp + playing an arpeggio for over an hour now. . . . 

zyncoder: master (f672128) 
zynthian-ui: master (37a155e) 
zynthian-sys: master (550dc55) 
zynthian-data: master (6712b1c) 
zynthian-webconf: master (798dd8d)

wyleu, 2019-05-10:
Does the xrun happen when the engine does nothing ...?
both have run for over an hour and a half now...?

wyleu, 2019-05-10:
and it's driving me bloody mad !!!

wyleu, 2019-05-10:
temp icon on amp2..

wyleu, 2019-05-10:
amp2 .... 
May 10 21:00:44 zynthian-amp2 startx[3016]: DEBUG:root:proc output: 
May 10 21:00:44 zynthian-amp2 startx[3016]: \set_control 3, 0.125984 
May 10 21:00:45 zynthian-amp2 startx[3016]: DEBUG:root:proc output: 
May 10 21:00:45 zynthian-amp2 startx[3016]: \set_control 3, 0.110236 
May 10 21:00:45 zynthian-amp2 startx[3016]: DEBUG:root:proc output: 
May 10 21:00:45 zynthian-amp2 startx[3016]: \set_control 3, 0.078740 
May 10 21:00:45 zynthian-amp2 startx[3016]: DEBUG:root:proc output: 
May 10 21:00:45 zynthian-amp2 startx[3016]: \set_control 3, 0.062992

wyleu, 2019-05-10:
nord ... AudioIjector ...

May 10 18:05:26 zynthian-nord startx[487]: ERROR:root:Jack XRUN! 
May 10 18:55:53 zynthian-nord startx[487]: ERROR:root:Jack XRUN! 
May 10 18:55:53 zynthian-nord startx[487]: ERROR:root:Jack XRUN! 
May 10 19:10:13 zynthian-nord startx[487]: ERROR:root:Jack XRUN! 
May 10 19:10:13 zynthian-nord startx[487]: ERROR:root:Jack XRUN! 
May 10 19:36:55 zynthian-nord startx[487]: ERROR:root:Jack XRUN! 
May 10 19:36:55 zynthian-nord startx[487]: ERROR:root:Jack XRUN!

wyleu, 2019-05-11:
Still both running the next morning...

guysoft, 2019-05-12:
``Jack XRUN`` tends to happen when jack settings are too demanding on the CPU, and it skipps/dies.

riban-bw, 2019-05-12:
Restarting zynthian fixes this (not ttymidi - that is another issue) so it would seem JACK server is still functioning but zynthian has lost comms. This could be python jack client?

jofemodo, 2019-05-12:
Hi @guysoft !
It's not a CPU problem. The problem appears when no engine is loaded and no playing at all. CPU is almost idle, etc. It's something a lot more bizarre ... believe me ;-)

jofemodo, 2019-05-12:
@riban-bw , could you take a look to your jackpeak code? It's been added quite recently and it's a perfect candidate for causing a problem like this. Tomorrow i will try disabling it to see if the problem remains ... 

Python jack client is not a strong candidate because MIDI goes from zyncoder to engines.  Of course, zyncoder library is a perfect candidate too ;-D

riban-bw, 2019-05-13:
I will take a look at both of those modules which I have touched recently. (Maybe I am the perfect candidate!)

I observed today that this issue occurred exactly one hour after the last zynthian log message which was
```
JackPosixSemaphore::TimedWait err = Connection timed out
SuspendRefNum error
JackClient::Execute error name = jackpeak
JackPosixSemaphore::TimedWait err = Connection timed out
SuspendRefNum error
JackClient::Execute error name = ZynMidiRouter
```
This in turn had occurred 10 minutes and 4 seconds after the application had completed startup messages. This might be relevant.

JACK server appears to continue working. I was able to stop zynthian, start fluidsynth connected to JACK and trigger notes which played out. I could not connect jack midi until after I had restarted a2jmidid and could not connect ttymidi until after restarting ttymidi.

jofemodo, 2019-05-13:
Yes! I observed this sequence too. I also guess is related ... ;-)

riban-bw, 2019-05-13:
My earlier info was not accurate. When this issue occurs, stopping zynthian then starting fluidsynth and running jack_connect I am able to connect midi sources to fluidsynth and get audio out so JACK server and a2jmidid are both working. ttymidi is not until a restart. It _may_ be ttymidi that is causing the issue - worth investigating but I think we can probably rule out JACK server and a2jmidi for now.

riban-bw, 2019-05-13:
Update: Stopped ttymidi service then restarted zynthian. Lost midi connection after some tens of minutes so does not look like ttymidi is to blame. (Other issue with ttymidi not recovering might be related to this problem.)

jofemodo, 2019-05-13:
OK! Here is again:
```

JackPosixSemaphore::TimedWait err = Connection timed out
SuspendRefNum error
JackClient::Execute error name = ZynMidiRouter
ERROR:root:Jack Audio XRUN!
ERROR:root:Jack Audio XRUN!
Server is not running
Server is not running
Server is not running
Server is not running

```

jofemodo, 2019-05-13:
https://github.com/jackaudio/jack2/issues/307

riban-bw, 2019-05-13:
The hour offset in the log messages may be relevant. That jack2 issue mentions a loop due to a comparison between times with negative value. Maybe a similar trigger here.

I have compiled jack2 1.9.12 and am currently running zynthian against it. Let's wait and see if it goes wrong...

riban-bw, 2019-05-13:
JACK2 1.9.12 experiences the same issue.

riban-bw, 2019-05-13:
This does not seem to be triggered by xruns. I set the buffer size small to trigger xruns and this issue did not occur when the xruns began.

riban-bw, 2019-05-13:
Did Aruk work without this issue before? If so, when was the last working build of Aruk? (I haven't used Aruk before so don't know whether this is a new issue or has always existed.) If we can test previous versions we may be able to pinpoint when things fell apart.

jofemodo, 2019-05-14:
If the issue is also present in 1.9.12, please, send the error log. It shouldn't contain the:

`JackPosixSemaphore::TimedWait err = Connection timed out`

line, that is the first error line in the 1.9.11 error trace. Remember falkTx 's words:

`JackPosixSemaphore is not used anymore since 1.9.11-RC1, please update your jack2 version and try again.`

Anyway, i will double check the 1.9.12 ;-)

This issue is getting hard to solve ... bufff!!

jofemodo, 2019-05-14:
Hi @riban-bw!
I've compiled the 1.9.12 and zynthian is running stable now, several hours after.
Could you double-check it? What procedure did you use for compiling, installing and running the 1.9.12? 

Thanks!

wyleu, 2019-05-14:
Does this mean I've got to spend another night listening to a very slow arpeggio on two different machines again...?

riban-bw, 2019-05-14:
@wyleu no this issue happens (often within about 10 minutes of boot) without any user input. The test is simply to boot Zynthian then periodically check if playing a note on a connected MIDI keyboard triggers a sound out of the Zynth. It can also be detected by monitoring the logs for "Server is not running".

@jofemodo I compiled by:

- git checkout tags/v1.9.12
- cd jack2
- ./waf configure
- ./waf
- ./waf install
- systemctl stop zynthian
- systemctl stop a2jmidid
- systemctl stop jack2
- /usr/bin/jackd <command line parameters - I can't check from here>
- systemctl start a2jmidid
- systemctl start zynthian

I also checked `/usr/bin/jackd --version` shows 1.9.12.

jofemodo, 2019-05-14:
Are you sure? This should install in "/usr/local" ... documentation says that:

http://jackaudio.org/faq/build_info.html

I followed the same steps and in my zynthian it was installed in "/usr/local" ...

Take care, because calling the binary without the full path will take the jackd installed in /usr/local/bin.

```
root@zynthian:~# jackd --version
jackdmp version 1.9.12 tmpdir /dev/shm protocol 8
root@zynthian:~# /usr/bin/jackd --version
jackdmp 1.9.11
Copyright 2001-2005 Paul Davis and others.
Copyright 2004-2014 Grame.
jackdmp comes with ABSOLUTELY NO WARRANTY
This is free software, and you are welcome to redistribute it
under certain conditions; see the file COPYING for details
jackdmp version 1.9.11 tmpdir /dev/shm protocol 8
```

Regards,

jofemodo, 2019-05-14:
Also, starting a2jmidid & zynthian services will force the starting of jack2 service, that will use the old binary (1.9.11).

Please, first confirm that you installed it in "/usr/local", and then you should change the binary path in the service file:

`/etc/systemd/system/jack2.service`

Regards,

jofemodo, 2019-05-14:
OK! I can confirm that the problem is still there ;-( Upgrading to v1.9.12 doesn't solve it!
I've contacted falkTX ... perhaps he can give us some tip ;-)

wyleu, 2019-05-14:
Yep both mine still have the fault.

wyleu, 2019-05-14:
The zynthian-amp2.local machine crashed in the same fashion with a running sequence, after about 10 minutes.

wyleu, 2019-05-14:
zynthian-nord.local also just gone down with a running sequence. Machine built this morning from a new image downloaded last night & updated today.

riban-bw, 2019-05-14:
Guys! I _may_ have it.... Each time I get the JackPosixSemaphore error (or similar) which triggers this issue the timestamp of the log messages jumps by an hour. This coincides with htpdate log indicating a change of time by one hour. It looks like one or more of the time servers are returning an the wrong time / timezone which in turn is upsetting JACK server. I have stopped the htpdate service and will see if the issue occurs...

riban-bw, 2019-05-14:
Yep! This seems to be it. I left it for an hour without htpdate running and everything was fine. I ran `htpdate -q www.pool.ntp.org` which reported "Offset 3600.000 seconds". I then ran `htpdate -s www.pool.ntp.org` and immediately Zynthian failed with "JackPosixSemaphore::TimedWait err = Connection timed out". This definitely looks like the issue. As a work-around we could disable htpdate service until we have resolved the problem. Also, we don't really need to update the time after first boot - it is only a synthesizer!

guysoft, 2019-05-14:
That's an epic bug

riban-bw, 2019-05-14:
For some reason, htpdate is reporting one hour offset. I can't see why. ntpdate works fine. Why are we using htpdate rather than ntpdate?

I would suggest:

- Set time on boot only
- Monitor for JACK server failure and restart zynthian (or at least jack clients) on error
- Use ntpdate

jofemodo, 2019-05-14:
Hi @riban-bw !

Yeahhhhhhhhh! You are a good hunter, my friend!! ;-) 

Regarding your suggestions:

+ The only moment we need to update the time/date is just before updating the software. Anyway, having a "more or less" updated date/time is interesting. I would like to maintain the daemon running if the problem is solved. In Gorgona htpdate has been running without causing any problem.

+ We are using htpdate because it performs better in restricted networks and normally have no problems with routers and firewalls.

Anyway, it seems that the problem should be solved:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=876687

I will try to keep using htpdate, but if the problem persist, i will follow your advice ;-)

Regards,

jofemodo, 2019-05-14:
htpdate v1.2 seems to solve the problem:

http://www.vervest.org/htp/?download

riban-bw, 2019-05-15:
I stopped htpdate service and the time was still being corrected each time I changed it. I found that systemd-timesyncd is also running. We don't need both running. I also found that manually setting the time has the same effect of breaking jack clients. This only happens if the time change is greater than 2149s.

wyleu, 2019-05-17:
> Closed #93 <https://github.com/zynthian/zynthian-sys/issues/93> via
> 045561d

You don't get this problem with a Fender Rhodes !! :-D

jofemodo, 2019-05-30:
We still have the issue:

```
May 24 01:18:19 zynthian startx[454]: libjackpeak registering as 'jackpeak'.
May 24 01:18:19 zynthian startx[454]: libjackpeak created input ports
May 30 11:33:04 zynthian startx[454]: JackPosixSemaphore::TimedWait err = Connection timed out
May 30 11:33:04 zynthian startx[454]: SuspendRefNum error
May 30 11:33:04 zynthian startx[454]: JackClient::Execute error name = ZynMidiRouter
May 30 11:33:04 zynthian startx[454]: JackPosixSemaphore::TimedWait err = Connection timed out
May 30 11:33:04 zynthian startx[454]: SuspendRefNum error
May 30 11:33:04 zynthian startx[454]: JackClient::Execute error name = jackpeak
May 30 11:33:04 zynthian startx[454]: Server is not running
May 30 11:33:04 zynthian startx[454]: Server is not running
May 30 11:33:06 zynthian startx[454]: Server is not running
May 30 11:33:06 zynthian startx[454]: Server is not running

```

As soon as the system update the date (currently only systemd-timesyn is running, no htpdate), jackd crashes.

We can easily solve the issue by not updating the date/time, but we need to do it before updating the software. I'm thinking about it ...

Anyway, jackd should not do that!!! ;-( I will open an issue in jack2 repository ...

jofemodo, 2019-05-30:
https://github.com/jackaudio/jack2/issues/469

riban-bw, 2019-05-30:
We could update time only on boot.
We could add a watchdog which monitors for various conditions and tracts accordingly.

jofemodo, 2019-05-30:
Normally, jack daemon starts before network is ready. It should be like that, because we need the sound subsystem ASAP, while the network can become available in any moment and it's not a core component. Zynthian can work without network, but not without jackd ;-)

The watchdog could restart the UI if it crashes, but can not avoid the problem if date/time is changed. 

We could disable automatic date/time sync and do it only when needed, aka, when updating the software, but we should restart the UI after updating, although no changes are downloaded. This would improve current behaviour, but it's a sub-optimal solution.

riban-bw, 2019-05-31:
I quite agree that we should avoid any delay to start-up. We also want to avoid an unexpected failure of JACK, e.g. performing a time update when using the synth so performing the time update before a software update makes sense. This would mean that the RPi clock will be wrong after each boot until an update is performed (no RTC) which may confuse fault diagnostics, etc. It may also give odd timestamps for files (.mid, .wav, etc.) and could have an impact on configuration files, e.g. if system uses save timestamp in any decisions.

jofemodo, 2019-06-03:
We are making progress ... ;-)

https://github.com/jackaudio/jack2/issues/469

I think we could have hope of a proper solution ;-)

jofemodo, 2019-06-04:
Now it works like a charm. As i suspected, we was not correctly installing & running the version 1.9.12.
After configuring LD_LIBRARY_PATH in the right places, everything works fine and you can change date/time as many times as you want without earing a glitch ;-)

commit 045561d9 (2019-05-16): + Fix #93: Disable htpdate service and add condition for not using it on stretch (Aruk) updates. Stretch includes a better synchronization service by default.
+ Add needed packages and config files for the WIFI Hostspot: hostapd, dnsmasq
+ New "set_wifi.sh" script for configuring wifi mode: off/on/hostspot
+ New "wifi-setup" service for configuring the wifi mode on boot
commit fdfa6d83 (2019-06-04): Fix #93: Install jack audio 1.9.12 and modify services and scripts for use it. Also configure LD_LIBRARY_PATH when needed.
closed 2019-06-04 by jofemodo