zynthian/zynthian-sys #94: Issues with JACK will slow reboot

riban-bw, 2019-05-11:
If JACK2 Audio Server or Alsa to Jack Bridge have any issues then the shutdown (and hence reboot) is delayed whist the service is forced down. The time for each of these is in minutes which can delay a reboot / shutdown by significant duration. This is a poor user experience, e.g. a user has an issue and wants to reboot their Zynthian - takes many minutes. This would be unacceptable for live performance and embarrassing for sessions.

jofemodo, 2019-05-13:
In such a case, a simple unplug + plug would do the task perfectly. Also, i never found that problem, because until now jack/a2jmidid never failed. Very soon jack stability problems will be solved ...

jofemodo, 2020-03-10:
I think this is solved now. Please, re-open if it's not ...

closed 2020-03-10 by jofemodo