zynthian/zynthian-sys #95: zynthian midi autoroute stops

riban-bw, 2019-05-11:
I left zynthian running for a few hours with Noize Mak3r loaded. It had a USB MIDI keyboard (QX49), USB MIDI interface (MIDI Sport 2x2) and hardware MIDI (ttyMIDI) connected. When I returned it was not responding to MIDI input from any channel and would not play any audio out, e.g. Test Audio. Checking MIDI and Audio Routing with jack_lsp I observed that synth audio remained routed to output but there was no MIDI routing. Turning QX49 off and on did not trigger a reconnection of MIDI. The rest of the zynthian UI seemed to be operational, e.g. I was able to remove and add synth engine. A restart of the zynthian ui (from command line `systemctrl restart zynthian`) resulted in normal operation.

riban-bw, 2019-05-11:
May be related - ttymidi would not connect even after restarting zynthian. Had to reboot to get it to work.

riban-bw, 2019-05-11:
JACK continued to run throughout.

jofemodo, 2019-05-12:
Could you check the logs?

`journalctl -u zynthian`

I think it's the same issue than #93 ...

riban-bw, 2019-05-13:
I think this is likely to be related to #93 so will close this issue as a duplicate.

closed 2019-05-13 by riban-bw