zynthian/zynthian-sys #98: Add support for ST7735R display

riban-bw, 2019-05-17:
The Sainsmart 1.8" TFT is a low cost SPI TFT display based on ST7735R chipset. This is not currently supported by Zynthian. fbtft driver supports ST7735R based displays. It would be advantageous to add support for this range of displays .

riban-bw, 2019-05-17:
I have already figured out how to do this for the Sainsmart 1.8" display. It would be advantageous to make this generic for all ST7735R displays with appropriate configuration in webconf.

- Add section to zynthian-webconf/lib/display_config_handler.py
```
['Sainsmart 1.8', {
	'DISPLAY_CONFIG': 'dtoverlay=sainsmart18,rotate=90',
	'DISPLAY_WIDTH': '180',
	'DISPLAY_HEIGHT': '128',
	'FRAMEBUFFER': '/dev/fb1'
}],
```
Maybe that should be ST7735R rather than Sainsmart.
- Create sainsmart18.dtbo from sainsmart18.dts:
```
dtc -@ -I dts –O dtb –o sainsmart18..dtbo sainsmart18..dts. 
```
I have a sainsmart18.dts source file which works. Where should we store this?
```
/*
 * Device Tree overlay for SPI TFT LCD with ST7735R chip 160x128 (1,8") 
 *
 */

/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835", "brcm,bcm2708", "brcm,bcm2709";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			status = "okay";

			spidev@0{
				status = "disabled";
			};

		};
	};

	fragment@1 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			sainsmart18@0{
				compatible = "sitronix,st7735r";
				reg = <0>;
				pinctrl-names = "default";

				name = "sainsmart18";
				spi-max-frequency = <48000000>;
				rotate = <90>;
				buswidth = <8>;
				fps = <50>;
				height = <160>;
				width = <128>;
				reset-gpios = <&gpio 25 0>;
				dc-gpios = <&gpio 24 0>;
				cs-gpios = <&gpio 8 0>;
				debug = <0>;
			};
		};
	};
};
```
- Add zynthian-sys/custom/display/Sainsmart18/boot/overlays/sainsmart18.dtbo (Note: Folder must have same name as used in webconf)
- To enable backlight add zynthian-sys/custom/display/Sainsmart18/config/zynthian_custom_config.sh
```
gpio -g mode 12 pwm
gpio -g pwm 12 1000
```
or
```
gpio -g mode 12 output
gpio -g pwm 12 1
```
Note: Backlight control GPI should ideally be configurable from webconf. Need to figure out how to do that.

jofemodo, 2019-05-17:
When done, it should be added to Webconf too...

-----------------

BTW, where did you get the DTS? Is it from FBDBI project? I see that it's "dead" and the newer FBTFT, that is already in the official linux kernel staging. Curiously, in the FBTFT repo there is no trace of Sainsmart18 DTS ...

It would be interesting to support all the FBTFT devices ... 

Regards,

riban-bw, 2019-05-18:
The dts is a combination of efforts from various sources on the web and a bit of fixing by me. It does seem to have been dropped out of the fbtft overlay although the driver supports it so it still works. I have done some more experimentation and found that adding the line:

`led-gpios = <&gpio 12 1>;`

to the dts does in fact turn on the backlight so we don't need the gpio control script.

I agree we should support as many displays as possible. I haven't got entirely my head around how the overlay works. I can't see what pokes this overlay and what in it loads fbtft. For now it would be good if I didn't have to keep making these changes each time I install Zynthian.

I will create a patch.

riban-bw, 2019-05-18:
Patches in branch commits:
Zynthian-sys: f363f5f1934efeb0f7a300aecfe1cf0f9e6635db
Zynthian-webconf: af8f6362181760de0bffcedd66f80d4cad73cfcd

riban-bw, 2021-07-05:
I think this should now work.

closed 2021-07-05 by riban-bw