zynthian/zynthian-sys #99: /etc/hosts using wrong hostname for localhost

smiths73v3, 2019-05-18:
/etc/hosts contains:
127.0.1.1   raspberrypi
>sudo true 
sudo: unable to resolve host zynthian

Should contain entry for host name specified in /etc/hostname
>cat /etc/hostname
zynthian

set /etc/hosts to contain entry:
127.0.1.1  zynthian

`>sed -e "s/raspberrypi/`cat /etc/hostname`/" /etc/hosts > /etc/hosts.new 
>mv /etc/hosts /etc/hosts.old
> mv /etc/hosts.new /etc/hosts
>sudo true
>
`

guysoft, 2019-05-19:
The hostname is being changed twice in the image build.
From the build.log of last nightly.

Change one form ZynthianOS's configuration:
```
+ cp /CustomPiOS/modules/base/end_chroot_script chroot_script
+ chmod 755 chroot_script
+ cp /CustomPiOS/common.sh common.sh
+ chmod 755 common.sh
+ chroot . usr/bin/qemu-arm-static /bin/bash /chroot_script
+ '[' -n '' ']'
+ '[' -n '' ']'
+ '[' -f /etc/hostname ']'
+ read FILE_HOST_NAME
+ FILE_HOST_NAME=zynthian
+ echo zynthian
+ sed -i -e s@zynthian@zynthian@g /etc/hosts
+ '[' -d custompios_export ']'
+ rm chroot_script
```

Change two from the zynthian-sys repo:

```
export ZYNTHIAN_CHANGE_HOSTNAME=yes
+ ZYNTHIAN_CHANGE_HOSTNAME=yes
```

smiths73v3, 2019-05-19:
hostname entry in /etc/hostname does not appear in /etc/hosts with a 127.0.x.x address, causing hostname lookup failures.

>\>cat /etc/hostname
>zynthian
>\>cat /etc/hosts
>127.0.0.1       localhost
>::1             localhost ip6-localhost ip6-loopback
>ff02::1         ip6-allnodes
>ff02::2         ip6-allrouters
>
>127.0.1.1       raspberrypi

Where the last entry should be "zynthian" not "raspberrypi" or whatever /etc/hostname has been changed to.

In the script above, the sed modify to /etc/hosts appears wrong. I believe it should be:
>\>sed -i -e s@raspberrypi@zynthian@g /etc/hosts

jofemodo, 2019-05-20:
Sorry, i can't reproduce the error.

smiths73v3, 2019-05-25:
Possible fix, small change to setup script. See pull request.

jofemodo, 2019-05-30:
OK! I've merged the pull request. I will change the webconf code accordingly.

closed 2019-05-30 by jofemodo

smiths73v3, 2019-06-05:
Verified in my RC2, this is resolved.

commit dd0c4e42 (2019-05-30): Fix #99: Improve hostname-change code for being more robust.