zynthian/zynthian-sys #100: Add screen setting for Standard Pi 7" Touchscreen.

wyleu, 2019-05-20:
The standard 7" touchscreen display when used with 'standard' mounting comes up inverted.

A clear and concise description of what you want to happen.

Add LCD_ROTATE=2 to 7" touchscreen setting, as IT will be far easier fro someone to remove this if they don't want it, than the overhead of finding the correct syntax.

riban-bw, 2019-05-20:
It may be a good idea as a generic policy to add all parameters to the default configurations, even if they have default values to allow anyone to tweak a configuration without delving into the source code. Maybe the range of parameters might be restricted to the useful ones, e.g. some display driver overlays have a debug setting which we may wish to omit.

There is the risk that the configuration becomes bloated or confusion ensues over which elements do what or are actually required, especially if configuration becomes obsolete.

jofemodo, 2020-06-10:
Hi @wyleu!

Is this issue still an issue?
In such a case ... could you create a PR for it?

Thanks!

jofemodo, 2021-07-05:
This should be solved via "Display Calibration" tool.

closed 2021-07-05 by jofemodo