zynthian/zynthian-sys #101: WiFi Hotspot mode not working

riban-bw, 2019-05-20:
- Navigate to http://zynthian.local/api/sys-wifi
- Select "Hotspot" from Mode list
- Click "Save"
Expected behaviour: Hotspot enabled.
Actual behaviour: WiFi On mode selected.

jofemodo, 2019-05-20:
hostapd & dnsmasq packages were not correctly installed by the setup script.

riban-bw, 2019-05-20:
This does not seem to be fixed by and update today. I enable WiFi and I see the Zynthian connected to my access point. If I subsequently enable WiFi Hotpot I do not see the Zynthian as an access point using a WiFi monitor on my Android phone and cannot see the network from Ubuntu Mate or Windows laptops. The webconf also behaves the same, i.e. selecting Hotspot results in the webconf showing Wifi on.

During an update I see messages:

sed: can't read /etc/hostapd/hostapd.conf: Not a directory

jofemodo, 2019-05-20:
It should be fixed, but you have to wait until the new SD image is built, my friend ;-)

riban-bw, 2019-05-25:
WiFi Hotspot now working.

commit 554649a1 (2019-05-20): Fix #101: Fix installation of hostapd and dnsmasq packages.
closed 2019-05-25 by riban-bw