zynthian/zynthian-sys #102: Stable / bleeding edge updates

riban-bw, 2019-05-21:
I want to be able to run a Zynthian with a stable configuration and software updates so that I may use it for production purpose. I want to be able to run a Zynthian with latest changes and fixes so that I may test issue resolution and contribute to the project.

It may be advantageous to allow a user to select between different update methods. This may involve having "Stable" and "Develop" or "Beta" or similar git branches. The user may select from the UI or webconf which branch to follow and updates will pull from that branch.

It should be simple to allow a user to test a feature then return to the stable branch.

Development may use a two step merge process:

- Fix applied to a new branch
- Fix branch merged to develop branch
- Develop branch merged to stable branch when deemed stable

It is possible for users to manually change their branches using command line git commands but some (many?) may find this daunting and prone to error.

There should be fair warning to users that following the development branch may render their device unstable.

It may be appropriate to use more user friendly language rather than git / branch / etc. Some project use the term Beta Tester for users who are willing to be early adopters.

jofemodo, 2019-05-21:
Yes. Zynthian is reaching a point where all this start to have sense ;-)
I think the easier will be to have 2 main branches:

+ Master Branch (testing)
+ Stable Branch (production)

On webconf, we should add a simple option for changing between them. Also, we could have an "advanced" branch settings, that allow to change the branch for every repo separately. That will be very useful for developers.

riban-bw, 2019-05-21:
That would be good. We should probably standardise on the naming of the two branches for all repos. I wonder whether the use of "Master" may lead users to believe it is the one they need rather than Stable? Of course this could be obfuscated in webconf.

guysoft, 2019-05-22:
Depends how you want to go around it.

Another way would be to have a release cycle.

1. You add features that make the release unsable
2. at some point you feature freeze and work on stability and completeness.
3. You release a release candidate from the nightly builds, and keep until all problems a fixed and users are not reporting the RC is giving problems
4. You release 
back to 1

riban-bw, 2019-05-22:
I suspected a combination of both, rolling updates to the current stable whilst more substantial changes are in a formal release. I quite like the concept of a rolling update of stable unless there is a substantial change. This is how the project seems to have been rolling so far with Aruk release relating to a change of OS release (Stretch). Everything else has been rolling. This is great but I propose we ensure fixes to the stable branch are fully tested in the unstable branch first. Of course this sounds simpler than it actually is. Applying some patches from unstable to stable without applying them all may prove more challenging than a formal update. I want to see enhancements in the stable version as early as possible. To take a recent example, audio level meters is a boon for many which seems silly waiting until the next stable release which might take months to produce.

We need to minimise overhead whilst providing fairly simple access to stable and development versions.

Maybe there is a middle ground where we have the stable version and the development version and a staging version. When there are features or bugs that we feel should be moved to the stable release we branch develop to staging and encourage users to test and use this version for non-critical activity, e.g. personal practice, composing, etc. i.e. not performance or studio rehearsal sessions. There could be a defined period for feedback after which staging is merged into stable. Develop continues to be available for new issues. I guess this is similar to @guysofts suggestion. (Waffle....)

guysoft, 2019-05-24:
Its my experience that you have two states.
"a nightly" and "a release that was tested". Unless you are a massive project like Debian, or something that people run on production, there is no place really for "a stable branch", because any change you add could break it just as badly as a development branch. And we don't have the capacity to both maintain two branches, and have people testing them.

mheidt, 2019-06-01:
I am in favour of keeping it as simple as possible. I am aware of the hell of merging and like the idea of feature freezes better.

Nevertheless in case we want to change the branches via webconf, this might become handy:

https://gitpython.readthedocs.io/en/stable/tutorial.html

riban-bw, 2019-06-08:
I think it would be advantageous to allow user to select between branches via webconf and / or UI to allow users to quickly toggle between testing new features and having a stable zynthian. (Not everyone has the luxury of testing, staging and production zynthians!)

I understand the concerns about a rolling release. The reason I suggested it is because that is what has been available so far. Gorgona Omega has been updated regularly and I think this approach has benefit but also risk. I too fear merge hell - I have fought with git more times than I would care to remember. This is best managed by a product owner (@jofemodo) managing the merges to staging branch and insisting / ensuring that PRs are fit, i.e. they will merge neatly without excessive effort. This may mean ensuring a PR / development branch is synchronised with production branch and avoids touching files being updated by other changes. This can be imposed by selecting changes to apply carefully, maybe only one change at a time (slow but safe). Aruk differs in that it is a substantial change, moving to a new OS release and deserves extra change management.

So this idea is to have these branches:

- Production - The most stable and default version
- Staging - Production + changes already tested individually - next version for users to easily test
- Development - Production + individual change (several development branches, each representing a fix or enhancement)

We can have the equivalent to agile sprints where Staging has a set of Development branches merged at the start of the sprint (feature freeze) and undergoes testing until being merged into Production within a week of two. (I know that isn't the same as a sprint - cut me some slack!)

I would anticipate that users could toggle between Production and Staging but may also be able to select Development branches (maybe we have an option to allow toggle between listing those two or all branches).

mheidt, 2019-06-08:
I would suggest tags instead of branches.
Each daily gets a tag.
And additionally to that we have one "stable" tag. 
And in the webconf you can switch between stable and master.

riban-bw, 2019-06-08:
Tags relate to specific commits which adds undue overhead. Say I create a branch with my fantastic fix it enhancement and want others to test it. Should I create a tag and day, test this tag then, when they say, "great but it should also do this" do I commit changes and tell everyone to use another tag (which I need to name differently to the first)? It feels awkward whereas using branches means that we advertise a fix and whenever someone pulls and checks it out they have the latest iteration. It also means we have a single name for the rolling Staging version rather that new names each day.

mheidt, 2019-06-08:
You are thinking as if you had to coordinate 50 developers.
I was thinking keeping it as it is, developing into master. 
When we release Anuk, that day gets the stable tag.
We carry on developing in master. When there should be a bug in Anuk, we create a branch from that stable tag and cherry pick the fix...and mark this branch as stable...the same for subsequent bugs.
Months later we decide to release Anuk 2.0, we will have a feature freeze and are at the point we are now.

guysoft, 2019-06-10:
Agree with @mheidt

@riban-bw I agree you should be able to switch branches, but I don't think we should support it, because you will get people trying unsable stuff in a custom configuration with no idea how to fix their issue.

jofemodo, 2021-03-19:
I just changed the default branch on all zynthian repos to testing. I prefer to have testing as default, so you can see the latest commits when you access. Anyway, it's not very important for the update system, and i will change the SD setup scripts for using stable.

I also tested the webconf "repository_mod" branch and merge on testing. If you agree, we could merge on stable so everything will be ready for "the change" ;-)

riban-bw, 2021-03-19:
Ship it!!! Yes! This will be good to deploy. We should accompany it with some comms.

jofemodo, 2021-03-19:
OK! The "change" is commited. Everybody updating their zynthian will be moved from master to stable branch. Repos on development branches wont be migrated.

Congratulations @riban-bw ! I hope this is the beginning of a new development era on zynthian ;-)

closed 2021-03-19 by jofemodo